seemoo-lab/nexmon #529: nexutil does not build on raspberry pi os (2022-04-04) 64bit

realies, 2022-05-22:
```
root@raspberry:/home/pi/nexmon/utilities/nexutil# make
cd ../libargp && make
make[1]: Entering directory '/home/pi/nexmon/utilities/libargp'
/ndk-build NDK_APPLICATION_MK=`pwd`/Application.mk NDK_APP_OUT=. TARGET_PLATFORM=android-21
/bin/sh: 1: /ndk-build: not found
make[1]: *** [Makefile:2: libs/armeabi/libargp.a] Error 127
make[1]: Leaving directory '/home/pi/nexmon/utilities/libargp'
make: *** [Makefile:19: ../libargp/local/armeabi/libargp.a] Error 2
```

DrSchottky, 2022-05-22:
https://github.com/DrSchottky/nexmon/commit/1738351bfd1da485f1718c5883801ac069f4bafe

dzervas, 2022-06-19:
That's not the only problem. The Makefile changes into android mode cause of the aarch64 inside uname

GuoJingtao-1997, 2022-07-30:
I also encounter this problem and I think it may be because the Makefile switches to android mode and cause the "ndk-build not found" problem. I try to filter the aarch64 but cause another issue:
```
make: *** No rule to make target 'libs/armeabi/nexutil', needed by 'all'.  Stop.
```
Any suggestion for you guys? Thanks a lot

DrSchottky, 2022-07-30:
As already said, https://github.com/seemoo-lab/nexmon/pull/530

GuoJingtao-1997, 2022-07-30:
Yes, I just followed your commits and successfully install it. Thanks a lot.