seemoo-lab/nexmon #533: Error Trying to Build Nexmon Driver

Lucario117, 2022-06-24:
Log below shows the error
Any tips on resloving the issue?

BUILDING DRIVER for kernel 4.19 brcmfmac_4.19.y-nexmon/brcmfmac.ko (details: l           og/driver.log)
/home/pi/RaspPiNexmonScripts/nexmon/patches/driver/brcmfmac_4.19.y-nexmon/debug.           c:28:5: error: redefinition of ‘brcmf_debug_create_memdump’
   28 | int brcmf_debug_create_memdump(struct brcmf_bus *bus, const void *data,
      |     ^~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/pi/RaspPiNexmonScripts/nexmon/patches/driver/brcmfma           c_4.19.y-nexmon/bus.h:20,
                 from /home/pi/RaspPiNexmonScripts/nexmon/patches/driver/brcmfma           c_4.19.y-nexmon/debug.c:24:
/home/pi/RaspPiNexmonScripts/nexmon/patches/driver/brcmfmac_4.19.y-nexmon/debug.           h:155:5: note: previous definition of ‘brcmf_debug_create_memdump’ was here
  155 | int brcmf_debug_create_memdump(struct brcmf_bus *bus, const void *data,
      |     ^~~~~~~~~~~~~~~~~~~~~~~~~~
/home/pi/RaspPiNexmonScripts/nexmon/patches/driver/brcmfmac_4.19.y-nexmon/debug.           c:61:5: error: redefinition of ‘brcmf_debugfs_add_entry’
   61 | int brcmf_debugfs_add_entry(struct brcmf_pub *drvr, const char *fn,
      |     ^~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/pi/RaspPiNexmonScripts/nexmon/patches/driver/brcmfma           c_4.19.y-nexmon/bus.h:20,
                 from /home/pi/RaspPiNexmonScripts/nexmon/patches/driver/brcmfma           c_4.19.y-nexmon/debug.c:24:
/home/pi/RaspPiNexmonScripts/nexmon/patches/driver/brcmfmac_4.19.y-nexmon/debug.           h:149:5: note: previous definition of ‘brcmf_debugfs_add_entry’ was here
  149 | int brcmf_debugfs_add_entry(struct brcmf_pub *drvr, const char *fn,
      |     ^~~~~~~~~~~~~~~~~~~~~~~
make[2]: *** [scripts/Makefile.build:304: /home/pi/RaspPiNexmonScripts/nexmon/pa           tches/driver/brcmfmac_4.19.y-nexmon/debug.o] Error 1
make[2]: *** Waiting for unfinished jobs....
/home/pi/RaspPiNexmonScripts/nexmon/patches/driver/brcmfmac_4.19.y-nexmon/core.c           :39:10: fatal error: pcie.h: No such file or directory
   39 | #include "pcie.h"
      |          ^~~~~~~~
compilation terminated.
make[2]: *** [scripts/Makefile.build:303: /home/pi/RaspPiNexmonScripts/nexmon/pa           tches/driver/brcmfmac_4.19.y-nexmon/core.o] Error 1
make[1]: *** [Makefile:1522: _module_/home/pi/RaspPiNexmonScripts/nexmon/patches           /driver/brcmfmac_4.19.y-nexmon] Error 2
make: *** [Makefile:50: brcmfmac.ko] Error 2
cd /home/pi/RaspPiNexmonScripts/nexmon/

thomerc1, 2022-09-30:
I also have an issue with this. Below is the output.

  APPLYING FLASHPATCHES gen/flashpatches.mk => brcmfmac43455-sdio.bin (details: log/flashpatches.log)
  APPLYING PATCHES gen/nexmon.mk => brcmfmac43455-sdio.bin (details: log/patches.log)
  BUILDING DRIVER for kernel 4.19 brcmfmac_4.19.y-nexmon/brcmfmac.ko (details: log/driver.log)
/home/pi/MonitorMode/nexmon/patches/driver/brcmfmac_4.19.y-nexmon/core.c:39:10: fatal error: pcie.h: No such file or directory
 #include "pcie.h"
          ^~~~~~~~
compilation terminated.
make[2]: *** [scripts/Makefile.build:303: /home/pi/MonitorMode/nexmon/patches/driver/brcmfmac_4.19.y-nexmon/core.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [Makefile:1519: _module_/home/pi/MonitorMode/nexmon/patches/driver/brcmfmac_4.19.y-nexmon] Error 2
make: *** [Makefile:50: brcmfmac.ko] Error 2

xnumad, 2023-10-15:
@thomerc1 See #535

moshangqingfeng, 2024-04-02:
Have you solved this problem? Can you share your method？