seemoo-lab/nexmon #534: Missing MAC on mon0 interface

WilliamASumner, 2022-07-03:
# TL;DR
Adding a monitor interface works and I can tcpdump on it, but it is missing a MAC, which [owl](https://github.com/seemoo-lab/owl)  seems to be expecting (the default address is always `00:00:00:00:00:00`). *Is this the expected behavior from nexmon?* I am assuming no because the [tutorial](https://owlink.org/2019/05/16/howto-use-airdrop-on-raspberry-pi-3.html) I'm following doesn't indicate there's a need to set one outside of creating the interface.

---
# What I'm using
I'm using a Rapsberry Pi Zero W (not W2) with Raspberry Pi OS Lite:
```bash
lsb_release -a
Distributor ID:	Raspbian
Description:	Raspbian GNU/Linux 11 (bullseye)
Release:	11
Codename:	bullseye
```
```bash
uname -a
Linux raspberrypi 5.15.32+ #1538 Thu Mar 31 19:37:58 BST 2022 armv6l GNU/Linux
```
# What I've tried 
I've tried both the `patches/bcm43430a1/7_45_41_26` and `patches/bcm43430a1/7_45_41_46` firmware patches (as well as the bcm4355 patches) and they all at least allow for the creation of the `mon0` interface:
```bash
sudo iw phy `iw dev wlan0 info | gawk '/wiphy/ {printf "phy" $2}'` interface add mon0 type monitor
sudo ifconfig mon0 up
sudo nexutil -k6
ip a
...
mon0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UNKNOWN group default qlen 1000
    link/ieee802.11/radiotap 00:00:00:00:00:00 brd ff:ff:ff:ff:ff:ff
...
```

I can even use tcpdump and see a bunch of output:
```bash
tcpdump -i mon0
...
10:35:34.675153 836223206us tsft 2437 MHz -71dBm signal Request-To-Send ...
10:35:34.675304 836223387us tsft 2437 MHz -51dBm signal BA ...
10:35:34.679050 836227061us tsft 2437 MHz -68dBm signal Beacon ...
10:35:34.679959 836227479us tsft 2437 MHz -64dBm signal Beacon ...
10:35:34.680134 836227892us tsft 2437 MHz -64dBm signal Beacon ...
```

Note: I've disabled `wpa_supplicant` because I won't be needing it.

# The Problem 
I'm not expecting anyone to solve this for me, but I am stuck at understanding where my actual issue is either in owl, nexmon or somewhere else - and I also want to double check I'm not doing something dumb before I try to understand how nexmon sets the initial address. I've seen a lot of issues on here that address whether or not a wireless chip supports monitor mode, but I can clearly see that it's working (see console output above).

Despite the monitor mode interface working, I can't use the [owl](https://github.com/seemoo-lab/owl) tool. Running with the recommended command results in:
```bash
sudo owl -i mon0 -N
...
10:37:55 ERROR: tun: unable to set HW address
10:37:55 ERROR: Could not open device: awdl0
10:37:55 ERROR: could not initialize core
```

Seeing the first line I went back to the `ifconfig` and noticed the `00:00:00:00:00:00` address. Is this expected behavior in nexmon? I ask because I am able to get owl to start by modifying the source to return the same MAC as my `wlan0` interface. However the [Opendrop](https://github.com/seemoo-lab/opendrop) tool I really want to run (which needs owl) can't be seen by any peers. Owl seems to expect some address other than `00:00:00:00:00:00`... is there something trivial I am missing to get one assigned? Trying to use `ip link dev set mon0 address xxx` and `macchanger` both complain about it being an unsupported operation:
```bash
sudo ip link set dev mon0 address b7:72:ab:39:ec:51
RTNETLINK answers: Operation not supported
```

```bash
sudo macchanger -r mon0
...
[ERROR] Could not change MAC: interface up or insufficient permissions: Operation not supported
```

DrSchottky, 2022-07-09:
Try this workaround (for kernel 5.15)

- Replace nexmon/patches/driver/brcmfmac_5.15.y-nexmon/cfg80211.c with this https://gist.github.com/DrSchottky/809d0eca857895d41224a6de73fded30
- Recompile and reload the driver
- Do not create a mon0 vif ( iw phy etc etc) but switch wlan0 from managed to monitor
  - sudo ip link set dev wlan0 down
  - sudo iw dev wlan0 set type monitor
  - sudo nexutil -m2
  - sudo ip link set dev wlan0 up
- Use wlan0 as your monitor interface

DrSchottky, 2022-07-09:
I commited a change that should fix.
@matthiasseemoo is it possible to merge this? https://github.com/seemoo-lab/nexmon/pull/530

nendonerd, 2022-10-28:
> 

raspi os bullseye, kernel version 5.15.61+, replacing "cfg80211.c", and recompiling the firmware makes wlan0 disappeared. After a reboot, wlan0 appeared, but `iw dev wlan0 set type monitor` resulted in `command failed: Operation not supported (-95)`

closed 2023-11-14 by jlinktu

DonnerGreif1771, 2024-09-01:
Hi, this issue has been closed for some time now, but I'm currently facing the same problem. I'm using a Raspberry Pi Zero 2 W with Raspberry Pi OS lite 32-bit and kernel version 6.6.31. I was able to build and install the firmware patch for bcm43436b0, but my monitor interface does not have an mac address as well. Furthermore, I get the same owl error as mentioned above. Maybe you have any ideas on how to solve this? Thanks you.

commit a80131bd (2023-11-14): Merge: Add aarch64 support to nexutil (#530)

* Add aarch64 support to nexutil

* Revert changes on types.h, nexutil shouldn't depend on this

* Make nexutil no longer depending on types.h

* Update interface type when an existing interface is put in monitor mode

* Reuse MAC addr for the newly created mon vif - Closes #534