seemoo-lab/nexmon #541: port to other mmwave devices

nisarnabeel, 2022-09-20:
Hi, tp link ad 7200 is not in stock in market, could this work on other devices using  Qualcomm’s QCA9500 ?

matthiasseemoo, 2022-09-20:
Its a Broadcom/Cypress only solution.

nisarnabeel, 2022-09-20:
Sorry may you please elaborate , slightly new in the field, i already worked on talon ad 7200, however not mine, i want do to some own exps,
For example both mikrotik wAP 60G and talon use QCA 9500 chipset from qualcomm, i know u reverse engineered talon ad 7200, would you please provide more insights?
I shall be highly thankful
Thank you!

nisarnabeel, 2022-09-20:
Will it be possible for example to use mikrotik wAP 60G?

nisarnabeel, 2022-09-20:
It might be a dumb question, would be it possible to use same patch for nightgear x10 r9000 or mikrotik i mentioned, i shall be highly thankful to you. ( in both cases 60Ghz radio has same wifi chip as tp link)

matthiasseemoo, 2022-09-20:
Ah mmware communication is an exception. Generally, the nexmon firmware patching approch can be applied to any firmware, however, you first need to get an understanding of that firmware. For 2.5 and 5 GHz Wi-Fi chips, we only have that understanding for Broadcom/Cypress devices.

nisarnabeel, 2022-09-20:
Thank you for taking your time to reply, now its clear :)

nisarnabeel, 2022-09-21:
One last thing, u mentioned 2.4 and 5 Ghz,  nexmon is available for broadcom chips, so nexmom could be patched to ideally theoretically to all broadcom devices (a lot already i see the list) btw i use asus rt ac 86u.

While for mmwave its not possible to may be use a patch u developed for tp link ad 7200 to other mmwave routers i mentioned though using same wifi chip? As u said its an exception the mm communication. ( is it correct the way I understood) ?

thank you and sorry for too many queries,
But i think if talons are not in market , there is no mmwave research platform:(

matthiasseemoo, 2022-09-22:
The patches focus on the actual wi-fi firmwares, so if you have another device, with the same chip, you can often use the patched firmware. If the firmware version does not fit to the driver of the other device, then you might need to port the firmware patches to the firmware of the other device. Everything else that is modified in the routers’ firmwares is just tooling around the wi-fi patches.