seemoo-lab/nexmon #550: BCM4389

TQMatvey, 2023-01-05:
Hello, @matthiasseemoo, i have spent many weeks researching monitor mode for BCM4389, had 0 luck, i can provide remote access to S21 Ultra, and any needed blobs, really hope you could help.

Here is and interesting topic i found but it didnt help at all :<
https://forum.xda-developers.com/t/get-bcm4389-into-monitor-mode-for-wifi-sniffing.4525011/

jlinktu, 2023-01-06:
Hi @TQMatvey,
the `bcm4389c1` as found on the S21 Ultra is already in the pipeline and will be supported here soon.

savox-326, 2023-01-07:
@jlinktu 
hi, could update patch for bcm4375 also? at OneUI 3 and newer in kernel changed driver. It is was bcmdhd_101_12 now it is become bcmdhd_101_16. sorry, that my english isn't good (reference to my previous issue)

shandongtlb, 2023-01-09:
> 你好@TQMatvey， 在`bcm4389c1`S21 Ultra 上找到的功能已经在准备中，很快就会在这里得到支持。

Waitting for you, thanks

TQMatvey, 2023-01-12:
> Hi @TQMatvey, the `bcm4389c1` as found on the S21 Ultra is already in the pipeline and will be supported here soon.

any ETAs or any way to see the progress? i am also very interested at reverse engineering, and patching firmwares

shandongtlb, 2023-01-23:
> Hi @TQMatvey, the `bcm4389c1` as found on the S21 Ultra is already in the pipeline and will be supported here soon.

When? Thanks!

shandongtlb, 2023-02-08:
> Hi @TQMatvey, the `bcm4389c1` as found on the S21 Ultra is already in the pipeline and will be supported here soon.

So, when? Im waitting for you, thanks!

jlinktu, 2023-02-09:
A bit more patience required.

shandongtlb, 2023-02-11:
> A bit more patience required.

Okay, waitting for you thanks!

TQMatvey, 2023-03-07:
Hello, it's been 2 months, tiny bump, any news?

TQMatvey, 2023-06-20:
Uhhh any news..?

TQMatvey, 2023-10-26:
@jlinktu could you send nexmon-magisk.zip for testing on S21 Ultra?)

jlinktu, 2023-10-26:
Hi @TQMatvey,
no, I don't provide already patched firmware. You have to build it yourself.
But you can try using the one for the Pixel 7 Pro, might work as well.

TQMatvey, 2023-10-27:
> Hi @TQMatvey, no, I don't provide already patched firmware. You have to build it yourself. But you can try using the one for the Pixel 7 Pro, might work as well.

trying to `make` in nexmon/patches/bcm4389c1/20_101_36_2/nexmon ends up in errors

[image]

i have tried different NDK versions, sourced setup_env.sh..

jlinktu, 2023-10-27:
Try with NDK r11c ([download here](https://github.com/android/ndk/wiki/Unsupported-Downloads#r11c)) as stated [here](https://github.com/seemoo-lab/nexmon#how-to-build-the-utilities).

jlinktu, 2023-10-27:
Btw. the firmware image on the Galaxy S21 Ultra is named `bcmdhd_sta.bin_c1` and for the Pixel 7 Pro it is called `fw_bcmdhd.bin`. So you would need to adapt the build process a bit to be useful...

TQMatvey, 2023-10-27:
> Try with NDK r11c ([download here](https://github.com/android/ndk/wiki/Unsupported-Downloads#r11c)) as stated [here](https://github.com/seemoo-lab/nexmon#how-to-build-the-utilities).

did not help...
https://katb.in/anekuxoyono

TQMatvey, 2023-10-28:
upd: fixed, had to go into utilities, and compile there first

TQMatvey, 2023-10-28:
flashed, wifi is dead, i adapted fw name and path (/vendor/firmware/wifi/bcmdhd_sta.bin_c1

```
nexutil -V
__nex_driver_io: error ret=-1 errno=22
__nex_driver_io: error ret=-1 errno=22
Segmentation fault
```
not sure what to do from here at all...

savox-326, 2023-11-25:
@jlinktu, so, I see that new firmwares (bcm4389, bcm4398) are not supporting monitor and injectoins. Does it mean that newly bcms unsupport it (very hard to add). Or we just need wait until you add it

jlinktu, 2023-12-01:
Hi @savox-326,

it is still possible to add monitor mode and frame injection to those firmwares

if you want it quick and simple, you might copy the code from one of the prior chips like the bcm4375, of course you need to adapt some bits here and there, add the right dummy function addresses and structs and their members

however, as the 4389 and 4398 are 802.11ax and 802.11be chips they should have new things that might be worth adding, so if I would add monitor and frame injection I want to do it in a proper/clean way, but I currently don't have the time for that - might be something for the future though

regarding the 4389, on samsung phones they ship monitor and manufacturer testing firmwares that should already contain functionalities for monitor mode and frame injection, you might want to play with these if you don't want to do the patching

savox-326, 2023-12-08:
@jlinktu can you tell how find addresses of functions? In example latest bcm4375 firmware has 0x13c68b length but the ucode extractor references to 0x289a58 or bigger values of length. Same history with patches. So can you explain what to do?

jlinktu, 2023-12-09:
Matthias' PhD thesis ([here](https://github.com/seemoo-lab/nexmon#read-my-phd-thesis)) and the linked papers ([here](https://github.com/seemoo-lab/nexmon#read-our-papers)) provide detailed insights on how to work with these firmwares. I.a. that the firmware blob is not necessarily loaded at address `0x0`.

jjbyrnes29, 2024-06-06:
I'd hate to necro an open issue here, but if there's any update on support for the BCM4398 on the Pixel devices, I'd love the ability for monitor mode as well

jlinktu, 2024-06-06:
This issue is on the bcm4389, not on the bcm4398. But, yes, at some point we might add monitor mode for it here.