seemoo-lab/nexmon #551: Issues when trying to install firmwaare for Rasperry Pi 4

michalisk13, 2023-01-18:
Hi,

I am facing issues when I try to install the firmware for Raspberry pi running on kernel version 5.15.88-v8+

The wifi module is BCM4345/6 with firmware version 7.45.241.

Here is the error when I run make install-firmware

```
insmod: ERROR: could not insert module /home/pi/nexmon/patches/driver/brcmfmac_5.15.y-nexmon/brcmfmac.ko: Invalid parameters
make: *** [Makefile:192: install-firmware] Error 1

```

The dmesg output looks like this:
```

[ 1567.533995] brcmfmac: disagrees about version of symbol netlink_kernel_release
[ 1567.533998] brcmfmac: Unknown symbol netlink_kernel_release (err -22)
[ 1567.534009] brcmfmac: disagrees about version of symbol wiphy_read_of_freq_limits
[ 1567.534012] brcmfmac: Unknown symbol wiphy_read_of_freq_limits (err -22)
[ 1567.534038] brcmfmac: disagrees about version of symbol netif_tx_wake_queue
[ 1567.534042] brcmfmac: Unknown symbol netif_tx_wake_queue (err -22)
[ 1567.534056] brcmfmac: disagrees about version of symbol netif_rx_ni
[ 1567.534060] brcmfmac: Unknown symbol netif_rx_ni (err -22)
[ 1567.534066] brcmfmac: disagrees about version of symbol _dev_warn
[ 1567.534070] brcmfmac: Unknown symbol _dev_warn (err -22)
[ 1567.534081] brcmfmac: disagrees about version of symbol brcmu_pktq_pdeq_match
[ 1567.534084] brcmfmac: Unknown symbol brcmu_pktq_pdeq_match (err -22)
[ 1567.534095] brcmfmac: disagrees about version of symbol cfg80211_vendor_cmd_reply
[ 1567.534099] brcmfmac: Unknown symbol cfg80211_vendor_cmd_reply (err -22)
[ 1567.534122] brcmfmac: disagrees about version of symbol __pskb_pull_tail
[ 1567.534126] brcmfmac: Unknown symbol __pskb_pull_tail (err -22)
[ 1567.534132] brcmfmac: disagrees about version of symbol netif_rx
[ 1567.534135] brcmfmac: Unknown symbol netif_rx (err -22)
[ 1567.534142] brcmfmac: disagrees about version of symbol __netdev_alloc_skb
[ 1567.534146] brcmfmac: Unknown symbol __netdev_alloc_skb (err -22)
[ 1567.534159] brcmfmac: disagrees about version of symbol cfg80211_mgmt_tx_status
[ 1567.534163] brcmfmac: Unknown symbol cfg80211_mgmt_tx_status (err -22)
[ 1567.534169] brcmfmac: disagrees about version of symbol skb_trim
[ 1567.534173] brcmfmac: Unknown symbol skb_trim (err -22)
[ 1567.534178] brcmfmac: disagrees about version of symbol skb_unlink
[ 1567.534182] brcmfmac: Unknown symbol skb_unlink (err -22)
[ 1567.534206] brcmfmac: disagrees about version of symbol brcmu_pktq_mdeq
[ 1567.534210] brcmfmac: Unknown symbol brcmu_pktq_mdeq (err -22)
[ 1567.534220] brcmfmac: disagrees about version of symbol cfg80211_unregister_wdev
[ 1567.534224] brcmfmac: Unknown symbol cfg80211_unregister_wdev (err -22)
[ 1567.534234] brcmfmac: disagrees about version of symbol brcmu_pktq_penq
[ 1567.534238] brcmfmac: Unknown symbol brcmu_pktq_penq (err -22)
[ 1567.534261] brcmfmac: disagrees about version of symbol brcmu_pktq_pdeq_tail
[ 1567.534265] brcmfmac: Unknown symbol brcmu_pktq_pdeq_tail (err -22)
[ 1567.534273] brcmfmac: disagrees about version of symbol __dev_kfree_skb_any
[ 1567.534277] brcmfmac: Unknown symbol __dev_kfree_skb_any (err -22)
[ 1567.534296] brcmfmac: disagrees about version of symbol netif_carrier_off
[ 1567.534300] brcmfmac: Unknown symbol netif_carrier_off (err -22)
[ 1567.534317] brcmfmac: disagrees about version of symbol netif_carrier_on
[ 1567.534321] brcmfmac: Unknown symbol netif_carrier_on (err -22)
[ 1567.534331] brcmfmac: disagrees about version of symbol brcmu_pkt_buf_get_skb
[ 1567.534335] brcmfmac: Unknown symbol brcmu_pkt_buf_get_skb (err -22)
[ 1567.534352] brcmfmac: disagrees about version of symbol cfg80211_report_wowlan_wakeup
[ 1567.534356] brcmfmac: Unknown symbol cfg80211_report_wowlan_wakeup (err -22)
[ 1567.534388] brcmfmac: disagrees about version of symbol cfg80211_inform_bss_data
[ 1567.534392] brcmfmac: Unknown symbol cfg80211_inform_bss_data (err -22)
[ 1567.534414] brcmfmac: disagrees about version of symbol cfg80211_connect_done
[ 1567.534418] brcmfmac: Unknown symbol cfg80211_connect_done (err -22)
[ 1567.534429] brcmfmac: disagrees about version of symbol wiphy_free
[ 1567.534433] brcmfmac: Unknown symbol wiphy_free (err -22)
[ 1567.534450] brcmfmac: disagrees about version of symbol cfg80211_cqm_rssi_notify
[ 1567.534453] brcmfmac: Unknown symbol cfg80211_cqm_rssi_notify (err -22)
```

Any suggestions on how I can deal with it? 
Is it the firmware version of the Cypress WiFi or is it the kernel version of my Pi

Thanks,
M

michalisk13, 2023-01-19:
The problem seems to be on the raspberry pi side.
I am investigating it. 

Thanks,
M

closed 2023-01-19 by michalisk13

renewedbullet, 2023-08-15:
what was your fix ?