seemoo-lab/nexmon #552: BCM4375

savox-326, 2023-01-18:
Update patch for new binaries at S20, N20 series. From OneUI 3 even with permissive SELinux we cannot use monitor mode. please update

savox-326, 2023-08-10:
@jlinktu so, how long do we have to wait until the 18.41.113 version's patch to come out?

savox-326, 2023-12-13:
@jlinktu I don't know what it should look like, but my compiled bcm4375 (s20 firmware) but my compiled firmware (with patch) no has wlc_monitor_attach if check with strings command though compilation finished successfully

jlinktu, 2023-12-13:
Not sure what I am supposed to do with this information... thank you for telling me, I guess.
Or is there a hidden question in there?

savox-326, 2023-12-13:
> Not sure what I am supposed to do with this information... thank you for telling me, I guess. Or is there a hidden question in there?
Monitor mode finds nothing, maybe because of this

jlinktu, 2023-12-13:
Some assume I'm a wizard 🧙🏻‍♂️ who magically knows what is happening on your local inner machine - surprise: I'm not.

You might help the situation by answering the following questions:
- What do you want to achieve? / What is your goal?
- How are you trying to get there? / What are you doing?
- Is there something specific you want from me? (Try to formulate a question if that is the case.)

The more detailed and precise your answers are the easier it is for me to help you out.

savox-326, 2023-12-13:
Sorry for that, I just thought you know what I was meaning about, cause thehe is many issues with that problem
My device is N9860 (Snapdragon) running on Android 13 (OneUI 5.1), with custom kernel where SELinux always permessive. Stock version of bcmdhd_sta.bin_b1 was equal 18.41.113 or higher. I replaced it to nexmon's unpatched 18.41.8.9 (because that version has monitor and injection patch). It works as stock binary. After patching also works perfectly. But after running monitor mode (with nexutil, with libfakeioctl) airodump starts and sticks on 0 cnannel and finds nothing. Yesterday with strings command I noticed, that my compiled binary no has wlc_monitor_attach string though meanwhile bcm4358 and etc has it

jlinktu, 2023-12-13:
1. `strings` outputs the printable characters contained in the file you are running it on. If your patched firmware file doesn't contain the character sequence "wlc_monitor_attach" in a printable form, `strings` can't output it. This is not equivalent to your firmware missing a function with the C-naming `wlc_monitor_attach`. Nevertheless, you are partly on the right track as the stock firmware `18.41.8.9 (STA)` probably doesn't have the `wlc_monitor_attach` function compiled in. Therefore, Matthias added it manually to his patch code: https://github.com/seemoo-lab/nexmon/blob/ae8addba003ceb68a4217c014242d5c747eeaf36/patches/bcm4375b1/18_41_8_9_sta/nexmon/src/monitormode.c#L97
which means, that the your patched firmware contains it. Also if there is no string "wlc_monitor_attach" in it.
2. You write that you are "[...] running monitor mode (with nexutil, with libfakeioctl) [...]". How exactly are you doing this? (commands) Are there any additional commands involved regarding configuration of the interface? I am not too familiar with `airodump`, therefore I can't really help you at this end. But you can check if the monitor mode is working in general by listening on the interface with e.g. `tcpdump`.

savox-326, 2023-12-13:
I write this for launch monitor mode:
ifconfig wlan0 down
ifconfig wlan0 up
nexutil -m2

Nexutil -m returns "monitor -2147483646" after -m2 parameter and "monitor -2147483647" after m1. Tcpdump also captures nothing

jlinktu, 2023-12-13:
Try setting a channel before enabling monitor mode.
Make sure that there is traffic on this channel too.
E.g. for channel 6
```
ifconfig wlan0 down
ifconfig wlan0 up
nexutil -k6
nexutil -m2
```
Does this change anything?

savox-326, 2023-12-13:
Also no result

jlinktu, 2023-12-13:
What does reading out the channel return?
`nexutil -k`

savox-326, 2023-12-13:
Says "chanspec: 0x1006, 6" and nothing capturing

jlinktu, 2023-12-13:
Well, I guess matthias has tested the patch before committing and pushing it, so I assume it will work somehow. Maybe not directly with `airodump`.
Are you sure there is traffic on that channel?
What about 5GHz channels? Does it work there?
Any obvious errors in the kernel log?
If none of the above helps, you could check directly on the chip if you receive frames in monitor mode by extending the patch.

savox-326, 2023-12-18:
@jlinktu I was busy in last several days. Today I checked all that you wrote. 5ghz is also same as 2.4ghz. After that command:
`ifconfig wlan0 up`
Dmesg shows that spaming log:
`I[0:dhd_rpm_state_t:  506] dhd_runtimepm_state: DHD Idle state!! -  idletime :5, wdtick :100`
Maybe it was working for matthias and others cause he had samsungs with Exynos

jlinktu, 2023-12-19:
This just means that the bus is idle for a specific time duration.

But maybe you have to disable power saving. You could do this by adding the following calls
```C
set_mpc(wlc_for_chanspec, 0);
set_mpc(wlc_other, 0);
```
here:

https://github.com/seemoo-lab/nexmon/blob/29d990bdaea869cb32f5eaa2dbb7efacdcd405aa/patches/bcm4375b1/18_41_8_9_sta/nexmon/src/ioctl.c#L295
   
I don't think that Exynos makes the difference. Its the same WiFi chip.

If the above still doesn't help, there might be a compatibility issue between the firmware and driver.

savox-326, 2023-12-26:
I'll try to figure out how find target patching addresses in new firmware. I hope I will can do that. So, even I use nexmon rom extraction tool, it works but fills binary these errors: ioctl -95, I also tried bcm4389 extraction tool, same result. Will I need bcm's rom?

jlinktu, 2023-12-26:
You can find a ROM dump of the `bcm4375b1` [here](https://github.com/seemoo-lab/bcm_misc). This is a clean dump - meaning that it comes without flashpatches applied. For analyzing firmwares you want to add the flashpatches of the firmware version you are currently looking into. Fortunately, we already provide Makefile targets for this purpose. All you have to do is:
- Dump a clean ROM; or use [the already existing one](https://github.com/seemoo-lab/bcm_misc/tree/bc9fea211568b2fbfbd32df0a3769bf0fe498959/bcm4375b1)
- Place this clean `rom.bin` in the chip specific firmware directory `nexmon/firmwares/bcm4375b1`
- Change into the firmware specific sub-directory `nexmon/firmwares/bcm4375b1/18_41_113_sta`
- Create a full firmware image from the clean ROM, flashpatches, and firmware blob by running `make complete.bin`
The output `complete.bin` is what you want to analyze.

savox-326, 2023-12-31:
Hi, again. I adapted 18.41.8.9 firmware patch to 18.41.113. Just replacing addresses. So the firmware works same as other versions. So, I find one bug on every firmware versions (18.38.18,  18.41.8.9,  18.41.113). If I switch mode to -m1 using nexutil, and choose open wifi (without password) airodump and tcpdump begins showing something. But that data is incorrect. Airodump shows incorrect MACs sometimes first catched one ends with 45:10 (even with different devices)
[image]
[image]
`nexutil -m2` is not showing anything even with any setting, like I said you before

savox-326, 2024-01-08:
@jlinktu I added printf into firmware for some debug but `dhdutil consoledump` says `Operation not supported on transport endpoint`. But DHD_DEBUG have been enabled in bcmdhd long time. Yeah dmesg also empty. I don't know what's wrong with that

jlinktu, 2024-01-08:
If the firmware doesn't support the `"membytes"` get var, dumping the console this way won't work, even when compiling the driver with debugging enabled. But you can also read out the console by using an own IOCTL, e.g. like [this](https://github.com/seemoo-lab/nexmon/blob/3e8fe2a9872ada3ec74f887dfb1e62969dd28c4f/patches/bcm4375b1/18_41_8_9_sta/nexmon/src/ioctl.c#L324).