seemoo-lab/nexmon #554: Modern phone that will capture EAPOL frames?

johnpaul7, 2023-01-23:
For pentesting reasons, I am often needing to capture WPA handshakes which I generally do with my laptop. It would be great if I could do this with a modern Android phone.

I bought a Nexus 6P a couple of years ago for this exact purpose, given that it was on the list of devices that support monitor mode, but unfortunately I tried and failed to get it to successfully capture EAPOL data frames (it would always truncate the packets, see #461).

I'm now looking at buying an S20 Ultra (z3s / G988B). I thought perhaps before I buy it this time(!), to check whether anyone has been able to capture EAPOL frames with it? I can see that I would need to recompile the kernel to turn off SELinux (#471), but I've seen a few other issues (#519 #552) saying that monitor mode doesn't work? Can anyone confirm either way whether this model is worth buying?

If not, what is the most modern phone I can use to capture EAPOL frames (without using external adaptors)?

Thank you!

savox-326, 2023-01-23:
you needn't buy exactly Exynos model. I successfully powered up wifi with Exynos binary driver (bcmdhd_sta.bin_b1 and  other .bin_b1 and .blob files) If you just replace stock binaries to Nexmon patched ones, your wifi doesn't work. They added monitor to old binary for OneUI 2 that used another driver in kernel, it was bcmdhd_101_12. Now, it is bcmdhd_101_16. After switched it to old one, wifi begins work with old binary (that was patched by nexmon), I could connect to spots, and turn up my one but couldn't use monitor, i mean that i ran airodump-ng and got only channel -1 (more or less like this). I think, it just need update. If you want modern phone with eapol, just try OnePlus 3/3T I never heard any problem with his monitor mode

johnpaul7, 2023-01-26:
Thank you @savox-326 for this info. I'm not quite sure I understand everything you said. Are you saying the S20 can't do monitor mode at all? Is there an older kernel that monitor mode used to work with? Why does the table in the README say that S20 can do monitor mode if it's not possible? I think the OnePlus 3/3T is not quite modern enough for me. The S20, which is already a few years old, has octa-core, 108MP camera, 16GB ram whereas the OnePlus 3 is quad-core, 16MP, 6GB. But thank you for the suggestion anyway!

savox-326, 2023-01-29:
@johnpaul7 OneUI is a proprietary firmware for Samsung devices. OneUI 2 is based on Android 10, the Nexmon patch for S20 was also specific to this version. Reading issues, I noticed that owners with OneUI 3 and higher firmware reported a non-working monitor. most likely, when everyone was on OneUI 2, the monitor worked and no one opened an issue. I tried porting wifi driver from OneUI 2 to OneUI 4 (in kernel), wifi worked but monitor still not

yesimxev, 2023-02-27:
You can also capture handskaes with most phones (which they use qcacld3.0 driver) because monitor mode is supported. The drawback is you can't use fixed channel and injection yet. That is availablw on qcacld2.0 phonea though but they are rare

davidrozen76, 2023-03-14:
Nexmon works on Samsung Galaxy S10, S20 and S21 FE.

johnpaul7, 2023-03-14:
@yesimxev What is qcacld? Which phones can use this?

johnpaul7, 2023-03-14:
@davidrozen76 Are you sure nexmon works with the S20? Despite the fact that it's in the table of supported devices, there seem to be several open issues of people claiming otherwise.

davidrozen76, 2023-03-15:
> @davidrozen76 Are you sure nexmon works with the S20? Despite the fact that it's in the table of supported devices, there seem to be several open issues of people claiming otherwise.

Yes, it works perfectly.

savox-326, 2023-03-20:
> > @davidrozen76 Are you sure nexmon works with the S20? Despite the fact that it's in the table of supported devices, there seem to be several open issues of people claiming otherwise.
> 
> Yes, it works perfectly.

What did you do for work it? What is firmware version you have? I recompiled kernel with old driver for just work nexmon's firmware. Wifi AP connectivity begin works, hotspot also, but monitor mode finds nothings always

davidrozen76, 2023-03-20:
> > > @davidrozen76 Are you sure nexmon works with the S20? Despite the fact that it's in the table of supported devices, there seem to be several open issues of people claiming otherwise.
> > 
> > 
> > Yes, it works perfectly.
> 
> What did you do for work it? What is firmware version you have? I recompiled kernel with old driver for just work nexmon's firmware. Wifi AP connectivity began work, hotspot also, but monitor mode finds nothing always and stuck on 0 channel (-1 if I touch arrow button)

18_41_8_9

savox-326, 2023-03-20:
@davidrozen76 I mean firmware version of your system (OneUI 4.1 or 5.0/5.1). So, don't know why that doesn't work on my device

davidrozen76, 2023-03-22:
> @davidrozen76 I mean firmware version of your system (OneUI 4.1 or 5.0/5.1). So, don't know why that doesn't work on my device

It works regardless of OneUI's version as long as you use firmware version 18.41.8.9

johnpaul7, 2023-03-23:
> > @davidrozen76 Are you sure nexmon works with the S20? Despite the fact that it's in the table of supported devices, there seem to be several open issues of people claiming otherwise.
> 
> Yes, it works perfectly.

@davidrozen76 This sounds promising. Do you have an S20 yourself? I would greatly appeciate if you could verify whether nexmon is truly able to capture the entire EAPOL frames in a 4-way WPA handshake without truncating them? (see #461 for the issues I was having with the Nexus 6P which was also supposed to work perfectly). Basically does airmon-ng / aircrack-ng work?

yesimxev, 2023-03-24:
I think Nexus 6P is the only one which had this EAPOL issue

savox-326, 2023-03-24:
@davidrozen76 so why for others it wasn't work? Maybe you just didn't update bootloader or monitor works only exynos models (but why other things work)

johnpaul7, 2023-03-24:
> I think Nexus 6P is the only one which had this EAPOL issue

@yesimxev The Nexus 6P (bcm4358) and the S7 (bcm43596a0) both have the EAPOL issue (see #231), despite both monitor mode and packet injection working otherwise. I still don't know why it's only the EAPOL frames that are affected.

@matthiasseemoo (#231) suggested it might be that

> the d11 core splits each received data frame and only passes the first part to the arm firmware and the rest directly to the host

But I couldn't confirm if that was the reason or not. I wondered whether it was a security implementation?

Maybe it would be good to have another column on the supported devices table that says whether they can capture EAPOL handshakes? I think this would be very useful for a lot of people as it's probably one of the more common use cases for monitor mode.

> Monitor mode and frame injection both work flawlessly. On S10, S20, S21 and S21 FE, as I mentioned earlier. They all use bcm4375b1.

@davidrozen76 I would be very happy if this is true. But I would really like to know if anyone can confirm for sure that they have specifically captured a full WPA handshake? This is the only reason I would buy the phone and I'd prefer not to spend several hundred dollars and find out it has the same problem.

davidrozen76, 2023-03-25:
@johnpaul7 yes, the exynos variants of the models specified above can fully capture EAPOLs with no malformed packets.

johnpaul7, 2023-03-26:
@davidrozen76 Wonderful! Thank you!

closed 2023-03-26 by johnpaul7

savox-326, 2023-04-30:
@davidrozen76 could you give me your telegram? 😅 I want to start nexmon on my snapdragon device. On dmesg i see message like this: "Bus is in power save state. Skip proccessing rest of ring buffers" and this "DHD idle state" from DHD (Broadcom wifi driver)

KikMyaz, 2023-08-29:
> > > > @davidrozen76 Are you sure nexmon works with the S20? Despite the fact that it's in the table of supported devices, there seem to be several open issues of people claiming otherwise.
> > > 
> > > 
> > > Yes, it works perfectly.
> > 
> > 
> > @davidrozen76 This sounds promising. Do you have an S20 yourself? I would greatly appeciate if you could verify whether nexmon is truly able to capture the entire EAPOL frames in a 4-way WPA handshake without truncating them? (see #461 for the issues I was having with the Nexus 6P which was also supposed to work perfectly). Basically does airmon-ng / aircrack-ng work?
> 
> Monitor mode and frame injection both work flawlessly. On S10, S20, S21 and S21 FE, as I mentioned earlier. They all use bcm4375b1.

Uhh unless you can show us some evidence this is probably not true... I have got S10 (both Snapdragon and Exynos) on One UI 2 (on Q) and after patching with nexmon they are both reporting the same ioctl 95 error.