seemoo-lab/nexmon #555: rpi 3b bcm43430: aireplay-ng is useless

wtswkz, 2023-02-14:
The firmware is successfully patched with kernel 5.15.61,  and the wifi interface can be set to monitor mode, and airodump-ng works fine. 
But the aireplay-ng deauth attack failed, it cannot deassociate sta from the AP. However, if I use my laptop's wifi interface to perfom the aireplay-ng deauth attack, then the sta can be successfully deassociate from the AP. 
So what might be the reason of this problem?

AzureDominus, 2023-04-22:
Same problem

Gh0stD3V, 2023-07-14:
pretty sure the pi3b+ and 4 dont support Injection

DrSchottky, 2023-10-01:
Injection on 5.15+ is broken until this https://github.com/seemoo-lab/nexmon/pull/561/commits/a0262c781933f682e301954654672099e6057557 gets merged