seemoo-lab/nexmon #556: question about patching wifi fw?

claud-wait, 2023-02-26:
hi dev, so after reading README.md and supported wifi fw, i see that my phone had a similar wifi fw bcm43436 like rapberry pie and it has similar/near to/version 9.88.36 my question: can i patch my wifi fw ? and thanks for reading

ShayLynia, 2023-02-26:
You will need a network adapter that can handle packet injection, and
monitor mode first, then yes, should be able to. YouTube is a valuable
resource, I find using a good walkthrough at the same time is helpful in
case I miss something. If I remember right Network Chuck covers RPI and
WiFi fw patching
*From the pc of James Savidge and Shannon Pollard*

claud-wait, 2023-02-26:
> You will need a network adapter that can handle packet injection, and monitor mode first, then yes, should be able to. YouTube is a valuable resource, I find using a good walkthrough at the same [...]

right but i search for 3 day and i didnt find any solution i successfully enable monitor mode by patching kernel but no frame injection so i ask community for help if i can port the fw both (my phone/ rpie ) use same radio chipset but version are near thanks for reply anyway