seemoo-lab/nexmon #567: Recommended samsung phone

tomer-avrahami, 2023-07-04:
Hi,

Which Samsung phone version is the most recommneded for packet injection project?

Thanks,
Tomer

davidrozen76, 2023-07-05:
Nexmon is supported on Samsung Galaxy S7, S10, S20 and S21 FE. I suppose S20's and S21's hardware is better than older devices which might yield better results.

tomer-avrahami, 2023-07-05:
Thanks!
Is it possible to consult with you privately?
If so, this is my mail:
tomer.avrahami@wiliot.com

savox-326, 2023-07-06:
@davidrozen76 could you tell how actions you made for launch monitor? Simple bcm wifi firmware replacing not helpful for my device (Note 20 Ultra, Snapdragon)

savox-326, 2023-12-12:
@davidrozen76 can you give me your s21fe's wifi firmware? My compiled one no has wlc_monitor_attach function name checked with strings | grep

closed 2023-12-26 by jlinktu