seemoo-lab/nexmon #572: request support for bcmdhd4389

Biohazardousrom, 2023-08-16:
Hi there 

 i would request support for the bcmdhd4389 broadcom chip and firmware. Atm we have firmware  with wifi monitor mode but would like to use nexmon on the tensor boards including pixel 6/pro, pixel 6a and pixel 7/pro. Also as it stands the app fc's after trying to grant root permission on pixel 6 and pixel 

thank you

ouroboros420, 2023-08-16:
i second this, would be awesome to have support for those phones

Biohazardousrom, 2023-08-16:
i can provide any logs or info that you need as well. 

heres a link to the firmware. every thing needed will be in the system/vendor folder of the magisk modules repo here

https://github.com/Biohazardousrom/nh-magisk-wifi-firmware-gs101-gs201/tree/master/system/vendor

and here is a link to the xda page explaining wifi sniffer and how to use it

https://forum.xda-developers.com/t/get-bcm4389-into-monitor-mode-for-wifi-sniffing.4525011/

jlinktu, 2023-08-16:
I already started adding `bcm4389c1` firmware version `20.101.36.2` taken from a pixel 7 pro locally, I can push this together with a `rom_extraction` patch during the next days, might also add the firmware you were linking to (version `20.101.43`), it usually is quickly done to add a new version as long as there is already a reference
however, I haven't finished frame injection and monitor mode yet, this will have to wait a bit

Biohazardousrom, 2023-08-16:
thank you for getting back to me. 

just to let you know that the fw_bcmdhd_monitor.bin firmware already has monitor mode built in and through radiotap0, but the caveat is that as of right now we have to use the wifi_sniffer binary to bring up radiotap0. also i pulled the firmware from a precompiled  vendor image used to build android 14 beta. from what ive found it should work with any device that uses the 4389 bcm chipset

jlinktu, 2023-08-16:
thanks, I am aware of the monitor and manufacturer builds that come in addition to the default firmware
in the end, `radiotap0` and `wlan0` are just separate interfaces of different types to the same SoC

savox-326, 2023-08-18:
@jlinktu i know that sounds weird, but does it possible to use monitor mode on UWB SoC? Most samsung devices uses SR100. And, so... when will the patch for bcm4375 for 18.41.113 be ready

rdtm443, 2023-09-15:
Not in a position to drop a request, but would really love to see nexmon coming to bcmdhd4389 :)

savox-326, 2023-09-18:
@jlinktu what's about bcm4375? Will patch for new firmware coming?

Kavanyy, 2025-06-04:
Are there any news on this? I want to add monitor mode support for Pixel 9 devices and would love to have a reference.
Thanks in advance for your reply!

> I already started adding `bcm4389c1` firmware version `20.101.36.2` taken from a pixel 7 pro locally, I can push this together with a `rom_extraction` patch during the next days, might also add the [...]

Kavanyy, 2025-07-01:
Dear @jlinktu 

By now im having some trouble trying to implement packet injection, i already reversed some structures needed for It and tried following the injection template for the bcm4375.

Currently im not able to send packages nor even dump debug messages on the console when triggering the NEX_INJECT_FRAME ioctl through nexutil using the corresponding ioctl implemented on ioctl.c.

Did you manage to develop It on the bcm4389 or maybe on the bcm4398? So i can keep investigating as right now im stuck.

Or at least probably point me in the right direction? I would be more than grateful for your guidance.