seemoo-lab/nexmon #575: RPI Zero 2W | patches/bcm43436b0/9_88_4_65/nexmon doesn't find libisl.so.10, even though is is there

linuzzx, 2023-10-12:
``I followed every step until

> Go to the patches folder for the bcm43430a1/bcm43455c0/bcm43436b0 chipset: cd patches/bcm43430a1/7_45_41_46/nexmon/ / patches/bcm43455c0/<7_45_154 or 7_45_189>/nexmon/ / cd patches/bcm43436b0/9_88_4_65/nexmon/
> 
> Compile a patched firmware: make

when I make, I get the error 

```
/home/linus/nexmon/buildtools/gcc-arm-none-eabi-5_4-2016q2-linux-armv7l/bin/../lib/gcc/arm-none-eabi/5.4.1/cc1: error while loading shared libraries: libisl.so.10: cannot open shared object file: No such file or directory
make: *** [Makefile:66: obj/injection.o] Error 1
```
Even though `ls /usr/lib/arm-linux-gnueabihf/` shows that I successfully compiled it and it is present.

Full log:

```
root@pizero:/home/linus/nexmon/patches/bcm43436b0/9_88_4_65/nexmon# make

          ###########   ###########   ##########    ##########
         ############  ############  ############  ############
         ##            ##            ##   ##   ##  ##        ##
         ##            ##            ##   ##   ##  ##        ##
         ###########   ####  ######  ##   ##   ##  ##    ######
          ###########  ####  #       ##   ##   ##  ##    #    #
                   ##  ##    ######  ##   ##   ##  ##    #    #
                   ##  ##    #       ##   ##   ##  ##    #    #
         ############  ##### ######  ##   ##   ##  ##### ######
         ###########    ###########  ##   ##   ##   ##########

            S E C U R E   M O B I L E   N E T W O R K I N G

                               presents:

              # ###   ###  #   # # ###  ###   ###  # ###
              ##   # #   #  # #  ##   ##   # #   # ##   #
              #    # #####   #   #    #    # #   # #    #
              #    # #      # #  #    #    # #   # #    #
              #    #  #### #   # #    #    #  ###  #    #

                The C-based Firmware Patching Framework

                           !!! WARNING !!!
    Our software may damage your hardware and may void your hardware’s
     warranty! You use our tools at your own risk and responsibility

  COLLECTING STATISTICS read /home/linus/nexmon/STATISTICS.md for more information
  COMPILING src/injection.c => obj/injection.o (details: log/compiler.log)
/home/linus/nexmon/buildtools/gcc-arm-none-eabi-5_4-2016q2-linux-armv7l/bin/../lib/gcc/arm-none-eabi/5.4.1/cc1: error while loading shared libraries: libisl.so.10: cannot open shared object file: No such file or directory
make: *** [Makefile:66: obj/injection.o] Error 1
root@pizero:/home/linus/nexmon/patches/bcm43436b0/9_88_4_65/nexmon# /usr/lib/arm-linux-gnueabihf/
bash: /usr/lib/arm-linux-gnueabihf/: Is a directory
root@pizero:/home/linus/nexmon/patches/bcm43436b0/9_88_4_65/nexmon# ls /usr/lib/arm-linux-gnueabihf/
audit                                libfuse.so.2.9.9                  libnss_mdns4.so.2
avahi                                libg.a                            libnss_mdns6_minimal.so.2
awk                                  libgcc_s.so.1                     libnss_mdns6.so.2
blas                                 libgcrypt.so.20                   libnss_mdns_minimal.so.2
bluetooth                            libgcrypt.so.20.2.8               libnss_mdns.so.2
cifs-utils                           libgdbm_compat.so.4               libnssutil3.so
crt1.o                               libgdbm_compat.so.4.0.0           libntfs-3g.so.883
crti.o                               libgdbm.so.6                      libntfs-3g.so.883.0.0
crtn.o                               libgdbm.so.6.0.0                  libogg.so.0
dhcpcd                               libgd.so.3                        libogg.so.0.8.4
e2fsprogs                            libgd.so.3.0.8                    libopcodes-2.35.2-system.so
engines-1.1                          libgfortran.so.5                  liborc-0.4.so.0
gawk                                 libgfortran.so.5.0.0              liborc-0.4.so.0.32.0
gconv                                libgio-2.0.so.0                   liborc-test-0.4.so.0
gcrt1.o                              libgio-2.0.so.0.6600.8            liborc-test-0.4.so.0.32.0
gio                                  libglib-2.0.so.0                  libp11-kit.so.0
glib-2.0                             libglib-2.0.so.0.6600.8           libp11-kit.so.0.3.0
[... 219 lines cut ...]
libFLAC.so.8                         libnpth.so.0                      libxxhash.so.0.8.0
libFLAC.so.8.3.0                     libnpth.so.0.1.2                  libyaml-0.so.2
libfl.so                             libns-9.16.44-Raspbian.so         libyaml-0.so.2.0.6
libfl.so.2                           libnsl-2.31.so                    libz.a
libfl.so.2.0.0                       libnsl.a                          libz.so
libfmt.so.7                          libnsl.so                         libz.so.1
libfmt.so.7.1.3                      libnsl.so.1                       libz.so.1.2.11
libfontconfig.so.1                   libnsl.so.2                       libzstd.so.1
libfontconfig.so.1.12.0              libnsl.so.2.0.1                   libzstd.so.1.4.8
libform.so.6                         libnspr4.so                       Mcrt1.o
libform.so.6.2                       libnss3.so                        ModemManager
libformw.so.6                        libnss_compat-2.31.so             NetworkManager
libformw.so.6.2                      libnss_compat.so                  nss
libfreetype.a                        libnss_compat.so.2                perl
libfreetype.so                       libnss_dns-2.31.so                perl5
libfreetype.so.6                     libnss_dns.so                     perl-base
libfreetype.so.6.17.4                libnss_dns.so.2                   pkgconfig
libfribidi.so.0                      libnss_files-2.31.so              rsyslog
libfribidi.so.0.4.0                  libnss_files.so                   samba
libfstrm.so.0                        libnss_files.so.2                 sasl2
libfstrm.so.0.1.0                    libnss_hesiod-2.31.so             Scrt1.o
libftdi1.so.2                        libnss_hesiod.so                  security
libftdi1.so.2.5.0                    libnss_hesiod.so.2                xtables
libfuse.so.2                         libnss_mdns4_minimal.so.2
```

icountu, 2023-12-22:
Same issue 😭

wmehilos, 2024-06-29:
I fixed this by changing the symlink commands (for both libisl and libmpfr) a tad; at least on my Pi 5 the versions of the libs the commands in the readme were trying to symlink didn't exist. I'm running Raspberry Pi OS Bookworm. 

`$ uname -a`

`Linux hostname 6.6.31+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.6.31-1+rpt1 (2024-05-29) aarch64 GNU/Linux`

From the readme, the command: 

`sudo ln -s /usr/lib/arm-linux-gnueabihf/libisl.so.23.0.0  /usr/lib/arm-linux-gnueabihf/libisl.so.10`

links `/usr/lib/arm-linux-gnueabihf/libisl.so.10` to `/usr/lib/arm-linux-gnueabihf/libisl.so.23.0.0`, which doesn't exist on my Pi 5. 

Same deal with the mpfr library in the next `ln` command. 

The fix is simple, but you'll want to double check what the latest version of the libs you have installed on your Pi. 

First, remove the old symlinks:

`sudo rm /usr/lib/arm-linux-gnueabihf/libisl.so.10`
`sudo rm /usr/lib/arm-linux-gnueabihf/libmpfr.so.4`

Then, get a list of each to see what version you have installed. Find the file with no `->` after it. In my case: 

`$ ls -l /usr/lib/arm-linux-gnueabihf/libisl*`

`lrwxrwxrwx 1 root root      16 Jan  2 04:58 /usr/lib/arm-linux-gnueabihf/libisl.so.23 -> libisl.so.23.2.0`

`-rw-r--r-- 1 root root 1199120 Jan  2 04:58 /usr/lib/arm-linux-gnueabihf/libisl.so.23.2.0`

and 

`$ ls -l /usr/lib/arm-linux-gnueabihf/libmpfr*`

`lrwxrwxrwx 1 root root     16 Jan 14  2023 /usr/lib/arm-linux-gnueabihf/libmpfr.so.6 -> libmpfr.so.6.2.0`

`-rw-r--r-- 1 root root 330676 Jan 14  2023 /usr/lib/arm-linux-gnueabihf/libmpfr.so.6.2.0`

So, in my case, I should change 

`sudo ln -s /usr/lib/arm-linux-gnueabihf/libisl.so.23.0.0  /usr/lib/arm-linux-gnueabihf/libisl.so.10`

to 

`sudo ln -s /usr/lib/arm-linux-gnueabihf/libisl.so.23.2.0  /usr/lib/arm-linux-gnueabihf/libisl.so.10`

and the same for the `libmpfr` library.

I don't know if there is some technical reason to not symlink to the other symlink (libisl.so.23) in that directory for `libisl`, which would have avoided this issue in the first place, since presumably these libraries get small point updates often but that symlink to version 23 or whatever won't as often. I tried it that way too and `make` didn't complain, but I'm not an authority on this at all. 

I continued to have unrelated issues backing up my stock firmware, but YMMV on a different Pi so I won't get into that here.

Xendr1k, 2024-10-12:
This problem occurs when you don't download "nexmon" to "cd/". I don't quite understand why but that's how it happens.