seemoo-lab/nexmon #577: what do jnmah and jmah mean?

cruisezhou, 2023-10-15:
Hi, I am learning the fabulous project currently. For the ucode diassembler, I found a slight difference from b43/disassembler/ and  b43-v3/disassembler/. In b43-v3/disassembler, there are some instructions which havn't been mentioned before. They are:
jmah(0xB0) jnmah(0xB1) jboh2(0x80) jnboh2(0x81)  and two commented instructions --merd(0x102) and cazz(0x103).
Does anyone have some infomation about how there instructions oprate? 
Thanks.

jlinktu, 2023-10-17:
Hi, there are italian words involved... e.g. "boh" just means "I don't know".
Those are instructions that are not completely understood or no matching name was found.
If you are into understanding the D11, you might be interested in our recently published [D11 emulator](https://github.com/seemoo-lab/d11-emu). It is WIP but might help you better understand the instructions.

cruisezhou, 2023-10-18:
> Hi, there are italian words involved... e.g. "boh" just means "I don't know". Those are instructions that are not completely understood or no matching name was found. If you are into understanding [...]

Hi, Thanks for your reply! I would like to study the D11 emulator.