seemoo-lab/nexmon #578: ucode.bin missing during Make process for firmware image bcm43439a0

redheron5, 2023-10-20:
Hello,
I am trying to follow your build process on a Raspberry Pi 4 Model B with linux kernel version 15.15.  When attempting to make at the top folder level, I get an error when extracting the ucode for firmware image 43439a0.  Looking in the subdirectories, it appears that the ucode.bin file is indeed missing.

Below is the error I'm getting: 

  EXTRACTING UCODE
ERR: ram file empty or unavailable.
make: *** [Makefile:17: ucode.bin] Error 1

redheron5, 2023-10-20:
It looks like the ucode.bin file was maybe supposed to be generated in the patches make process?  Trying to run make in that folder I got this error:
COLLECTING STATISTICS read /home/emily.shields/nexmon/STATISTICS.md for more information
  COMPILING src/ioctl.c => obj/ioctl.o (details: log/compiler.log)
/home/emily.shields/nexmon/buildtools/gcc-arm-none-eabi-5_4-2016q2-linux-armv7l/bin/../lib/gcc/arm-none-eabi/5.4.1/cc1: error while loading shared libraries: libisl.so.10: wrong ELF class: ELFCLASS64

Does this mean I improperly installed some 32 bit libraries on my 64 bit Raspberian?

redheron5, 2023-10-20:
Come to think of it, when I had to build /usr/lib/arm-linux-gnueabihf/libisl.so.10 from source, ./configure failed because I needed to specify an architecture, and I wasn't sure which to specify.

jlinktu, 2023-10-20:
Those are probably two unrelated issues.

Regarding `libisl`, there is a dedicated section for 64 bit raspberry pi os [here](https://github.com/seemoo-lab/nexmon#on-64bit-raspberry-pi-os).

The failed ucode extraction seems to be a different issue, I'll have a look soon. Until then, you can just remove the failing target from the Makefile, as you do not need it for the RPi4.

redheron5, 2023-10-20:
I'm a dummy - you said use kernel 5.4, not 15 -> let me try that and see if it works.

WildWestCyberSecurity, 2024-05-17:
I ran into similar issue, using latest raspios-bookworm-arm64-lite image.

To fix libisl.so issue make sure before you create symbolic link outlined in official readme [here](https://github.com/seemoo-lab/nexmon#on-64bit-raspberry-pi-os).

Instructions 4-5:
```sudo ln -s /usr/lib/arm-linux-gnueabihf/libisl.so.23.0.0  /usr/lib/arm-linux-gnueabihf/libisl.so.10
sudo ln -s /usr/lib/arm-linux-gnueabihf/libmpfr.so.6.1.0 /usr/lib/arm-linux-gnueabihf/libmpfr.so.4```

DO MAKE SURE libisl.so.23.0.0 does indeed exist as in my case it was named libisl.so.23 and same for libmpfr.so.6.1.0 and this one for me was simply libmpfr.so.6

Updated instructions 4-5 that got me *to almost * be able to compile patch for bcm43455c0/7_45_234_4ca95bb_CY.

```  ln -s /usr/lib/arm-linux-gnueabihf/libisl.so.23  /usr/lib/arm-linux-gnueabihf/libisl.so.10
  ln -s /usr/lib/arm-linux-gnueabihf/libmpfr.so.6 /usr/lib/arm-linux-gnueabihf/libmpfr.so.4
``` 

I am however running into this issue when i attempt make file(running make in nexmon root):
```
EXTRACTING UCODE
ERR: ram file empty or unavailable.
make[3]: *** [Makefile:17: ucode.bin] Error 1
make[3]: Leaving directory '/home/pi/nexmon/firmwares/bcm43439a0/7_95_49_2271bb6'
make[2]: *** [Makefile:8: 7_95_49_2271bb6/] Error 2
make[2]: Leaving directory '/home/pi/nexmon/firmwares/bcm43439a0'
make[1]: *** [Makefile:7: bcm43439a0/] Error 2
make[1]: Leaving directory '/home/pi/nexmon/firmwares'
make: *** [Makefile:5: firmwares] Error 2
```

running make on patches/bcm43455c0/7_45_234_4ca95bb_CY/nexmon dir:
```
  COLLECTING STATISTICS read /home/pi/nexmon/STATISTICS.md for more information
  COMPILING src/console.c => obj/console.o (details: log/compiler.log)
  COMPILING src/ioctl.c => obj/ioctl.o (details: log/compiler.log)
  COMPILING src/patch.c => obj/patch.o (details: log/compiler.log)
  COMPILING src/version.c => obj/version.o (details: log/compiler.log)
make: *** No rule to make target '/home/pi/nexmon/firmwares/bcm43455c0/7_45_234_4ca95bb_CY/ucode.bin', needed by 'gen/ucode.bin'.  Stop.
```

I will update if I find anything else out :P hope this helps anyone!

WildWestCyberSecurity, 2024-05-17:
> I ran into similar issue, using latest raspios-bookworm-arm64-lite image.
> 
> To fix libisl.so issue make sure before you create symbolic link outlined in official readme [here](https://github.com/seemoo-lab/nexmon#on-64bit-raspberry-pi-os).
> 
> Instructions 4-5: `sudo ln -s /usr/lib/arm-linux-gnueabihf/libisl.so.23.0.0 /usr/lib/arm-linux-gnueabihf/libisl.so.10` `sudo ln -s /usr/lib/arm-linux-gnueabihf/libmpfr.so.6.1.0 /usr/lib/arm-linux-gnueabihf/libmpfr.so.4`
> 
> DO MAKE SURE libisl.so.23.0.0 does indeed exist as in my case it was named libisl.so.23 and same for libmpfr.so.6.1.0 and this one for me was simply libmpfr.so.6
> 
> Updated instructions 4-5 that got me *to almost * be able to compile patch for bcm43455c0/7_45_234_4ca95bb_CY. _running as root instead sudo (sudo su)_ `ln -s /usr/lib/arm-linux-gnueabihf/libisl.so.23 /usr/lib/arm-linux-gnueabihf/libisl.so.10` `ln -s /usr/lib/arm-linux-gnueabihf/libmpfr.so.6 /usr/lib/arm-linux-gnueabihf/libmpfr.so.4`
> 
> I am however running into this issue when i attempt make file(running make in nexmon root): `EXTRACTING UCODE ERR: ram file empty or unavailable. make[3]: *** [Makefile:17: ucode.bin] Error 1 make[3]: Leaving directory '/home/pi/nexmon/firmwares/bcm43439a0/7_95_49_2271bb6' make[2]: *** [Makefile:8: 7_95_49_2271bb6/] Error 2 make[2]: Leaving directory '/home/pi/nexmon/firmwares/bcm43439a0' make[1]: *** [Makefile:7: bcm43439a0/] Error 2 make[1]: Leaving directory '/home/pi/nexmon/firmwares' make: *** [Makefile:5: firmwares] Error 2`
> 
> running make on patches/bcm43455c0/7_45_234_4ca95bb_CY/nexmon dir: `COLLECTING STATISTICS read /home/pi/nexmon/STATISTICS.md for more information COMPILING src/console.c => obj/console.o (details: log/compiler.log) COMPILING src/ioctl.c => obj/ioctl.o (details: log/compiler.log) COMPILING src/patch.c => obj/patch.o (details: log/compiler.log) COMPILING src/version.c => obj/version.o (details: log/compiler.log) make: *** No rule to make target '/home/pi/nexmon/firmwares/bcm43455c0/7_45_234_4ca95bb_CY/ucode.bin', needed by 'gen/ucode.bin'. Stop.`
> 
> I will update if I find anything else out :P hope this helps anyone!

I was successfully able to get this to work/compile, figured out ucode issue I was running into on my system.

For whatever reason bcm43439a0 was giving compile errors during ucode process, so I simply removed this dir using `rm -rf nexmon/firmwares/bcm43439a0`

I then was able to run make in nexmon root dir, as well as patch for bcm43455c0/7_45_234_4ca95bb_CY.

Here is results success!
`snip...  GENERATING LINKER FILE gen/nexmon.pre => gen/nexmon.ld
  GENERATING LINKER FILE gen/nexmon.pre => gen/flashpatches.ld
  GENERATING LINKER FILE gen/memory.ld
  LINKING OBJECTS => gen/patch.elf (details: log/linker.log, log/linker.err)
  GENERATING MAKE FILE gen/nexmon.pre => gen/nexmon.mk
  GENERATING MAKE FILE gen/nexmon.pre => gen/flashpatches.mk
  APPLYING FLASHPATCHES gen/flashpatches.mk => cyfmac43455-sdio-standard.bin (details: log/flashpatches.log)
  APPLYING PATCHES gen/nexmon.mk => cyfmac43455-sdio-standard.bin (details: log/patches.log)
  BUILDING DRIVER for kernel 6.6 brcmfmac_6.6.y-nexmon/brcmfmac.ko (details: log/driver.log
`

Now we have super awesome firmware :sunglasses: , (it compiles but will it run lol I will update again if not working)

p.s Anyone running *LITE* raspos  version should also run `apt install xxd` as it was required during build process but missing as i guess lite version.

jlinktu, 2024-05-17:
Wasn't expecting `xxd` to not being installed by default. It probably makes sense to use something else then, will change this at some point..