seemoo-lab/nexmon #581: Husarion OS for ROSBot not able to install patch

BrendanxP, 2023-10-27:
I am trying to get Nexmon working directly on the hardware that drives the ROSBot 2R, which is a Raspberry Pi 4B. They have [pre-built images for the robots](https://husarion.com/software/os/installation/), called Husarion OS, with multiple versions based on different Linux kernels. I have tried getting the toolbox to work on two of the versions, but both fail to install the patch and end up in an error during `make install firmware`. 
I followed the guide precisely as stated, though, I had to add sources to install the raspberrypi-headers. I did get it installed on Raspberry Pi OS with closely similar Linux kernels. Therefore, I am confused about what adjustments in these images from Husarion could cause the toolbox to not install at all.

1. Version: ROS2 Humble on top of Ubuntu 22
[Direct download link](https://robot-os-images.s3.eu-central-1.amazonaws.com/husarion-rpi-ros2-humble-2023-08-31.img.xz)
Ubuntu 22.04.03 LTS (Jammy Jellyfish)
Kernel: 5.15.0-1041-raspi
Architecture: Aarch64

The error it gives:
`COLLECTING STATISTICS read /home/husarion/nexmon/STATISTICS.md for more information
  COMPILING src/version.c => obj/version.o (details: log/compiler.log)
  GENERATING LINKER FILE gen/nexmon.pre => gen/nexmon.ld
  GENERATING LINKER FILE gen/nexmon.pre => gen/flashpatches.ld
  LINKING OBJECTS => gen/patch.elf (details: log/linker.log, log/linker.err)
  GENERATING MAKE FILE gen/nexmon.pre => gen/nexmon.mk
  GENERATING MAKE FILE gen/nexmon.pre => gen/flashpatches.mk
  APPLYING FLASHPATCHES gen/flashpatches.mk => brcmfmac43455-sdio.bin (details: log/flashpatches.log)
  APPLYING PATCHES gen/nexmon.mk => brcmfmac43455-sdio.bin (details: log/patches.log)
  BUILDING DRIVER for kernel 5.15 brcmfmac_5.15.y-nexmon/brcmfmac.ko (details: log/driver.log)
  COPYING brcmfmac43455-sdio.bin => /lib/firmware/brcm/brcmfmac43455-sdio.bin
  RELOADING brcmfmac
insmod: ERROR: could not insert module /home/husarion/nexmon/patches/driver/brcmfmac_5.15.y-nexmon/brcmfmac.ko: Unknown symbol in module
make: *** [Makefile:192: install-firmware] Error 1`

2. Version ROS Noetic on top of Ubuntu 20
[Direct download link](https://robot-os-images.s3.eu-central-1.amazonaws.com/husarion-rpi-ros-noetic-2023-08-31.img.xz)
Ubuntu: 20.04.6 LTS (Focal Fossa)
Kernel: 5.4.0-1052-raspi

The error it gives:
`COLLECTING STATISTICS read /home/husarion/nexmon/STATISTICS.md for more information
  COMPILING src/ioctl.c => obj/ioctl.o (details: log/compiler.log)
/bin/sh: 1: /home/husarion/nexmon/buildtools/gcc-arm-none-eabi-5_4-2016q2-linux-armv7l/bin/arm-none-eabi-gcc: not found
make: *** [Makefile:123: obj/ioctl.o] Error 127`
Even though the file does exist and has perfect ownership and r/w/x rights.

Any help is appreciated. I only need one version to run, so either the ROS2 Humble or ROS Noetic is fine, whichever is easier to fix.

jlinktu, 2023-10-27:
The `brcmfmac` with its modifications, the Makefiles, and the firmware patches are made to work with Raspberry Pi OS ([README](https://github.com/seemoo-lab/nexmon#build-patches-for-bcm43430a1-on-the-rpi3zero-w-or-bcm434355c0-on-the-rpi3rpi4-or-bcm43436b0-on-the-rpi-zero-2w-using-raspbianraspberry-pi-os-recommended)). If you want to adapt it to work with Husarion OS, e.g. ROS 2 Humble, get the corresponding kernel sources (as it is based on Ubuntu linux-raspi you might want to look [here](https://code.launchpad.net/~ubuntu-kernel/ubuntu/+source/linux-raspi/+git)) and adapt `brcmfmac` s.t. you can use nexmon firmwares with it. You might also need to use a different firmware..in the ROS 2 Humble image you linked, the firmware is located here: `/usr/lib/firmware/brcm/brcmfmac43455-sdio.bin`. Of course you also have to adapt the Makefiles to work under Husarion OS instead of Raspberry Pi OS.

BrendanxP, 2023-10-30:
> The `brcmfmac` with its modifications, the Makefiles, and the firmware patches are made to work with Raspberry Pi OS [...]

Thank you @jlinktu, this is very helpful. I will have a look if I can get it to work like you described!