seemoo-lab/nexmon #585: RPI Zero 2W | Not all versions supported?

dolfje, 2023-11-16:
I tried to compile nexmon for Raspberry Zero 2W, but it didn't work. After searching I noticed that it loaded the brcmfmac43436**s**-sdio version instead of the compiled brcmfmac43436-sdio. So I searched the internet and learned there are at least 2 versions of the wifi chip used:
* BCM43430/2 "brcm/brcmfmac43436-sdio"
* BCM43430/1 "brcm/brcmfmac43436**s**-sdio"

Nexmon only has support for brcmfmac43436-sdio
https://github.com/seemoo-lab/nexmon/tree/master/firmwares/bcm43436b0/9_88_4_65

I tried to copy the brcmfmac43436-sdio to brcmfmac43436**s**-sdio, but that didn't work.
This guy also had the same problem https://github.com/seemoo-lab/nexmon/issues/564#issuecomment-1642353360

So does anybody has a clue for the next steps?

dolfje, 2023-11-17:
So compiling and patching the brcmfmac43430-sdio in this branch and copying to brcmfmac43436s-sdio works

closed 2023-11-17 by dolfje

jlinktu, 2023-11-17:
Hi @dolfje,
I assume the original firmware blob that gets loaded on your device is this one: [brcmfmac43436s-sdio.bin](https://github.com/RPi-Distro/firmware-nonfree/blob/995129a86915fa3165a24d0a29419f6349f8f36b/debian/config/brcm80211/brcm/brcmfmac43436s-sdio.bin).
According to the build-string that it contains, it is build against the `bcm43430a1` ROM:
```
43430a1-roml/sdio-... Version: 7.45.96.s1 (gf031a129) CRC: 6670a1e Date: Wed 2023-06-14 07:28:04 CST Ucode Ver: 1043.20744 FWID: 01-70bd2af7
```
I think the labeling of the firmware file as `43436` is simply wrong / misleading. It should be named `43430`.

This also explains why using a patched version of a `brcmfmac43430-sdio` firmware works for you.

As a side note:
The [brcmfmac43436-sdio.bin](https://github.com/RPi-Distro/firmware-nonfree/blob/995129a86915fa3165a24d0a29419f6349f8f36b/debian/config/brcm80211/brcm/brcmfmac43436-sdio.bin) (_note the missing **s**_) that gets loaded on the other versions of the RPi Zero 2 W is build against the `43436b0` ROM. Hence, in this case the naming makes sense.