seemoo-lab/nexmon #586: The Nexus5 phone cannot connect to the network

doge0823, 2023-11-18:
I am a graduate student from China. I just installed nexmon on the Nexus5 mobile phone whose network card is bcm4339 to collect CSI data. However, after setting wlan0 as Monitor mode, the mobile phone could not scan the nearby WiFi and the SIM card could not connect to the network. Is there any way to make the Nexus5 mobile card connect to the network normally?

jlinktu, 2023-11-18:
Monitor interfaces are not ment to be connected to an access point.

The normal Android WiFi operations are not compatible with running Nexmon CSI patched firmware.

The Nexmon CSI patch only affects the WiFi firmware and does not touch the mobile network at all. I don't think your mobile network issue is directly related to nexmon.

doge0823, 2023-11-18:
> Monitor interfaces are not ment to be connected to an access point.
>
> The normal Android WiFi operations are not compatible with running Nexmon CSI patched firmware.
>
> The Nexmon CSI patch only [...]

Thank you for your reply. After receiving the reply, I changed the SIM card of different operators as soon as possible, and found that the SIM card of "China Mobile" operator can connect to the network, which I think is related to the compatibility between operators and mobile phones.
About the Nexmon CSI patch, I have another question. Is the modification of the bcm4339 network card irreversible? Can it be restored only by brushing the mobile phone system again? (I tried to restore the normal WiFi function only by "Resume factory setting")

jlinktu, 2023-11-18:
The firmware blob is a binary file residing in the file system, which gets loaded into the memory of the chip by the driver when the chip is fired up. You can go back to default by putting the original firmware file: https://github.com/seemoo-lab/nexmon/blob/master/firmwares/bcm4339/6_37_34_43/fw_bcmdhd.bin there instead of the patched one.

doge0823, 2023-11-19:
> The firmware blob is a binary file residing in the file system of the host, which gets loaded into the memory of the chip by the driver when the chip is fired up. You can go back to default by putting the original firmware file: https://github.com/seemoo-lab/nexmon/blob/master/firmwares/bcm4339/6_37_34_43/fw_bcmdhd.bin there instead of the patched one.

I am new to the research field, and forgive me for asking this question, which may seem immature to you, but after I replaced the file "/sdcard/fw_bcmdhd.bin" on Nexus5 with "fw_bcmdhd.bin" downloaded on Github, my phone still couldn't connect to the network. Are there any additional commands or actions that need to be executed to make this work?

jlinktu, 2023-11-19:
`/sdcard/fw_bcmdhd.bin` is only a temporary location, you want to place it at `/vendor/firmware/fw_bcmdhd.bin`. Just like the `make install-firmware` does it with the patched firmware:
https://github.com/seemoo-lab/nexmon/blob/e4e99ebcc8baa139928dc35bf17d2e8d02a76219/patches/bcm4339/6_37_34_43/nexmon/Makefile#L157

closed 2023-12-28 by jlinktu