seemoo-lab/nexmon #587: Is this an InfoDisclosure vulnerability??

danielcunn123, 2023-12-05:
CWE-214: Invocation of Process Using Visible Sensitive Information
CWE-497: Exposure of Sensitive System Information to an Unauthorized Control Sphere
CWE-548: Exposure of Information Through Directory Listing

Through static code analysis it's possible to determine the source code directory used for driver compilation, otherwise known as the Nexmon PATH. The driver directly references source code locations whereas common drivers implement relative path references based on binary location, such as the linux bluetooth.ko driver.

Is it even possible to reference 'warn_slowpath_fmt'  to a relative location?

Unsure if this is present within nexmon patched android firmware.

An attacker with local access may gain insight to compiler and source code locations on the system, also determine modified wireless firmware with low privilages.

[image]
[image]
[image]
[image]

jlinktu, 2023-12-05:
This seems to be specific to the `brcmfmac` building within nexmon. Firmware patches should not be affected by this.

danielcunn123, 2023-12-06:
> This seems to be specific to the `brcmfmac` building within nexmon. Firmware patches should not be affected by this.

Yes the kernel driver to be more specific.

Since Nexmon is technically release software its best pratice to use relative paths when compiling drivers, so patching 'NEXMON_ROOT' should resolve the path issue.

A similar issue was rust incorrectly remapping 'rust-src' and defaulting to fullpath for rust applications, disclosing many developer usernames within many binaries...