seemoo-lab/nexmon #590: error run "source setup_env.sh" in the repository\'s root directory

Empire-digital-solutions, 2024-01-02:
I am trying to installing nexmon on a raspberry pi zero 2w.
when I try to run make command in "nexmon/patches/bcm43436b0/9_88_4_65/nexmon" I get this error;
Makefile:176: *** run 'source setup_env.sh' in the repository\'s root directory. Stop.
I did as the error said by running source setup_env.sh but it still throws this error.
          Any help or guidance would be appreciated thank you in advance

jlinktu, 2024-01-02:
Did you run `source setup_env.sh` in the **nexmon root directory** as told by the message or in `nexmon/patches/bcm43436b0/9_88_4_65/nexmon`?

Empire-digital-solutions, 2024-01-02:
I ran it in the root directory

jlinktu, 2024-01-02:
This error message is triggered if the environment variable `NEXMON_SETUP_ENV` is not set. It is set to your current environment when sourcing `setup_env.sh`.
Did you switch to a different shell maybe? You have to continue in the same shell environment in that you run `source setup_env.sh` or add it to one of the auto-setup files if you want the environment to be setup for every shell instance by default.

Empire-digital-solutions, 2024-01-02:
Am using the same shell so I assume that having the it in the auto-setup files wouldn't fix the problem

jlinktu, 2024-01-02:
Enter the nexmon root directory and run the following commands:
```
source setup_env.sh
env | grep NEXMON_SETUP_ENV
```
This should produce:
```
NEXMON_SETUP_ENV=1
```
Is this the case?

Empire-digital-solutions, 2024-01-03:
yes that is the output

jlinktu, 2024-01-03:
And the original error message persists even with making sure that `NEXMON_SETUP_ENV` is set?

Empire-digital-solutions, 2024-01-03:
Yes that is right

jlinktu, 2024-01-03:
That doesn't really make sense to me.

You followed the README?

Can you give me more details about your OS, version, steps you performed? Anything that helps me to reproduce this behavior..

Empire-digital-solutions, 2024-01-04:
Sure, I am using raspberry pi OS (legacy 64bit lite) version 6.1.21-v8+, and i did follow the README exactly except for one thing I commented out line #4 In the Makefile: `include $(FW_Path)/definitions.mk` because it was giving me this error that i could not solve: `Makefile:4: /firmwares/bcm43436b0/9_88_4_65/definitions.mk: No such file or directory
make: *** No rule to make target '/firmwares/bcm43436b0/9_88_4_65/definitions.mk'. Stop`

jlinktu, 2024-01-04:
`definitions.mk` is required, the build process won't work without it, as it holds crucial information about the firmware.

Both errors are related to the same issue: From the `Makefile`'s perspective, the environment set from `source setup_env.sh` is not seen.

Can you please paste a copy of your full procedure (including commands and outputs) of what you are doing? (redact personal data)

Empire-digital-solutions, 2024-01-04:
1. `sudo su`
2. `apt-get` update && apt-get upgrade`
[image]
3.`exit`
4.`sudo apt install raspberrypi-kernel-headers git libgmp3-dev gawk qpdf bison flex make autoconf libtool texinfo`
   this output is from running it the second time because I did not think it would take more than one page(i can't scroll)
[image]
5.`git clone https://github.com/seemoo-lab/nexmon.git`
[image]
6.`cd nexmon`
7.`sudo dpkg --add-architecture armhf`
   no output
8.`sudo apt-get update`
9.`sudo apt-get install libc6:armhf libisl23:armhf libmpfr6:armhf libmpc3:armhf libstdc++6:armhf`
[image]
10.`sudo ln -s /usr/lib/arm-linux-gnueabihf/libisl.so.23.0.0  /usr/lib/arm-linux-gnueabihf/libisl.so.10`
11.`sudo ln -s /usr/lib/arm-linux-gnueabihf/libmpfr.so.6.1.0 /usr/lib/arm-linux-gnueabihf/libmpfr.so.4`
12.`cd patches/bcm43436b0/9_88_4_65/nexmon/`
13.`make`
[image]

So I commented out line 4 of the make file then I get the error I mentioned when I opened this thread

jlinktu, 2024-01-05:
After step 11 and before step 12 you are supposed to run `source setup_env.sh`.

Empire-digital-solutions, 2024-01-05:
Sorry, I did but I forgot to write it

jlinktu, 2024-01-05:
Are you sure? This would perfectly explain the behavior.
Just to rule out that we are talking past each other, you do:
1. open a command line
1. change into nexmon repo `cd nexmon`
1. setup the environment `source setup_env.sh` (any output?)
1. change into patches dir `cd patches/bcm43436b0/9_88_4_65/nexmon/`
1. run `make`

Empire-digital-solutions, 2024-01-05:
I erased the SD card and flashed a new image and started the process over and now I have this error
[image]

jlinktu, 2024-01-06:
There should be the following symbolic link:
```
/usr/lib/arm-linux-gnueabihf/libmpc.so.3
```
pointing so something like the following (version might be a different one):
```
/usr/lib/arm-linux-gnueabihf/libmpc.so.3.3.1
```
`libmpc.so.3` should be installed in [this step](https://github.com/seemoo-lab/nexmon?tab=readme-ov-file#on-64bit-raspberry-pi-os) of the README.
If you did that and it still doesn't work you might want to reinstall it, e.g. like this: `sudo apt-get --reinstall install libmpc3:armhf`.
However, if it was already installed but is broken, there must be a reason for that. Maybe your sdcard is broken or your installed image is corrupted...

If you run into more problems like that, please make sure to follow the README carefully and/or consider replacing your sdcard and starting from scratch.

Empire-digital-solutions, 2024-01-07:
little busy right now but I'll try that for sure and tell you what happens

Empire-digital-solutions, 2024-01-09:
I did end up solving the problem, it turns out that I was supposed to run all the commands as root including `source setup_env.sh`.
I found the solution in issue #257

closed 2024-01-09 by Empire-digital-solutions