seemoo-lab/nexmon #594: Raspberry Pi 4 firmware version not in patches

jayofelony, 2024-01-24:
```
[    8.764237] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[    8.795849] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4345/6 wl0: Apr 15 2021 03:03:20 version 7.45.234 (4ca95bb CY) FWID 01-996384e2
```
The latest nexmon firmware is 7.45.206, Raspberry Pi 5 uses 7.45.241 I think. Do you need me anything to do with this version, if so how?

jlinktu, 2024-01-25:
Raspberry Pi recently switched its firmware for the `cyw43455c0` in their [`firmware-nonfree` package](https://github.com/RPi-Distro/firmware-nonfree) from version `7.45.241` to version `7.45.234` because Infineon recommended them to do so if they want WPA3-SAE support, see [pull request#42](https://github.com/RPi-Distro/firmware-nonfree/pull/42).
Both firmwares (versions `7.45.241` and `7.45.234`) are very similar to the versions supported by nexmon, porting should be quite easy by comparing them, for example with `bindiff`.

Btw. if you want to use nexmon on a **Raspberry Pi 5** the way it is described in the README, you currenty have to boot with the 4K aligned kernel instead of the 16K aligned default one, because the `armhf` libraries are not yet compatible with it. You can do that by placing the following in your `/boot/config.txt` file:
```
kernel=kernel8.img
```

jayofelony, 2024-01-25:
I have nexmon working on the Raspberry Pi 5 by using @DrSchottky his repo. I used `QEMU_UNAME = "6.1.0-rpi7-rpi-2712"` while building an image with packer/ansible.

About the firmware versions, I have no idea how to do that. Do you want me to try anything, or not really required?

jlinktu, 2024-01-25:
If you have other tasks your time might be better invested there.
Of course, I don't want to encourage you. If you are really interested in this, just go ahead.
Though, porting the state of the currently supported firmwares to these two newer versions is trivial work from our perspective and can be done quickly, if required.

jayofelony, 2024-01-25:
I do find this very interesting, I can always try stuff and learn while doing it. Do you have a guide perhaps to help do stuff, do a comparison check etc.

jayofelony, 2024-01-25:
If there were an upgraded nexmon firmware from .241 would there be much difference/benefit from it?

jlinktu, 2024-01-25:
Newer releases usually contain bug and security related fixes - sometimes, releases also differ in their feature set. The features can be obtained for most firmware blobs from a release string that can be extracted like this:
```
# firmware version 7.45.241
$ strings cyfmac43455-sdio-standard.bin | tail -n 2 | head -n 1
43455c0-roml/43455_sdio-pno-aoe-pktfilter-pktctx-lpc-pwropt-43455_ftrs-wfds-mfp-dfsradar-wowlpf-idsup-idauth-noclminc-clm_min-obss-obssdump-swdiv Version: 7.45.241 (1a2f2fa CY) CRC: 959ad1c7 Date: Mon 2021-11-01 00:40:29 PDT Ucode Ver: 1043.2164 FWID 01-703fd60

# firmware version 7.45.234
$ strings cyfmac43455-sdio-standard.bin | tail -n 2 | head -n 1
43455c0-roml/43455_sdio-pno-aoe-pktfilter-pktctx-wfds-mfp-dfsradar-wowlpf-idsup-idauth-noclminc-clm_min-obss-obssdump-swdiv-gtkoe-roamprof-txbf-ve-sae-dpp-sr-okc-bpd Version: 7.45.234 (4ca95bb CY) CRC: 212e223d Date: Thu 2021-04-15 03:06:00 PDT Ucode Ver: 1043.2161 FWID 01-996384e2
```
The features that are embedded into these two firmware images are slightly different. Thus, if one needs a specific feature that can be found in a specific firmware image, it makes sense to use this one - which could also be a reason to add a specific firmware to nexmon.

In general, from the nexmon perspective, it makes little difference, as our additional features like monitor mode or frame injection most of the time don't depend on bug/security related fixes. But of course, we wan't to support security enhancements, which is a pro adding new releases. Another reason to add new releases can be compatibility issues between firmware and driver releases (e.g. older firmwares not working well together with newer drivers).

Specific to version `7.45.241`: I already added the firmware to nexmon (see [firmwares/bcm43455c0/7_45_241](https://github.com/seemoo-lab/nexmon/tree/master/firmwares/bcm43455c0/7_45_241)), there is just no port of the nexmon patch. But this could be easily done by copying it from version `7.45.206` [here](https://github.com/seemoo-lab/nexmon/tree/master/patches/bcm43455c0/7_45_206/nexmon) and adapting the patch addresses and version indicators inside + adapting function addresses if required in [`wrapper.c`](https://github.com/seemoo-lab/nexmon/blob/master/patches/common/wrapper.c).

jayofelony, 2024-01-25:
Okay, I will clone the repo and see how I do with porting it to version 7.45.241.

jayofelony, 2024-01-25:
Okay, so I cloned the repo. Copied the patch of 7.45.206 to 7.45.241. Found all these lines in the patch files and edited the version indicators.
[image]
But the other stuff you said about patch addresses, is just mumble jumble to me. As much as I like to be of any help here, I think this is a little over my head.

I have never done anything in C, I think I will just stick to Python for now.

jlinktu, 2024-01-26:
Alright.
Is there anything else or can this issue be closed?

closed 2024-01-26 by jayofelony