seemoo-lab/nexmon #595: firmware halted or crashed

jayofelony, 2024-02-01:
Output from `journalctl`:
`Feb 01 10:33:10 jayofelony kernel: ieee80211 phy0: brcmf_fw_crashed: Firmware has halted or crashed`

Device:
`Raspberry Pi 4`

WiFi chip:
`43455c0`

Firmware:
`nexmon 7.45.206`

```
[88411.161703] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4345/6 wl0: Jan 24 2024 20:30:46 version 7.45.206 (nexmon.org: 2.2.2-514-gf195c-2) FWID 01-88ee44ea
[88411.280828] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled
[88411.811377] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled
[88424.201595] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save disabled
[88424.249634] brcmfmac: brcmf_vif_add_validate: Attempt to add a MONITOR interface...
[88424.249665] brcmfmac: brcmf_mon_add_vif: brcmf_mon_add_vif called
[88424.249674] brcmfmac: brcmf_mon_add_vif: Adding vif "wlan0mon"
[90611.957556] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save disabled
[90614.985828] brcmfmac: brcmf_vif_add_validate: Attempt to add a MONITOR interface...
[90614.985846] brcmfmac: brcmf_vif_add_validate: ... there is already a monitor interface, returning EOPNOTSUPP
[90614.985853] ieee80211 phy0: brcmf_cfg80211_add_iface: iface validation failed: err=-95
[91535.369455] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save disabled
[91538.397427] brcmfmac: brcmf_vif_add_validate: Attempt to add a MONITOR interface...
[91538.397445] brcmfmac: brcmf_vif_add_validate: ... there is already a monitor interface, returning EOPNOTSUPP
[91538.397452] ieee80211 phy0: brcmf_cfg80211_add_iface: iface validation failed: err=-95
[93495.332828] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save disabled
[93498.360899] brcmfmac: brcmf_vif_add_validate: Attempt to add a MONITOR interface...
[93498.360916] brcmfmac: brcmf_vif_add_validate: ... there is already a monitor interface, returning EOPNOTSUPP
[93498.360923] ieee80211 phy0: brcmf_cfg80211_add_iface: iface validation failed: err=-95
[94005.518078] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save disabled
[94008.545083] brcmfmac: brcmf_vif_add_validate: Attempt to add a MONITOR interface...
[94008.545100] brcmfmac: brcmf_vif_add_validate: ... there is already a monitor interface, returning EOPNOTSUPP
[94008.545107] ieee80211 phy0: brcmf_cfg80211_add_iface: iface validation failed: err=-95
[130885.613184] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save disabled
[130888.639866] brcmfmac: brcmf_vif_add_validate: Attempt to add a MONITOR interface...
[130888.639884] brcmfmac: brcmf_vif_add_validate: ... there is already a monitor interface, returning EOPNOTSUPP
[130888.639891] ieee80211 phy0: brcmf_cfg80211_add_iface: iface validation failed: err=-95
[132870.122369] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save disabled
[132873.156327] brcmfmac: brcmf_vif_add_validate: Attempt to add a MONITOR interface...
[132873.156345] brcmfmac: brcmf_vif_add_validate: ... there is already a monitor interface, returning EOPNOTSUPP
[132873.156352] ieee80211 phy0: brcmf_cfg80211_add_iface: iface validation failed: err=-95
[135620.017168] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save disabled
[135623.046750] brcmfmac: brcmf_vif_add_validate: Attempt to add a MONITOR interface...
[135623.046769] brcmfmac: brcmf_vif_add_validate: ... there is already a monitor interface, returning EOPNOTSUPP
[135623.046776] ieee80211 phy0: brcmf_cfg80211_add_iface: iface validation failed: err=-95
[135798.650335] ieee80211 phy0: brcmf_fw_crashed: Firmware has halted or crashed
[135798.750228] ieee80211 phy0: brcmf_fil_cmd_data: bus is down. we have nothing to do.
[135798.750255] ieee80211 phy0: brcmf_net_mon_stop: BRCMF_C_SET_MONITOR error (-5)
[135798.977727] brcmfmac: F1 signature read @0x18000000=0x15264345
[135798.981613] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[135799.310392] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[135799.313703] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4345/6 wl0: Jan 24 2024 20:30:46 version 7.45.206 (nexmon.org: 2.2.2-514-gf195c-2) FWID 01-88ee44ea
[135799.443187] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled
[135799.960666] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled
[135801.219002] usbcore: deregistering interface driver brcmfmac
[135802.394468] brcmfmac: F1 signature read @0x18000000=0x15264345
[135802.398679] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[135802.705905] usbcore: registered new interface driver brcmfmac
[135802.708021] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[135802.711525] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4345/6 wl0: Jan 24 2024 20:30:46 version 7.45.206 (nexmon.org: 2.2.2-514-gf195c-2) FWID 01-88ee44ea
[135802.831390] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled
[135803.352685] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled
[135804.716690] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save disabled
[135805.407504] usbcore: deregistering interface driver brcmfmac
[135806.622508] brcmfmac: F1 signature read @0x18000000=0x15264345
[135806.626516] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[135806.716232] usbcore: registered new interface driver brcmfmac
```

jlinktu, 2024-02-01:
All I can see from that log is that a virtual monitor interface was added and then there were 7 more tries to add another one, which is not supported, only a single virtual monitor interface per interface can exist. Then the chip crashes and is restarted.

Can you provide me with the commands that you executed during this? Best would be everything that I need to replicate this behavior.

jayofelony, 2024-02-01:
```
uname -a
Linux jayofelony 6.1.0-rpi7-rpi-v8 #1 SMP PREEMPT Debian 1:6.1.63-1+rpt1 (2023-11-24) aarch64 GNU/Linux
```
RaspiOS 64bit lite Bookworm, using nexmon bcm43455c0 from DrSchottky his repo.

I am running Pwnagotchi, which uses a plugin to check journalctl for erros on the wifi firmware.
```
self.pattern = re.compile(r'brcmf_cfg80211_nexmon_set_channel.*?Set Channel failed')
self.pattern2 = re.compile(r'wifi error while hopping to channel')
self.pattern3 = re.compile(r'Firmware has halted or crashed')
self.pattern4 = re.compile(r'error 400: could not find interface wlan0mon')
```

You may not believe it, but when it sees a lot of accesspoints or quick injecting of packets it seems to crash.

jayofelony, 2024-02-01:
The command used to start monitor mode:
```bash
start_monitor_interface() {
  rfkill unblock all
  ifconfig wlan0 up
  sleep 3
  iw dev wlan0 set power_save off
  iw phy "$(iw phy | head -1 | cut -d" " -f2)" interface add wlan0mon type monitor
  sleep 2
  rfkill unblock all
  ifconfig wlan0 down
  ifconfig wlan0mon up
  iw dev wlan0mon set power_save off
}
```

The sleep command is needed, because on the RPi zero2w the firmware will crash immediately if we do not delay start monitor mode.

jlinktu, 2024-02-01:
If you are using @DrSchottky's version, you might want to ask there instead.

jayofelony, 2024-02-01:
`journalctl -k`
```
Feb 01 12:39:11 jayofelony kernel: brcmfmac: brcmf_vif_add_validate: Attempt to add a MONITOR interface...
Feb 01 12:39:11 jayofelony kernel: brcmfmac: brcmf_mon_add_vif: brcmf_mon_add_vif called
Feb 01 12:39:11 jayofelony kernel: brcmfmac: brcmf_mon_add_vif: Adding vif "wlan0mon"
Feb 01 12:39:12 jayofelony kernel: zram: Added device: zram0
Feb 01 12:39:12 jayofelony kernel: zram: Added device: zram1
Feb 01 12:39:12 jayofelony kernel: zram1: detected capacity change from 0 to 204800
Feb 01 12:39:12 jayofelony kernel: EXT4-fs (zram1): mounted filesystem with ordered data mode. Quota mode: none.
Feb 01 12:39:13 jayofelony kernel: zram: Added device: zram2
Feb 01 12:39:13 jayofelony kernel: zram2: detected capacity change from 0 to 40960
Feb 01 12:39:13 jayofelony kernel: EXT4-fs (zram2): mounted filesystem with ordered data mode. Quota mode: none.
Feb 01 12:39:18 jayofelony kernel: device wlan0mon entered promiscuous mode
Feb 01 12:40:08 jayofelony kernel: zram: Added device: zram3
Feb 01 12:40:08 jayofelony kernel: zram3: detected capacity change from 0 to 204800
Feb 01 12:40:08 jayofelony kernel: zram: Added device: zram4
Feb 01 12:40:08 jayofelony kernel: zram4: detected capacity change from 0 to 40960
Feb 01 12:40:11 jayofelony kernel: device wlan0mon left promiscuous mode
Feb 01 12:40:11 jayofelony kernel: device wlan0mon entered promiscuous mode
Feb 01 12:44:57 jayofelony kernel: ieee80211 phy0: brcmf_fw_crashed: Firmware has halted or crashed
Feb 01 12:44:57 jayofelony kernel: ieee80211 phy0: brcmf_fil_cmd_data: bus is down. we have nothing to do.
Feb 01 12:44:57 jayofelony kernel: ieee80211 phy0: brcmf_net_mon_stop: BRCMF_C_SET_MONITOR error (-5)
Feb 01 12:44:57 jayofelony kernel: device wlan0mon left promiscuous mode
Feb 01 12:44:57 jayofelony kernel: mmc1: card 0001 removed
Feb 01 12:44:57 jayofelony kernel: mmc1: new high speed SDIO card at address 0001
Feb 01 12:44:57 jayofelony kernel: brcmfmac: F1 signature read @0x18000000=0x15264345
Feb 01 12:44:57 jayofelony kernel: brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
Feb 01 12:44:57 jayofelony kernel: brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
Feb 01 12:44:57 jayofelony kernel: brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4345/6 wl0: Jan 24 2024 20:30:46 version 7.45.206 (nexmon.org: 2.2.2-514-gf195c-2) FWID 01-88ee44ea
Feb 01 12:44:57 jayofelony kernel: brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled
Feb 01 12:44:58 jayofelony kernel: brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled
```

`journalctl -S today`
```
Feb 01 12:44:51 jayofelony bettercap-launcher[943]: [12:44:51] [sys.log] [dbg] wifi hop changed
Feb 01 12:44:51 jayofelony bettercap-launcher[943]: [12:44:51] [sys.log] [inf] wifi sending association request to AP Ziggo (channel:11 encryption:WPA2)
Feb 01 12:44:52 jayofelony bettercap-launcher[943]: [12:44:52] [sys.log] [inf] wifi sending association request to AP Ziggo6611EA8 (channel:11 encryption:WPA2)
Feb 01 12:44:52 jayofelony bettercap-launcher[943]: [12:44:52] [sys.log] [inf] wifi deauthing client 70:f1:1c:48:fd:af (Shenzhen Ogemray Technology Co.,Ltd) from AP Ziggo6611EA8 (channel:11 encryption:WPA2)
Feb 01 12:44:54 jayofelony bettercap-launcher[943]: [12:44:54] [sys.log] [inf] wifi deauthing client d0:ba:e4:8b:0d:94 (Shanghai MXCHIP Information Technology Co., Ltd.) from AP Ziggo6611EA8 (channel:11 encryption:WPA2)
Feb 01 12:44:56 jayofelony bettercap-launcher[943]: [12:44:56] [sys.log] [dbg] wifi skipping deauth frame from d0:ba:e4:8b:0d:94
Feb 01 12:44:56 jayofelony bettercap-launcher[943]: [12:44:56] [sys.log] [inf] wifi deauthing client d2:ac:32:45:e5:1b from AP Ziggo6611EA8 (channel:11 encryption:WPA2)
Feb 01 12:44:57 jayofelony kernel: ieee80211 phy0: brcmf_fw_crashed: Firmware has halted or crashed
Feb 01 12:44:57 jayofelony kernel: ieee80211 phy0: brcmf_fil_cmd_data: bus is down. we have nothing to do.
Feb 01 12:44:57 jayofelony kernel: ieee80211 phy0: brcmf_net_mon_stop: BRCMF_C_SET_MONITOR error (-5)
Feb 01 12:44:57 jayofelony kernel: device wlan0mon left promiscuous mode
Feb 01 12:44:57 jayofelony bettercap-launcher[943]: [12:44:57] [sys.log] [err] wifi could not inject WiFi packet: send: No such device or address
Feb 01 12:44:57 jayofelony bettercap-launcher[943]: [12:44:57] [sys.log] [err] wifi could not inject WiFi packet: send: No such device or address
Feb 01 12:44:57 jayofelony NetworkManager[683]: <info>  [1706787897.2023] device (wlan0): state change: unavailable -> unmanaged (reason 'removed', sys-iface-state: 'removed')
Feb 01 12:44:57 jayofelony bettercap-launcher[943]: [12:44:57] [sys.log] [err] wifi could not inject WiFi packet: send: No such device or address
Feb 01 12:44:57 jayofelony bettercap-launcher[943]: [12:44:57] [sys.log] [err] wifi could not inject WiFi packet: send: No such device or address
Feb 01 12:44:57 jayofelony bettercap-launcher[943]: [12:44:57] [sys.log] [err] wifi could not inject WiFi packet: send: No such device or address
Feb 01 12:44:57 jayofelony bettercap-launcher[943]: [12:44:57] [sys.log] [err] wifi could not inject WiFi packet: send: No such device or address
Feb 01 12:44:57 jayofelony bettercap-launcher[943]: [12:44:57] [sys.log] [err] wifi could not inject WiFi packet: send: No such device or address
[... cut ...]
Feb 01 12:44:57 jayofelony bettercap-launcher[943]: [12:44:57] [sys.log] [err] wifi could not inject WiFi packet: send: No such device or address
Feb 01 12:44:57 jayofelony bettercap-launcher[943]: [12:44:57] [sys.log] [err] wifi could not inject WiFi packet: send: No such device or address
Feb 01 12:44:57 jayofelony bettercap-launcher[943]: [12:44:57] [sys.log] [err] wifi could not inject WiFi packet: send: No such device or address
Feb 01 12:44:57 jayofelony bettercap-launcher[943]: [12:44:57] [sys.log] [err] wifi could not inject WiFi packet: send: No such device or address
```

jayofelony, 2024-02-01:
@jlinktu Of course I also tried with your repository. I had to alter the installation a bit to use libmpfr-6.2.0 and libisl-23.2.0.
```
Feb 01 13:32:51 jayofelony bettercap-launcher[923]: [13:32:51] [sys.log] [inf] wifi deauthing client 70:f1:1c:48:fd:af (Shenzhen Ogemray Technology Co.,Ltd) from AP Ziggo6611EA8 (channel:11 encryption:WPA2)
Feb 01 13:32:51 jayofelony kernel: ieee80211 phy0: brcmf_fw_crashed: Firmware has halted or crashed
Feb 01 13:32:52 jayofelony kernel: ieee80211 phy0: brcmf_fil_cmd_data: bus is down. we have nothing to do.
Feb 01 13:32:52 jayofelony kernel: ieee80211 phy0: brcmf_net_mon_stop: BRCMF_C_SET_MONITOR error (-5)
Feb 01 13:32:52 jayofelony kernel: device wlan0mon left promiscuous mode
Feb 01 13:32:52 jayofelony bettercap-launcher[923]: [13:32:52] [sys.log] [err] wifi could not inject WiFi packet: send: No such device or address
Feb 01 13:32:52 jayofelony bettercap-launcher[923]: [13:32:52] [sys.log] [err] wifi could not inject WiFi packet: send: No such device or address
Feb 01 13:32:52 jayofelony NetworkManager[678]: <info>  [1706790772.0684] device (wlan0): state change: unavailable -> unmanaged (reason 'removed', sys-iface-state: 'removed')
Feb 01 13:32:52 jayofelony bettercap-launcher[923]: [13:32:52] [sys.log] [err] wifi could not inject WiFi packet: send: No such device or address
Feb 01 13:32:52 jayofelony bettercap-launcher[923]: [13:32:52] [sys.log] [err] wifi could not inject WiFi packet: send: No such device or address
Feb 01 13:32:52 jayofelony bettercap-launcher[923]: [13:32:52] [sys.log] [err] wifi could not inject WiFi packet: send: No such device or address
Feb 01 13:32:52 jayofelony NetworkManager[678]: <info>  [1706790772.0934] radio killswitch /sys/devices/platform/soc/fe300000.mmcnr/mmc_host/mmc1/mmc1:0001/mmc1:0001:1/ieee80211/phy0/rfkill1 disappeared
Feb 01 13:32:52 jayofelony bettercap-launcher[923]: [13:32:52] [sys.log] [err] wifi could not inject WiFi packet: send: No such device or address
Feb 01 13:32:52 jayofelony bettercap-launcher[923]: [13:32:52] [sys.log] [err] wifi could not inject WiFi packet: send: No such device or address
Feb 01 13:32:52 jayofelony bettercap-launcher[923]: [13:32:52] [sys.log] [err] wifi could not inject WiFi packet: send: No such device or address
Feb 01 13:32:52 jayofelony systemd[1]: Starting systemd-rfkill.service - Load/Save RF Kill Switch Status...
Feb 01 13:32:52 jayofelony bettercap-launcher[923]: [13:32:52] [sys.log] [err] wifi could not inject WiFi packet: send: No such device or address
Feb 01 13:32:52 jayofelony systemd[1]: Started systemd-rfkill.service - Load/Save RF Kill Switch Status.
Feb 01 13:32:52 jayofelony bettercap-launcher[923]: [13:32:52] [sys.log] [err] wifi could not inject WiFi packet: send: No such device or address
Feb 01 13:32:52 jayofelony bettercap-launcher[923]: [13:32:52] [sys.log] [err] wifi could not inject WiFi packet: send: No such device or address
Feb 01 13:32:52 jayofelony bettercap-launcher[923]: [13:32:52] [sys.log] [err] wifi could not inject WiFi packet: send: No such device or address
Feb 01 13:32:52 jayofelony kernel: mmc1: card 0001 removed
Feb 01 13:32:52 jayofelony bettercap-launcher[923]: [13:32:52] [sys.log] [err] wifi could not inject WiFi packet: send: No such device or address
Feb 01 13:32:52 jayofelony bettercap-launcher[923]: [13:32:52] [sys.log] [err] wifi could not inject WiFi packet: send: No such device or address
Feb 01 13:32:52 jayofelony bettercap-launcher[923]: [13:32:52] [sys.log] [err] wifi could not inject WiFi packet: send: No such device or address
Feb 01 13:32:52 jayofelony bettercap-launcher[923]: [13:32:52] [sys.log] [err] wifi could not inject WiFi packet: send: No such device or address
Feb 01 13:32:52 jayofelony bettercap-launcher[923]: [13:32:52] [sys.log] [err] wifi could not inject WiFi packet: send: No such device or address
Feb 01 13:32:52 jayofelony bettercap-launcher[923]: [13:32:52] [sys.log] [err] wifi could not inject WiFi packet: send: No such device or address
Feb 01 13:32:52 jayofelony bettercap-launcher[923]: [13:32:52] [sys.log] [err] wifi could not inject WiFi packet: send: No such device or address
Feb 01 13:32:52 jayofelony bettercap-launcher[923]: [13:32:52] [sys.log] [err] wifi could not inject WiFi packet: send: No such device or address
Feb 01 13:32:52 jayofelony bettercap-launcher[923]: [13:32:52] [sys.log] [err] wifi could not inject WiFi packet: send: No such device or address
Feb 01 13:32:52 jayofelony bettercap-launcher[923]: [13:32:52] [sys.log] [err] wifi could not inject WiFi packet: send: No such device or address
Feb 01 13:32:52 jayofelony kernel: mmc1: new high speed SDIO card at address 0001
Feb 01 13:32:52 jayofelony kernel: brcmfmac: F1 signature read @0x18000000=0x15264345
Feb 01 13:32:52 jayofelony bettercap-launcher[923]: [13:32:52] [sys.log] [err] wifi could not inject WiFi packet: send: No such device or address
Feb 01 13:32:52 jayofelony kernel: brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
Feb 01 13:32:52 jayofelony bettercap-launcher[923]: [13:32:52] [sys.log] [err] wifi could not inject WiFi packet: send: No such device or address
Feb 01 13:32:52 jayofelony bettercap-launcher[923]: [13:32:52] [sys.log] [err] wifi could not inject WiFi packet: send: No such device or address
Feb 01 13:32:52 jayofelony bettercap-launcher[923]: [13:32:52] [sys.log] [err] wifi could not inject WiFi packet: send: No such device or address
[... cut ...]
Feb 01 13:32:57 jayofelony systemd[1]: systemd-rfkill.service: Deactivated successfully.
Feb 01 13:33:01 jayofelony bettercap-launcher[923]: [13:33:01] [sys.log] [dbg] wifi station 54:67:51:b9:0b:d0 not seen in 42.116130834s, removing.
Feb 01 13:33:06 jayofelony NetworkManager[678]: <warn>  [1706790786.8674] device (wlan0): re-acquiring supplicant interface (#1).
Feb 01 13:33:09 jayofelony NetworkManager[678]: <error> [1706790789.8702] device (wlan0): Couldn't initialize supplicant interface: Failed to D-Bus activate wpa_supplicant service
``

jlinktu, 2024-02-01:
```
--- snip ---
Feb 01 13:32:51 jayofelony bettercap-launcher[923]: [13:32:51] [sys.log] [inf] wifi deauthing client 70:f1:1c:48:fd:af (Shenzhen Ogemray Technology Co.,Ltd) from AP Ziggo6611EA8 (channel:11 encryption:WPA2)
Feb 01 13:32:51 jayofelony kernel: ieee80211 phy0: brcmf_fw_crashed: Firmware has halted or crashed
--- snip ---
```
Leads me to think that the injection of the deauth frame kills the chip.

jayofelony, 2024-02-01:
That could surely be, although it had done some deauthing already prior to that part. Here's the rest right before the snip.
```
Feb 01 13:31:08 jayofelony bettercap-launcher[923]: [13:31:08] [sys.log] [dbg] wifi hop changed
Feb 01 13:31:08 jayofelony bettercap-launcher[923]: [13:31:08] [sys.log] [inf] wifi sending association request to AP Ziggo (channel:11 encryption:WPA2)
Feb 01 13:31:09 jayofelony bettercap-launcher[923]: [13:31:09] [sys.log] [inf] wifi sending association request to AP Ziggo6611EA8 (channel:11 encryption:WPA2)
Feb 01 13:31:09 jayofelony bettercap-launcher[923]: [13:31:09] [sys.log] [inf] wifi deauthing client 70:f1:1c:48:fd:af (Shenzhen Ogemray Technology Co.,Ltd) from AP Ziggo6611EA8 (channel:11 encryption:WPA2)
Feb 01 13:31:11 jayofelony bettercap-launcher[923]: [13:31:11] [sys.log] [inf] wifi deauthing client 8c:5a:25:58:1a:cf (ARRIS Group, Inc.) from AP Ziggo6611EA8 (channel:11 encryption:WPA2)
Feb 01 13:31:12 jayofelony bettercap-launcher[923]: [13:31:12] [sys.log] [inf] wifi deauthing client 8c:5a:25:58:1a:cf (ARRIS Group, Inc.) from AP Ziggo6611EA8 (channel:44 encryption:WPA2)
Feb 01 13:31:13 jayofelony bettercap-launcher[923]: [13:31:13] [sys.log] [dbg] wifi skipping deauth frame from 8c:5a:25:58:1a:cf
Feb 01 13:31:13 jayofelony bettercap-launcher[923]: [13:31:13] [sys.log] [dbg] wifi skipping deauth frame from 8c:5a:25:58:1a:cf
Feb 01 13:31:13 jayofelony bettercap-launcher[923]: [13:31:13] [sys.log] [inf] wifi deauthing client d0:ba:e4:8b:0d:94 (Shanghai MXCHIP Information Technology Co., Ltd.) from AP Ziggo6611EA8 (channel:11 encryption:WPA2)
Feb 01 13:31:14 jayofelony bettercap-launcher[923]: [13:31:14] [sys.log] [dbg] wifi got frame 1/4 of the 54:67:51:b9:0c:a6 <-> 8c:5a:25:58:1a:cf handshake (without PMKID) (anonce:f9bc5e214bef353303190781796c41e388c50d053fa0e7bb9bcf8026fa930348)
Feb 01 13:31:14 jayofelony bettercap-launcher[923]: [13:31:14] [sys.log] [dbg] wifi adding beacon frame to handshake for 54:67:51:b9:0c:a6
Feb 01 13:31:14 jayofelony bettercap-launcher[923]: [13:31:14] [sys.log] [dbg] wifi got frame 1/4 of the 54:67:51:b9:0c:a6 <-> 8c:5a:25:58:1a:cf handshake (without PMKID) (anonce:f9bc5e214bef353303190781796c41e388c50d053fa0e7bb9bcf8026fa930348)
Feb 01 13:31:14 jayofelony bettercap-launcher[923]: [13:31:14] [sys.log] [dbg] wifi adding beacon frame to handshake for 54:67:51:b9:0c:a6
Feb 01 13:31:14 jayofelony bettercap-launcher[923]: [13:31:14] [sys.log] [dbg] wifi got frame 2/4 of the 54:67:51:b9:0c:a6 <-> 8c:5a:25:58:1a:cf handshake (snonce:1a17d4cedb9ad8254e70596a97383b394927b38502d20011318411d1b9bd2a81 mic:2ed305795aeabf818a8edd422320186b)
Feb 01 13:31:14 jayofelony bettercap-launcher[923]: [13:31:14] [sys.log] [dbg] wifi got frame 3/4 of the 54:67:51:b9:0c:a6 <-> 8c:5a:25:58:1a:cf handshake (mic:6e49c2d65394731a6b8d78ba40aab2dc)
Feb 01 13:31:15 jayofelony bettercap-launcher[923]: [13:31:15] [sys.log] [inf] wifi deauthing client d2:ac:32:45:e5:1b from AP Ziggo6611EA8 (channel:11 encryption:WPA2)
Feb 01 13:31:21 jayofelony bettercap-launcher[923]: [13:31:21] [sys.log] [dbg] wifi got frame 1/4 of the 54:67:51:b9:0c:a6 <-> d0:ba:e4:8b:0d:94 handshake (without PMKID) (anonce:6d0d4810fcb4d1aad757769295f072024c700d0a9fc9d466352030dd8598008d)
Feb 01 13:31:21 jayofelony bettercap-launcher[923]: [13:31:21] [sys.log] [dbg] wifi adding beacon frame to handshake for 54:67:51:b9:0c:a6
Feb 01 13:31:21 jayofelony bettercap-launcher[923]: [13:31:21] [sys.log] [dbg] wifi got frame 2/4 of the 54:67:51:b9:0c:a6 <-> d0:ba:e4:8b:0d:94 handshake (snonce:290902b473e5a2bf672ef0c2afd9cab9ef6c3829b9cad9afc2f02e67c0a2e573 mic:ef1702df3cb907a45f58c2fab4248625)
Feb 01 13:31:21 jayofelony bettercap-launcher[923]: [13:31:21] [sys.log] [dbg] wifi got frame 3/4 of the 54:67:51:b9:0c:a6 <-> d0:ba:e4:8b:0d:94 handshake (mic:07d11b79edb95f9b3d340a0e16b01338)
Feb 01 13:31:22 jayofelony bettercap-launcher[923]: [13:31:22] [sys.log] [dbg] wifi got frame 2/4 of the 54:67:51:b9:0c:a6 <-> d2:ac:32:45:e5:1b handshake (snonce:ecef2e719a7dcf76fbf4d3a7a47e667c488dbf59f72067b1af94e66bb582baef mic:bba318f12d22fb48d01b58ccb2928b85)
Feb 01 13:31:22 jayofelony bettercap-launcher[923]: [13:31:22] [sys.log] [dbg] wifi got frame 1/4 of the 54:67:51:b9:0c:a6 <-> d2:ac:32:45:e5:1b handshake (without PMKID) (anonce:7c8e191dd423ab97a00097a5314fe400970b649f2fe76841613b0157e3b97e27)
Feb 01 13:31:22 jayofelony bettercap-launcher[923]: [13:31:22] [sys.log] [dbg] wifi adding beacon frame to handshake for 54:67:51:b9:0c:a6
Feb 01 13:31:22 jayofelony bettercap-launcher[923]: [13:31:22] [sys.log] [dbg] wifi got frame 3/4 of the 54:67:51:b9:0c:a6 <-> d2:ac:32:45:e5:1b handshake (mic:b129595076216540ef62bebcd027986c)
Feb 01 13:31:37 jayofelony bettercap-launcher[923]: [13:31:37] [sys.log] [dbg] wifi got frame 1/4 of the 54:67:51:b9:0c:a6 <-> d0:ba:e4:8b:0d:94 handshake (without PMKID) (anonce:7e86730435449ca9bc08e979e8efee58b664108ec0dd83f08657002acf3c4366)
Feb 01 13:31:37 jayofelony bettercap-launcher[923]: [13:31:37] [sys.log] [dbg] wifi adding beacon frame to handshake for 54:67:51:b9:0c:a6
Feb 01 13:31:37 jayofelony bettercap-launcher[923]: [13:31:37] [sys.log] [dbg] wifi got frame 2/4 of the 54:67:51:b9:0c:a6 <-> d0:ba:e4:8b:0d:94 handshake (snonce:d66583c137c5a9c34546237382596945e70c8cd64569598273234645c4a9c537 mic:28d11a2fafebe5013584230ed69cbd60)
[... cut ...]
Feb 01 13:32:47 jayofelony bettercap-launcher[923]: [13:32:47] [sys.log] [inf] wifi deauthing client d0:ba:e4:8b:0d:94 (Shanghai MXCHIP Information Technology Co., Ltd.) from AP Ziggo6611EA8 (channel:11 encryption:WPA2)
Feb 01 13:32:49 jayofelony bettercap-launcher[923]: [13:32:49] [sys.log] [inf] wifi deauthing client d2:ac:32:45:e5:1b from AP Ziggo6611EA8 (channel:11 encryption:WPA2)
Feb 01 13:32:51 jayofelony bettercap-launcher[923]: [13:32:51] [sys.log] [inf] wifi deauthing client 70:f1:1c:48:fd:af (Shenzhen Ogemray Technology Co.,Ltd) from AP Ziggo6611EA8 (channel:11 encryption:WPA2)
Feb 01 13:32:51 jayofelony kernel: ieee80211 phy0: brcmf_fw_crashed: Firmware has halted or crashed
```

jlinktu, 2024-02-01:
Please load the `brcmfmac.ko` kernel module with the following parameter `debug=0x100006` to create more verbose logging.

jayofelony, 2024-02-01:
How do I do that exactly?

jlinktu, 2024-02-01:
Add `debug=0x100006` ad the end of [this line](https://github.com/seemoo-lab/nexmon/blob/d21826cca28f8a5ba64c1b242d7ee2a2a858d765/patches/bcm43455c0/7_45_206/nexmon/Makefile#L197) and run `make install-firmware` again.

jayofelony, 2024-02-01:
I don't know if this is the more verbose debug you expected:

```
Feb 01 15:24:59 jayofelony bettercap-launcher[942]: [15:24:59] [sys.log] [inf] wifi wifi.ap.ttl set to 384
Feb 01 15:24:59 jayofelony bettercap-launcher[942]: [15:24:59] [sys.log] [inf] wifi wifi.sta.ttl set to 282
Feb 01 15:24:59 jayofelony bettercap-launcher[942]: [15:24:59] [sys.log] [inf] wifi wifi.rssi.min set to -123
Feb 01 15:25:00 jayofelony bettercap-launcher[942]: [15:25:00] [sys.log] [dbg] wifi setting hopping channels to 6,7,8,11,36,40,48,56,60,100,104,116,128,136,140
Feb 01 15:25:00 jayofelony bettercap-launcher[942]: [15:25:00] [sys.log] [dbg] wifi new frequencies: [2437 2442 2447 2462 5180 5200 5240 5280 5300 5500 5520 5580 5640 5680 5700]
Feb 01 15:25:00 jayofelony bettercap-launcher[942]: [15:25:00] [sys.log] [dbg] wifi hop changed
Feb 01 15:25:58 jayofelony bettercap-launcher[942]: [15:25:58] [sys.log] [dbg] wifi setting hopping channels to 11
Feb 01 15:25:58 jayofelony bettercap-launcher[942]: [15:25:58] [sys.log] [dbg] wifi new frequencies: [2462]
Feb 01 15:25:58 jayofelony bettercap-launcher[942]: [15:25:58] [sys.log] [dbg] wifi hop changed
Feb 01 15:25:58 jayofelony bettercap-launcher[942]: [15:25:58] [sys.log] [inf] wifi sending association request to AP Ziggo6611EA8 (channel:11 encryption:WPA2)
Feb 01 15:25:58 jayofelony bettercap-launcher[942]: [15:25:58] [sys.log] [inf] wifi deauthing client 70:f1:1c:48:fd:af (Shenzhen Ogemray Technology Co.,Ltd) from AP Ziggo6611EA8 (channel:11 encryption:WPA2)
Feb 01 15:26:00 jayofelony bettercap-launcher[942]: [15:26:00] [sys.log] [inf] wifi deauthing client d2:ac:32:45:e5:1b from AP Ziggo6611EA8 (channel:11 encryption:WPA2)
Feb 01 15:26:02 jayofelony bettercap-launcher[942]: [15:26:02] [sys.log] [dbg] wifi got frame 1/4 of the 54:67:51:b9:0c:a6 <-> 70:f1:1c:48:fd:af handshake (without PMKID) (anonce:aa936907973b2bc6523844d7f6ca14acb7e2491dcdfccd1ab0e52c9e3ec2e349)
Feb 01 15:26:02 jayofelony bettercap-launcher[942]: [15:26:02] [sys.log] [dbg] wifi adding beacon frame to handshake for 54:67:51:b9:0c:a6
Feb 01 15:26:02 jayofelony bettercap-launcher[942]: [15:26:02] [sys.log] [dbg] wifi got frame 2/4 of the 54:67:51:b9:0c:a6 <-> 70:f1:1c:48:fd:af handshake (snonce:87b8d296e9f6b279fa3212adb79925842bb1ca9f0ee446354f9b10e3a1e924ab mic:ebe63a09ded852714d67c795f3e2251b)
Feb 01 15:26:02 jayofelony bettercap-launcher[942]: [15:26:02] [sys.log] [dbg] wifi got frame 3/4 of the 54:67:51:b9:0c:a6 <-> 70:f1:1c:48:fd:af handshake (mic:dcb0c4daaa701ced9555088c00dc1591)
Feb 01 15:26:02 jayofelony bettercap-launcher[942]: [15:26:02] [sys.log] [inf] wifi deauthing client d0:ba:e4:8b:0d:94 (Shanghai MXCHIP Information Technology Co., Ltd.) from AP Ziggo6611EA8 (channel:11 encryption:WPA2)
Feb 01 15:26:03 jayofelony kernel: ieee80211 phy0: brcmf_fw_crashed: Firmware has halted or crashed
Feb 01 15:26:03 jayofelony kernel: ieee80211 phy0: brcmf_fil_cmd_data: bus is down. we have nothing to do.
Feb 01 15:26:03 jayofelony kernel: ieee80211 phy0: brcmf_net_mon_stop: BRCMF_C_SET_MONITOR error (-5)
Feb 01 15:26:03 jayofelony kernel: device wlan0mon left promiscuous mode
Feb 01 15:26:03 jayofelony bettercap-launcher[942]: [15:26:03] [sys.log] [err] wifi could not inject WiFi packet: send: No such device or address
Feb 01 15:26:03 jayofelony bettercap-launcher[942]: [15:26:03] [sys.log] [err] wifi could not inject WiFi packet: send: No such device or address
Feb 01 15:26:03 jayofelony bettercap-launcher[942]: [15:26:03] [sys.log] [err] wifi could not inject WiFi packet: send: No such device or address
Feb 01 15:26:03 jayofelony NetworkManager[670]: <info>  [1706797563.6724] device (wlan0): state change: unavailable -> unmanaged (reason 'removed', sys-iface-state: 'removed')
Feb 01 15:26:03 jayofelony bettercap-launcher[942]: [15:26:03] [sys.log] [err] wifi could not inject WiFi packet: send: No such device or address
Feb 01 15:26:03 jayofelony systemd[1]: Starting systemd-rfkill.service - Load/Save RF Kill Switch Status...
Feb 01 15:26:03 jayofelony NetworkManager[670]: <info>  [1706797563.7313] radio killswitch /sys/devices/platform/soc/fe300000.mmcnr/mmc_host/mmc1/mmc1:0001/mmc1:0001:1/ieee80211/phy0/rfkill1 disappeared
Feb 01 15:26:03 jayofelony systemd[1]: Started systemd-rfkill.service - Load/Save RF Kill Switch Status.
Feb 01 15:26:03 jayofelony kernel: mmc1: card 0001 removed
Feb 01 15:26:03 jayofelony kernel: mmc1: new high speed SDIO card at address 0001
Feb 01 15:26:03 jayofelony kernel: brcmfmac: F1 signature read @0x18000000=0x15264345
Feb 01 15:26:03 jayofelony kernel: brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
Feb 01 15:26:04 jayofelony kernel: brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
Feb 01 15:26:04 jayofelony kernel: brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4345/6 wl0: Feb  1 2024 15:11:25 version 7.45.206 (nexmon.org: 2.2.2-465-gd2182-dirty-5) FWID 01-88ee44ea
Feb 01 15:26:04 jayofelony NetworkManager[670]: <info>  [1706797564.2022] rfkill2: found Wi-Fi radio killswitch (at /sys/devices/platform/soc/fe300000.mmcnr/mmc_host/mmc1/mmc1:0001/mmc1:0001:1/ieee80211/phy1/rfkill2) (driver brcmfmac)
Feb 01 15:26:04 jayofelony NetworkManager[670]: <info>  [1706797564.2246] device (wlan0): driver supports Access Point (AP) mode
Feb 01 15:26:04 jayofelony NetworkManager[670]: <info>  [1706797564.2258] manager: (wlan0): new 802.11 Wi-Fi device (/org/freedesktop/NetworkManager/Devices/5)
[... cut ...]
Feb 01 15:26:10 jayofelony bettercap-launcher[942]: [15:26:10] [sys.log] [war] wifi error while hopping to channel 60: iw: out=command failed: No such device (-19) err=exit status 237
Feb 01 15:26:11 jayofelony bettercap-launcher[942]: [15:26:11] [sys.log] [war] wifi error while hopping to channel 100: iw: out=command failed: No such device (-19) err=exit status 237
Feb 01 15:26:11 jayofelony bettercap-launcher[942]: [15:26:11] [sys.log] [war] wifi error while hopping to channel 104: iw: out=command failed: No such device (-19) err=exit status 237
```

jlinktu, 2024-02-01:
Please provide the related output of `dmesg` instead.

jayofelony, 2024-02-01:
```
sudo dmesg | grep brcm
[    1.213523] irq_brcmstb_l2: registered L2 intc (/soc/interrupt-controller@7ef00100, parent irq: 27)
[    1.223708] brcm-pcie fd500000.pcie: host bridge /scb/pcie@7d500000 ranges:
[    1.223766] brcm-pcie fd500000.pcie:   No bus range found for /scb/pcie@7d500000, using [bus 00-ff]
[    1.223855] brcm-pcie fd500000.pcie:      MEM 0x0600000000..0x063fffffff -> 0x00c0000000
[    1.223946] brcm-pcie fd500000.pcie:   IB MEM 0x0000000000..0x00bfffffff -> 0x0400000000
[    1.224496] brcm-pcie fd500000.pcie: setting SCB_ACCESS_EN, READ_UR_MODE, MAX_BURST_SIZE
[    1.224928] brcm-pcie fd500000.pcie: PCI host bridge to bus 0000:00
[    1.336228] brcm-pcie fd500000.pcie: link up, 5.0 GT/s PCIe x1 (SSC)
[    1.339481] brcm-pcie fd500000.pcie: clkreq control enabled
[    2.512869] brcmstb-i2c fef04500.i2c:  @97500hz registered in polling mode
[    2.529003] brcmstb-i2c fef09500.i2c:  @97500hz registered in polling mode
[    7.683333] brcmfmac: loading out-of-tree module taints kernel.
[    7.952135] brcmfmac: F1 signature read @0x18000000=0x15264345
[    8.070178] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[    8.071027] usbcore: registered new interface driver brcmfmac
[    8.132171] Bluetooth: hci0: BCM4345C0 'brcm/BCM4345C0.raspberrypi,4-model-b.hcd' Patch
[    8.494027] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[    8.504246] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4345/6 wl0: Feb  1 2024 15:11:25 version 7.45.206 (nexmon.org: 2.2.2-465-gd2182-dirty-5) FWID 01-88ee44ea
[   11.276308] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled
[   12.162607] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled
[   14.296486] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save disabled
[   14.324592] brcmfmac: brcmf_vif_add_validate: Attempt to add a MONITOR interface...
[   14.324614] brcmfmac: brcmf_mon_add_vif: brcmf_mon_add_vif called
[   14.324618] brcmfmac: brcmf_mon_add_vif: Adding vif "wlan0mon"
[  724.993874] ieee80211 phy0: brcmf_fw_crashed: Firmware has halted or crashed
[  725.092130] ieee80211 phy0: brcmf_fil_cmd_data: bus is down. we have nothing to do.
[  725.092157] ieee80211 phy0: brcmf_net_mon_stop: BRCMF_C_SET_MONITOR error (-5)
[  725.301499] brcmfmac: F1 signature read @0x18000000=0x15264345
[  725.305485] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[  725.633166] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[  725.636488] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4345/6 wl0: Feb  1 2024 15:11:25 version 7.45.206 (nexmon.org: 2.2.2-465-gd2182-dirty-5) FWID 01-88ee44ea
[  725.758595] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled
[  726.303989] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled
```

jlinktu, 2024-02-01:
Check if debug output is set with `sudo cat /sys/module/brcmfmac/parameters/debug`, I would have expected more output.

jayofelony, 2024-02-01:
Doesnt seem like it

sudo cat /sys/module/brcmfmac/parameters/debug 
0

jayofelony, 2024-02-01:
[image]

jlinktu, 2024-02-01:
Any error or warning when you ran `make install-firmware` with this modification?

jayofelony, 2024-02-01:
None
[image]

jlinktu, 2024-02-01:
Let's try to add even more output with: `debug=0x120406`
And check in a second window the live output with `dmesg -w` to see if the output level is increased up to when the firmware crashes the first time.

jayofelony, 2024-02-01:
okay its running, itll be about 5-10 minutes

jayofelony, 2024-02-01:
```
9.645828] Adding 2097148k swap on /var/swap.  Priority:-2 extents:25 across:112771068k SSFS
[    9.678772] Bluetooth: hci0: BCM: features 0x2f
[    9.680070] Bluetooth: hci0: BCM43455 37.4MHz Raspberry Pi 3+-0190
[    9.680081] Bluetooth: hci0: BCM4345C0 (003.001.025) build 0382
[   10.394433] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[   10.394452] Bluetooth: BNEP filters: protocol multicast
[   10.394467] Bluetooth: BNEP socket layer initialized
[   10.399095] Bluetooth: MGMT ver 1.22
[   10.434404] NET: Registered PF_ALG protocol family
[   11.089189] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled
[   12.001260] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled
[   12.350136] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[   12.351075] bcmgenet fd580000.ethernet eth0: Link is Up - 1Gbps/Full - flow control rx/tx
[   14.110960] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save disabled
[   14.136061] brcmfmac: brcmf_vif_add_validate: Attempt to add a MONITOR interface...
[   14.136084] brcmfmac: brcmf_mon_add_vif: brcmf_mon_add_vif called
[   14.136088] brcmfmac: brcmf_mon_add_vif: Adding vif "wlan0mon"
[   14.835229] zram: Added device: zram0
[   14.844087] zram: Added device: zram1
[   14.877320] zram1: detected capacity change from 0 to 204800
[   14.926849] EXT4-fs (zram1): mounted filesystem with ordered data mode. Quota mode: none.
[   15.769491] zram: Added device: zram2
[   15.770181] zram2: detected capacity change from 0 to 40960
[   15.797255] EXT4-fs (zram2): mounted filesystem with ordered data mode. Quota mode: none.
[   21.215544] device wlan0mon entered promiscuous mode
[   54.105518] zram: Added device: zram3
[   54.106668] zram3: detected capacity change from 0 to 204800
[   54.124553] zram: Added device: zram4
[   54.125156] zram4: detected capacity change from 0 to 40960
[   56.531374] device wlan0mon left promiscuous mode
[   56.583291] device wlan0mon entered promiscuous mode
[  231.032567] ieee80211 phy0: brcmf_fil_cmd_data: bus is down. we have nothing to do.
[  231.032585] ieee80211 phy0: brcmf_net_mon_stop: BRCMF_C_SET_MONITOR error (-5)
[  231.033059] device wlan0mon left promiscuous mode
[  231.161564] usbcore: deregistering interface driver brcmfmac
[  231.224371] brcmfmac: brcmfmac_module_init No platform data available.
[  231.224625] brcmfmac: brcmf_ops_sdio_probe Enter
[  231.224633] brcmfmac: brcmf_ops_sdio_probe Class=0
[  231.224638] brcmfmac: brcmf_ops_sdio_probe sdio vendor ID: 0x02d0
[  231.224663] brcmfmac: brcmf_ops_sdio_probe sdio device ID: 0xa9a6
[  231.224678] brcmfmac: brcmf_ops_sdio_probe Function#: 1
[  231.226583] brcmfmac: brcmf_ops_sdio_probe Enter
[  231.226600] brcmfmac: brcmf_ops_sdio_probe Class=0
[  231.226605] brcmfmac: brcmf_ops_sdio_probe sdio vendor ID: 0x02d0
[  231.226610] brcmfmac: brcmf_ops_sdio_probe sdio device ID: 0xa9a6
[  231.226614] brcmfmac: brcmf_ops_sdio_probe Function#: 2
[  231.226620] brcmfmac: brcmf_ops_sdio_probe F2 found, calling brcmf_sdiod_probe...
[  231.227202] brcmfmac: brcmf_sdiod_probe set F2 blocksize to 512
[  231.227273] brcmfmac: brcmf_sdio_probe Enter
[  231.227513] brcmfmac: F1 signature read @0x18000000=0x15264345
[  231.227718] brcmfmac: brcmf_chip_recognition found AXI chip: BCM4345/6
[  231.230352] brcmfmac: brcmf_chip_cores_check  [1 ] core 0x800:51  base 0x18000000 wrap 0x18100000
[  231.230366] brcmfmac: brcmf_chip_cores_check  [2 ] core 0x812:54  base 0x18001000 wrap 0x18101000
[  231.230372] brcmfmac: brcmf_chip_cores_check  [3 ] core 0x83e:9   base 0x18002000 wrap 0x18102000
[  231.230377] brcmfmac: brcmf_chip_cores_check  [4 ] core 0x83c:14  base 0x18003000 wrap 0x18103000
[  231.230382] brcmfmac: brcmf_chip_cores_check  [5 ] core 0x829:21  base 0x18004000 wrap 0x18104000
[  231.230387] brcmfmac: brcmf_chip_cores_check  [6 ] core 0x135:0   base 0x00000000 wrap 0x18107000
[  231.230392] brcmfmac: brcmf_chip_cores_check  [7 ] core 0x240:0   base 0x00000000 wrap 0x00000000
[  231.230398] brcmfmac: brcmf_chip_set_passive Enter
[  231.231673] brcmfmac: brcmf_chip_get_raminfo RAM: base=0x198000 size=819200 (0xc8000) sr=0 (0x0)
[  231.231749] brcmfmac: brcmf_chip_setup ccrev=51, pmurev=27, pmucaps=0x39d05f1b
[  231.231755] brcmfmac: brcmf_get_module_param Enter, bus=0, chip=17221, rev=6
[  231.231772] brcmfmac: brcmf_sdiod_sgtable_alloc nents=35
[  231.231778] brcmfmac: brcmf_sdio_kso_init Enter
[  231.231798] brcmfmac: brcmf_sdio_drivestrengthinit No SDIO driver strength init needed for chip BCM4345/6 rev 6 pmurev 27
[  231.232272] brcmfmac: brcmf_sdio_probe completed!!
[  231.232291] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[  231.232298] brcmfmac: brcmf_fw_get_firmwares enter: dev=mmc1:0001:1
[  231.233071] brcmfmac: brcmf_fw_complete_request firmware brcm/brcmfmac43455-sdio.bin found
[... 422 lines cut ...]
[  231.547783] brcmfmac: brcmf_sdiod_recv_pkt addr = 0x18000000, size = 64
[  231.547828] brcmfmac: brcmf_sdio_read_control Enter
[  231.547864] brcmfmac: brcmf_sdiod_recv_pkt addr = 0x18000000, size = 64
[  231.548077] brcmfmac: brcmf_sdio_bus_txctl ctrl_frame complete, err=0
[  231.548092] brcmfmac: brcmf_sdio_bus_rxctl Enter
[  231.548114] brcmfmac: brcmf_sdio_bus_txctl Enter
[  231.548131] brcmfmac: brcmf_sdio_dpc Enter
[  231.548142] brcmfmac: brcmf_sdio_bus_sleep Enter: request WAKE currently WAKE
[  231.548147] brcmfmac: brcmf_sdio_clkctl Enter
[  231.548151] brcmfmac: brcmf_sdio_bus_sleep new state WAKE
[... cut ...]
[  231.548585] brcmfmac: brcmf_sdio_isr Enter
[  231.548672] brcmfmac: brcmf_sdio_dpc Enter
[  231.548682] brcmfmac: brcmf_sdio_bus_sleep Enter: request WAKE currently WAKE
[  231.548687] brcmfmac: brcmf_sdio_clkctl Enter
[  231.548691] brcmfmac: brcmf_sdio_bus_sleep new state WAKE
[  231.548695] brcmfmac: brcmf_sdio_bus_sleep Exit: err=0
[  231.548701] brcmfmac: brcmf_sdio_readframes Enter
[  231.548712] brcmfmac: brcmf_sdiod_recv_pkt addr = 0x18000000, size = 64
```

Than it just spams this WAKE / SLEEP stuff

jlinktu, 2024-02-01:
No crash as far as I can see.

jayofelony, 2024-02-01:
no, my bad. service for scanning wasnt runing LOL

jayofelony, 2024-02-01:
[image]
some more errors/warnings

DrSchottky, 2024-02-01:
For what I've seen it fails here
https://github.com/FlUxIuS/rt-ac86u_et/blob/master/release/src-rt-5.02hnd/bcmdrivers/broadcom/net/wl/impl51/4365/src/wl/sys/wlc_tx.c#L2892

jayofelony, 2024-02-01:
I trust you experts in this know what you are talking about, I don't even know if it's a nexmon bug or something that my software is doing wrong. Heck, for all I know it's an issue in the original part of the firmware.

jlinktu, 2024-02-01:
@DrSchottky already pointed out the correct location of where the crash occurs.
Together with the error log, I can tell the following:
The method in question fetches packets from a queue and tries to forward them to the transmit chain. Each of these packets have a station control block associated with them, and these station control blocks in turn have a BSS config associated.
At the point of failure, register `r7` is supposed to hold a pointer to such a BSS config of the packet that is currently processed. However, for some reason `r7` holds a value that is a faulty address, and when it is accessed, the core crashes. The faulty address is `be23f199`, which is simply the four bytes located at address `0x10`. That is because the value in `r7` is taken from an offset of `0x10` from the pointer (station control block) in `r6`, but `r6` holds the value `0`. This value is taken from an offset of the pointer in `r1`, which is `2355ac`. Thus, the problem seems to lie in an empty station control block pointer. I remember discussing that problem before and we had a patch that made the firmware to crash less likely, but couldn't fully fix it. Let me check tomorrow if I can find it.

jayofelony, 2024-02-01:
> @DrSchottky already pointed out the correct location of where the crash occurs. Together with the error log, I can tell the following: The method in question fetches packets from a queue and tries [...]

I am happy to test any patches you can think of!

jlinktu, 2024-02-02:
Can you please replace `src/sendframe.c` in the nexmon patch directory with [sendframe-patch-v1.zip](https://github.com/seemoo-lab/nexmon/files/14139075/sendframe-patch-v1.zip) (unzip it first), rerun `make install-firmware`.  Then go ahead as usual and after some time, or even better, when you see that bettercap tried to transmit several deauth frames in a short time period, get the console output like this:
```
sudo -s cat /sys/kernel/debug/ieee80211/`iw dev wlan0 info | gawk '/wiphy/ {printf "phy" $2}'`/forensics; echo
```
Same if you experience a crash + then the `dmesg` output could be helpful again.

jayofelony, 2024-02-02:
Crashlog:

```
[79828.576091] brcmfmac: brcmf_sdio_bus_sleep Enter: request WAKE currently WAKE
[79828.576096] brcmfmac: brcmf_sdio_clkctl Enter
[79828.576099] brcmfmac: brcmf_sdio_bus_sleep new state WAKE
[79828.576103] brcmfmac: brcmf_sdio_bus_sleep Exit: err=0
[79828.576109] brcmfmac: brcmf_sdio_readframes Enter
[79828.576116] brcmfmac: brcmf_sdiod_recv_pkt addr = 0x18000000, size = 64
[79828.576148] brcmfmac: brcmf_sdiod_recv_pkt addr = 0x18000000, size = 0
[79828.576164] brcmfmac: brcmf_sdiod_recv_pkt addr = 0x18000000, size = 64
[79828.576204] brcmfmac: brcmf_sdio_dpc Enter
[79828.576225] brcmfmac: brcmf_sdio_isr Enter
[79828.576290] brcmfmac: brcmf_sdio_bus_sleep Enter: request WAKE currently WAKE
[79828.576295] brcmfmac: brcmf_sdio_clkctl Enter
[79828.576298] brcmfmac: brcmf_sdio_bus_sleep new state WAKE
[79828.576302] brcmfmac: brcmf_sdio_bus_sleep Exit: err=0
[79828.576307] brcmfmac: brcmf_sdio_readframes Enter
[79828.576314] brcmfmac: brcmf_sdiod_recv_pkt addr = 0x18000000, size = 64
[79828.576346] brcmfmac: brcmf_sdiod_recv_pkt addr = 0x18000000, size = 0
[79828.576362] brcmfmac: brcmf_sdiod_recv_pkt addr = 0x18000000, size = 64
[79828.576395] brcmfmac: brcmf_sdio_dpc Enter
[79828.576402] brcmfmac: brcmf_sdio_bus_sleep Enter: request WAKE currently WAKE
[79828.576406] brcmfmac: brcmf_sdio_clkctl Enter
[79828.576409] brcmfmac: brcmf_sdio_bus_sleep new state WAKE
[79828.576413] brcmfmac: brcmf_sdio_bus_sleep Exit: err=0
[79828.589248] brcmfmac: brcmf_sdio_bus_sleep Enter: request WAKE currently WAKE
[79828.589268] brcmfmac: brcmf_sdio_clkctl Enter
[79828.589273] brcmfmac: brcmf_sdio_bus_sleep new state WAKE
[79828.589277] brcmfmac: brcmf_sdio_bus_sleep Exit: err=0
[79828.589352] brcmfmac: brcmf_sdiod_ramrw read 24 bytes at offset 0x00005ec4 in window 0x00258000
[79828.595611] brcmfmac: brcmf_sdio_isr Enter
[79828.595749] brcmfmac: brcmf_sdio_dpc Enter
[79828.595759] brcmfmac: brcmf_sdio_bus_sleep Enter: request WAKE currently WAKE
[79828.595764] brcmfmac: brcmf_sdio_clkctl Enter
[79828.595769] brcmfmac: brcmf_sdio_bus_sleep new state WAKE
[79828.595773] brcmfmac: brcmf_sdio_bus_sleep Exit: err=0
[79828.595780] brcmfmac: brcmf_sdio_readframes Enter
[79828.595791] brcmfmac: brcmf_sdiod_recv_pkt addr = 0x18000000, size = 64
[79828.595832] brcmfmac: brcmf_sdiod_recv_pkt addr = 0x18000000, size = 0
[79828.595859] brcmfmac: brcmf_sdio_isr Enter
[79828.595927] brcmfmac: brcmf_sdiod_recv_pkt addr = 0x18000000, size = 64
[79828.595961] brcmfmac: brcmf_sdiod_recv_pkt addr = 0x18000000, size = 0
[79828.595971] brcmfmac: brcmf_sdiod_recv_pkt addr = 0x18000000, size = 64
[79828.596005] brcmfmac: brcmf_sdio_sendfromq Enter
[79828.596014] brcmfmac: brcmf_sdio_txpkt Enter
[79828.596023] brcmfmac: brcmf_sdiod_send_pkt addr = 0x18000000, size = 2
[79828.596101] brcmfmac: brcmf_sdio_dpc Enter
[79828.596110] brcmfmac: brcmf_sdio_bus_sleep Enter: request WAKE currently WAKE
[79828.596114] brcmfmac: brcmf_sdio_clkctl Enter
[79828.596117] brcmfmac: brcmf_sdio_bus_sleep new state WAKE
[79828.596121] brcmfmac: brcmf_sdio_bus_sleep Exit: err=0
[79828.596126] brcmfmac: brcmf_sdio_readframes Enter
[79828.596133] brcmfmac: brcmf_sdiod_recv_pkt addr = 0x18000000, size = 64
[79828.609242] brcmfmac: brcmf_sdio_bus_sleep Enter: request WAKE currently WAKE
[79828.609263] brcmfmac: brcmf_sdio_clkctl Enter
[79828.609267] brcmfmac: brcmf_sdio_bus_sleep new state WAKE
[79828.609271] brcmfmac: brcmf_sdio_bus_sleep Exit: err=0
[79828.609344] brcmfmac: brcmf_sdiod_ramrw read 24 bytes at offset 0x00005ec4 in window 0x00258000
[79828.620955] brcmfmac: brcmf_sdio_isr Enter
[79828.621085] brcmfmac: brcmf_sdio_dpc Enter
[79828.621095] brcmfmac: brcmf_sdio_bus_sleep Enter: request WAKE currently WAKE
[79828.621101] brcmfmac: brcmf_sdio_clkctl Enter
[79828.621106] brcmfmac: brcmf_sdio_bus_sleep new state WAKE
[79828.621110] brcmfmac: brcmf_sdio_bus_sleep Exit: err=0
[79828.621118] brcmfmac: brcmf_sdio_readframes Enter
[79828.621128] brcmfmac: brcmf_sdiod_recv_pkt addr = 0x18000000, size = 64
[79828.621166] brcmfmac: brcmf_sdiod_recv_pkt addr = 0x18000000, size = 0
[79828.621195] brcmfmac: brcmf_sdiod_recv_pkt addr = 0x18000000, size = 64
[79828.621244] brcmfmac: brcmf_sdiod_recv_pkt addr = 0x18000000, size = 0
[... 1548 lines cut ...]
[79829.403580] brcmfmac: brcmf_fweh_register event handler registered for ACTION_FRAME_COMPLETE
[79829.403584] brcmfmac: brcmf_fweh_register event handler registered for ACTION_FRAME_OFF_CHAN_COMPLETE
[79829.403588] brcmfmac: brcmf_fweh_register event handler registered for PSK_SUP
[79829.403592] brcmfmac: brcmf_fweh_register event handler registered for RSSI
[79829.403597] brcmfmac: brcmf_fweh_register event handler registered for ESCAN_RESULT
[79829.403603] brcmfmac: brcmf_sdio_bus_txctl Enter
[79829.403615] brcmfmac: brcmf_sdio_dpc Enter
[79829.403623] brcmfmac: brcmf_sdio_bus_sleep Enter: request WAKE currently WAKE
[79829.403627] brcmfmac: brcmf_sdio_clkctl Enter
[79829.403631] brcmfmac: brcmf_sdio_bus_sleep new state WAKE
[79829.403635] brcmfmac: brcmf_sdio_bus_sleep Exit: err=0
[79829.403643] brcmfmac: brcmf_sdio_tx_ctrlframe Enter
[79829.403694] brcmfmac: brcmf_sdio_bus_txctl ctrl_frame complete, err=0
[79829.403699] brcmfmac: brcmf_sdio_bus_rxctl Enter
[79829.403741] brcmfmac: brcmf_sdio_isr Enter
[79829.403803] brcmfmac: brcmf_sdio_dpc Enter
[79829.403809] brcmfmac: brcmf_sdio_bus_sleep Enter: request WAKE currently WAKE
[79829.403814] brcmfmac: brcmf_sdio_clkctl Enter
[79829.403817] brcmfmac: brcmf_sdio_bus_sleep new state WAKE
[79829.403821] brcmfmac: brcmf_sdio_bus_sleep Exit: err=0
[79829.403827] brcmfmac: brcmf_sdio_readframes Enter
[79829.403834] brcmfmac: brcmf_sdiod_recv_pkt addr = 0x18000000, size = 64
```

Took well over 10 minutes, instead of about 3 minutes yesterday.

DrSchottky, 2024-02-02:
@jlinktu isn't there an UaF in your code?

jlinktu, 2024-02-02:
What is an UaF?

DrSchottky, 2024-02-02:
If scb is null p memory is being freed, but it's still passed to wlc_sendctl. Shouldn't sendframe immediately return after calling pkt_buf_free_skb?

jlinktu, 2024-02-02:
Ah sorry, brainlag, use after free..
Yes, you are right, was meant to be a return

jayofelony, 2024-02-02:
```
[85773.086903] brcmfmac: brcmf_sdio_bus_sleep Enter: request WAKE currently WAKE
[85773.086909] brcmfmac: brcmf_sdio_clkctl Enter
[85773.086913] brcmfmac: brcmf_sdio_bus_sleep new state WAKE
[85773.086917] brcmfmac: brcmf_sdio_bus_sleep Exit: err=0
[85773.086968] brcmfmac: brcmf_sdiod_ramrw read 4 bytes at offset 0x00007ffc in window 0x00258000
[85773.087005] brcmfmac: brcmf_sdio_readshared: invalid sdpcm_shared address 0xFE4B01B4
[85773.087010] brcmfmac: brcmf_sdio_readshared: unable to obtain sdpcm_shared info: rv=-22 (addr=0xfe4b01b4)
[85773.087056] brcmfmac: brcmf_sdio_firmware_callback failed: dev=mmc1:0001:1, err=-62
[85773.087068] brcmfmac: brcmf_ops_sdio_probe Enter
[85773.087073] brcmfmac: brcmf_ops_sdio_probe Class=2
[85773.087077] brcmfmac: brcmf_ops_sdio_probe sdio vendor ID: 0x02d0
[85773.087081] brcmfmac: brcmf_ops_sdio_probe sdio device ID: 0xa9a6
```

get this error straight away after running `make install-firmware`, wlan0 won't even come up.

jlinktu, 2024-02-02:
I will try to replicate your setup tomorrow to debug that, otherwise this creates too much overhead.
Will let you know if I need you to do something.

jayofelony, 2024-02-02:
https://github.com/jayofelony/pwnagotchi-bookworm/releases/download/v2.7.9/pwnagotchi-rpi-bookworm-2.7.9-arm64.img.xz

You can use this image. It's exactly what I use.

DrSchottky, 2024-02-02:
This can be used to reproduce the issue, Irun it for 30s and got the crash
`from scapy.all import *

dot11 = Dot11(addr1="00:ae:fa:81:e2:5e", addr2="e8:94:f6:c4:97:3f", addr3="e8:94:f6:c4:97:3f")
packet = RadioTap()/dot11/Dot11Deauth(reason=7)
sendp(packet,iface="mon0", loop=1)`

jayofelony, 2024-02-10:
Any update on this, need me to test something?

jlinktu, 2024-02-11:
No, just didn't find time to check yet.

jayofelony, 2024-03-13:
Checking back on this. Now that kernel 6.6 is out, is there much difference between the workings of a 64bit firmware vs 32bit? I suspect there will also be new version for the bcm43455c0, bcm434340/bcm43436 chipsets. Would it be an idea to update nexmon to the latest version now that many new version have been released in the meantime?

jlinktu, 2024-03-13:
The firmware blob is loaded onto the WiFi chip's ARM Cortex processor and is therefore independent of the hosts architecture, and also mostly independent of the kernel version (neglecting that driver and firmware need some sort of compatibility).
I am already in the process of adding the firmware that is currently used by Raspberry Pi OS.
@jayofelony As your question has absolutely no relation with this issue, please consider opening a dedicated issue if you have further question in the future.

jayofelony, 2024-03-26:
@jlinktu I noticed today my RPi02w with the bcm43430 also showed firmware halted or crashed with the following output in journalctl:
[image]