seemoo-lab/nexmon #596: Set channel failed: chspec=****

jayofelony, 2024-02-01:
[image]
Now that I know how to set a debugging flag, I thought I let it go beserk on my RPi Zero 2W running nexmon. And this error comes by a lot, and I mean **A LOT**. It does not seem to affect injection at all. Does not crash the firmware either. Maybe it is an easy fix, maybe it does not need a fix at all if you think so. But I thought I'd let you know.

Device:
`RPi Zero 2W`

OS:
`Linux jayofelony02w 6.1.0-rpi7-rpi-v8 #1 SMP PREEMPT Debian 1:6.1.63-1+rpt1 (2023-11-24) aarch64 GNU/Linux`
RaspiOS 64 bit lite bookworm

WiFi chip:
`bcm43430/1`

Other info:
I know there is a RPi Zero 2W with a bcm43430/2 WiFi chip that uses the 43436s-sdio.bin but I do not have that one to test sadly. If I do find one that has this chip I will ask them to run a debug on it to see how that one behaves.

jlinktu, 2024-02-02:
For an explanation of the error code `-110`, see https://github.com/seemoo-lab/nexmon/issues/573#issuecomment-1707841926

`chspec=4107` is the channel that is supposed to be set, where `4107`(`0x100b`) means the 20MHz channel number 11 in the 2.4GHz band. Which is a legit channel.

But I don't know why this timeout happens.

Btw. if you want to have the most detailed log possible, you can use `debug=0x1ffffe`. But be aware that this might result in your console being spammed with messages.

jayofelony, 2024-02-02:
When I have done the bcm43455c0 patch and gotten an output for you I will try that debug flag.

jayofelony, 2024-02-02:
```
[29264.278671] brcmfmac: brcmf_sdio_bus_sleep new state WAKE
[29264.278682] brcmfmac: brcmf_sdio_bus_sleep Exit: err=0
[29264.278705] brcmfmac: brcmf_sdiod_ramrw read 24 bytes at offset 0x00005ec4 in window 0x00068000
[29264.279521] brcmfmac: brcmf_sdio_bus_txctl ctrl_frame timeout
[29264.279541] brcmfmac: brcmf_fil_cmd_data Failed: error=-110
[29264.279557] brcmfmac: brcmf_cfg80211_nexmon_set_channel: Set Channel failed: chspec=4101, -110
[29264.295494] brcmfmac: brcmf_sdio_bus_watchdog Enter
[29264.295534] brcmfmac: brcmf_sdio_bus_sleep Enter: request WAKE currently WAKE
[29264.295548] brcmfmac: brcmf_sdio_clkctl Enter
[29264.295559] brcmfmac: brcmf_sdio_bus_sleep new state WAKE
[29264.295570] brcmfmac: brcmf_sdio_bus_sleep Exit: err=0
[29264.295592] brcmfmac: brcmf_sdiod_ramrw read 24 bytes at offset 0x00005ec4 in window 0x00068000
[29264.295669] brcmfmac: brcmf_sdio_bus_watchdog idle
[29264.295689] brcmfmac: brcmf_sdio_bus_sleep Enter: request SLEEP currently WAKE
```
@jlinktu, this log was made with the debug flag you mentioned.

jayofelony, 2024-02-02:
Ok, I will try and create a delay in the software. I will close this issue.

closed 2024-02-02 by jayofelony

jlinktu, 2024-02-02:
Delaying might be an option.
Another way could be to read back the channel value and try to set it again until you get back the right one.

jayofelony, 2024-02-03:
I will do both, as a precaution.