seemoo-lab/nexmon #599: unable to compile Raspberry Pi 5 kernel 6.6.y

renewedbullet, 2024-02-15:
Unable to compile manually on 6.6.y kernel, below is the error I get, kernel 6.6.y has passed all RPI requirements https://github.com/raspberrypi/linux

Is this something that can be fixed on my end or requires and update from nexmon?

root@raspberrypi:/home/pi/nexmon/patches/bcm43455c0/7_45_206/nexmon# make

          ###########   ###########   ##########    ##########
         ############  ############  ############  ############
         ##            ##            ##   ##   ##  ##        ##
         ##            ##            ##   ##   ##  ##        ##
         ###########   ####  ######  ##   ##   ##  ##    ######
          ###########  ####  #       ##   ##   ##  ##    #    #
                   ##  ##    ######  ##   ##   ##  ##    #    #
                   ##  ##    #       ##   ##   ##  ##    #    #
         ############  ##### ######  ##   ##   ##  ##### ######
         ###########    ###########  ##   ##   ##   ##########

            S E C U R E   M O B I L E   N E T W O R K I N G

                               presents:

              # ###   ###  #   # # ###  ###   ###  # ###
              ##   # #   #  # #  ##   ##   # #   # ##   #
              #    # #####   #   #    #    # #   # #    #
              #    # #      # #  #    #    # #   # #    #
              #    #  #### #   # #    #    #  ###  #    #

                The C-based Firmware Patching Framework

                           !!! WARNING !!!
    Our software may damage your hardware and may void your hardware’s
     warranty! You use our tools at your own risk and responsibility

  COLLECTING STATISTICS read /home/pi/nexmon/STATISTICS.md for more information
  COMPILING src/version.c => obj/version.o (details: log/compiler.log)
  GENERATING LINKER FILE gen/nexmon.pre => gen/nexmon.ld
  GENERATING LINKER FILE gen/nexmon.pre => gen/flashpatches.ld
  LINKING OBJECTS => gen/patch.elf (details: log/linker.log, log/linker.err)
  GENERATING MAKE FILE gen/nexmon.pre => gen/nexmon.mk
  GENERATING MAKE FILE gen/nexmon.pre => gen/flashpatches.mk
  APPLYING FLASHPATCHES gen/flashpatches.mk => brcmfmac43455-sdio.bin (details: log/flashpatches.log)
  APPLYING PATCHES gen/nexmon.mk => brcmfmac43455-sdio.bin (details: log/patches.log)
Makefile:59: Warning: Driver for kernel 6.6 not found

jlinktu, 2024-02-16:
Kernel 6.6 is not supported yet. Check [https://github.com/seemoo-lab/nexmon/tree/master/patches/driver](nexmon/patches/driver) directory for supported versions.

renewedbullet, 2024-02-16:
Can you guys patch it to work with 6.6.y for Raspberry OS since they have locked in their kernel 6.6.y? That is the question I'm asking I know it's not supported :)

jlinktu, 2024-02-16:
Yes, we can do that. But it is something that I have to queue for later.

@DrSchottky usually is quite fast in adding new version though. Are you already doing so?

DrSchottky, 2024-02-16:
Oh, I didn't know they had already moved to version 6.6. I'll try to find some time later :)

renewedbullet, 2024-02-16:
realtek driver supports a few other fixes. Is there anything I can provide to help with this?

renewedbullet, 2024-03-14:
@jlinktu now RPI has officially released 6.6 is it possible or do we need to wait for @DrSchottky ?

DrSchottky, 2024-03-14:
6.2 nexmon brmfmac works without any changes on 6.6, you just need to compile it with -DDEBUG and unload brcmfmac_wcc too before loading the .ko
I'll update my fork in the next few days

renewedbullet, 2024-03-14:
<3 @DrSchottky

closed 2024-03-16 by jlinktu