seemoo-lab/nexmon #600: unable to compile for RPI4 aarch64 kernel 5.15

nightmarepuma, 2024-02-23:
I am new to this but I am trying to follow the readme instructions to compile manually for my RPI4 5.15.84-v8+, but I am getting the following error: 
`BUILDING DRIVER for kernel 5.15 brcmfmac_5.15.y-nexmon/brcmfmac.ko (details: log/driver.log)
ERROR: modpost: "dmi_first_match" [/opt/nexmon/patches/driver/brcmfmac_5.15.y-nexmon/brcmfmac.ko] undefined!
ERROR: modpost: "_raw_spin_unlock" [/opt/nexmon/patches/driver/brcmfmac_5.15.y-nexmon/brcmfmac.ko] undefined!
ERROR: modpost: "__warn_printk" [/opt/nexmon/patches/driver/brcmfmac_5.15.y-nexmon/brcmfmac.ko] undefined!
ERROR: modpost: "__const_udelay" [/opt/nexmon/patches/driver/brcmfmac_5.15.y-nexmon/brcmfmac.ko] undefined!
ERROR: modpost: "efivar_entry_size" [/opt/nexmon/patches/driver/brcmfmac_5.15.y-nexmon/brcmfmac.ko] undefined!
ERROR: modpost: "efivar_entry_get" [/opt/nexmon/patches/driver/brcmfmac_5.15.y-nexmon/brcmfmac.ko] undefined!
ERROR: modpost: "cpu_hwcap_keys" [/opt/nexmon/patches/driver/brcmfmac_5.15.y-nexmon/brcmfmac.ko] undefined!
ERROR: modpost: "arm64_const_caps_ready" [/opt/nexmon/patches/driver/brcmfmac_5.15.y-nexmon/brcmfmac.ko] undefined!
ERROR: modpost: "dmi_get_system_info" [/opt/nexmon/patches/driver/brcmfmac_5.15.y-nexmon/brcmfmac.ko] undefined!
make[2]: *** [scripts/Makefile.modpost:133: /opt/nexmon/patches/driver/brcmfmac_5.15.y-nexmon/Module.symvers] Error 1
make[2]: *** Deleting file '/opt/nexmon/patches/driver/brcmfmac_5.15.y-nexmon/Module.symvers'
make[1]: *** [Makefile:1817: modules] Error 2
make: *** [Makefile:54: brcmfmac.ko] Error 2`

I have tried compiling for different firmware versions and got the same error. Do you have any idea how could this be fixed?

**uname -rm**
5.15.84-v8+ aarch64`

nightmarepuma, 2024-03-13:
@DrSchottky do you know how to solve this problem? Here are the commands I ran:
sudo su
apt-get update
sudo apt install raspberrypi-kernel-headers git libgmp3-dev gawk qpdf bison flex make autoconf libtool texinfo
wget https://raw.githubusercontent.com/RPi-Distro/rpi-source/master/rpi-source -O /usr/local/bin/rpi-source && chmod +x /usr/local/bin/rpi-source && /usr/local/bin/rpi-source -q --tag-update
rpi-source
git clone https://github.com/seemoo-lab/nexmon.git
cd nexmon
sudo dpkg --add-architecture armhf
sudo apt-get update
sudo apt-get install libc6:armhf libisl23:armhf libmpfr6:armhf libmpc3:armhf libstdc++6:armhf
sudo ln -s /usr/lib/arm-linux-gnueabihf/libisl.so.23.0.0 /usr/lib/arm-linux-gnueabihf/libisl.so.10
sudo ln -s /usr/lib/arm-linux-gnueabihf/libmpfr.so.6.1.0 /usr/lib/arm-linux-gnueabihf/libmpfr.so.4
source setup_env.sh
make
cd patches/bcm43455c0/7_45_189/nexmon/
make

From that last make, I get the error above.

DrSchottky, 2024-03-13:
Can you try installing `raspberrypi-kernel-headers` before running `apt update` to get headers matching your kernel? 
Or you can grab them from here http://archive.raspberrypi.org/debian/pool/main/r/raspberrypi-firmware/

nightmarepuma, 2024-03-14:
The manual download worked for me. Thank you so much

closed 2024-03-14 by nightmarepuma