seemoo-lab/nexmon #601: BPi M2 Zero installation

jayofelony, 2024-02-29:
I had the idea of trying Nexmon on the Banana Pi M2 Zero:
```
sudo dmesg | grep brcmf
[   69.044970] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43430-sdio for chip BCM43430/1
[   69.052213] brcmfmac mmc1:0001:1: Direct firmware load for brcm/brcmfmac43430-sdio.sinovoip,bpi-m2-zero.bin failed with error -2
[   69.052280] brcmfmac mmc1:0001:1: Falling back to sysfs fallback for: brcm/brcmfmac43430-sdio.sinovoip,bpi-m2-zero.bin
[   69.529424] brcmfmac: brcmf_c_process_txcap_blob: no txcap_blob available (err=-2)
[   69.530453] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM43430/1 wl0: Mar 30 2021 01:12:21 version 7.45.98.118 (7d96287 CY) FWID 01-32059766
```

[image]
I tried both versions of nexmon, both error with the same message. Has this been tried to install before, any things needs to be changed to install it on the BPi M2 Zero?

jlinktu, 2024-02-29:
The modified brcmfmac drivers in the nexmon repo are specific to Raspberry Pi. You could fetch the driver for whatever OS you are running on the Banana Pi and port the driver modifications to it.

jayofelony, 2024-02-29:
If only I knew how to do that haha. Than I certainly would do it!

jlinktu, 2024-03-01:
You could look at the differences between the original Raspberry Pi brcmfmac driver and the modified one in the nexmon repo, e.g. using `diff`. Then see where you can introduce the same modifications in the driver of your OS.

jayofelony, 2024-03-01:
I tried.. told me they differ haha I am no good at this. I have absolutely no idea what I am doing, so far out of my league.

closed 2024-03-07 by jayofelony