seemoo-lab/nexmon #606: RPi02W: Firmware halted or crashed

jayofelony, 2024-03-26:
[image]
This is a screen taken from journalctl today
```
uname -a
Linux jayofelony 6.6.20+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.6.20-1+rpt1 (2024-03-07) aarch64 GNU/Linux

Firmware: BCM43430/1 wl0: Aug  7 2017 00:46:29 version 7.45.41.46 (r666254 CY) FWID 01-f8a78378
```

jlinktu, 2024-03-26:
Unfortunately, the trap info only tells that the chip crashed while gathering debug info about the programmable state machine of the D11 core.

The crash might not be related to nexmon as the default patch doesn't modify the related microcode and doesn't directly access the D11 cores registers after initalization.

If you can get this reproducable, there might be a chance of locating the issue, but even then, this would probably require more time than I can invest right now.

jayofelony, 2024-03-26:
I was scrolling through 3 days of data in journalctl and this was the only occurrence. But I figured I tell you either way. I can catch the error and have brcmfmac reloaded or something else that works if needed.

closed 2024-03-28 by jlinktu