seemoo-lab/nexmon #607: BCM43430: firmware has halted or crashed

jayofelony, 2024-06-03:
Hello once more,
`uname -a`:
`Linux jayofelony 6.6.31+rpt-rpi-v6 #1 Raspbian 1:6.6.31-1+rpt1 (2024-05-29) armv6l GNU/Linux`

I ran into this issue after which the monitor mode quits, this is a copy/paste from `journalctl -xe`. Patched using the bcm43430/7_45_41_46 on a RPi 0W running latest Bookworm raspios kernel 6.6.y. No rush or anything, not even sure if this is the info you need. If you need more info let me know.

```
Jun 03 20:29:30 jayofelony kernel: ieee80211 phy0: brcmf_fw_crashed: Firmware has halted or crashed
Jun 03 20:29:31 jayofelony kernel: ieee80211 phy0: brcmf_fil_cmd_data: bus is down. we have nothing to do.
Jun 03 20:29:31 jayofelony kernel: ieee80211 phy0: brcmf_net_mon_stop: BRCMF_C_SET_MONITOR error (-5)
Jun 03 20:29:31 jayofelony kernel: brcmfmac mmc1:0001:1 wlan0mon (unregistering): left promiscuous mode
Jun 03 20:29:31 jayofelony kernel: 8<--- cut here ---
Jun 03 20:29:31 jayofelony kernel: Unable to handle kernel NULL pointer dereference at virtual address 00000004 when read
Jun 03 20:29:31 jayofelony kernel: [00000004] *pgd=00000000
Jun 03 20:29:31 jayofelony kernel: Internal error: Oops: 5 [#1] ARM
Jun 03 20:29:31 jayofelony kernel: Modules linked in: lz4 lz4_compress zram zsmalloc vc4 raspberrypi_hwmon snd_soc_hdmi_codec drm_display_helper cec drm_dma_helper drm_kms_helper snd_soc_core brcmfmac(O) b>
Jun 03 20:29:31 jayofelony kernel: CPU: 0 PID: 562 Comm: kworker/u3:2 Tainted: G         C O       6.6.31+rpt-rpi-v6 #1  Raspbian 1:6.6.31-1+rpt1
Jun 03 20:29:31 jayofelony kernel: Hardware name: BCM2835
Jun 03 20:29:31 jayofelony kernel: Workqueue: brcmf_wq/mmc1:0001:1 brcmf_sdio_dataworker [brcmfmac]
Jun 03 20:29:31 jayofelony kernel: PC is at brcmf_rx_hdrpull+0x14/0xa0 [brcmfmac]
Jun 03 20:29:31 jayofelony kernel: LR is at brcmf_rx_frame+0x68/0x168 [brcmfmac]
Jun 03 20:29:31 jayofelony kernel: pc : [<bf524be8>]    lr : [<bf52586c>]    psr: 60000013
Jun 03 20:29:31 jayofelony kernel: sp : df9c9e18  ip : bf5553c8  fp : c3ad3610
Jun 03 20:29:31 jayofelony kernel: r10: 00000040  r9 : 00000200  r8 : 00000600
Jun 03 20:29:31 jayofelony kernel: r7 : 00000000  r6 : 00000000  r5 : 00000000  r4 : c36710c0
Jun 03 20:29:31 jayofelony kernel: r3 : 00000000  r2 : df9c9e30  r1 : c36710c0  r0 : 00000000
Jun 03 20:29:31 jayofelony kernel: Flags: nZCv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment user
Jun 03 20:29:31 jayofelony kernel: Control: 00c5387d  Table: 0320c008  DAC: 00000055
Jun 03 20:29:31 jayofelony kernel: Register r0 information: NULL pointer
Jun 03 20:29:31 jayofelony kernel: Register r1 information: slab skbuff_head_cache start c36710c0 pointer offset 0 size 184
Jun 03 20:29:31 jayofelony kernel: Register r2 information: 2-page vmalloc region starting at 0xdf9c8000 allocated at kernel_clone+0xa0/0x32c
Jun 03 20:29:31 jayofelony kernel: Register r3 information: NULL pointer
Jun 03 20:29:31 jayofelony kernel: Register r4 information: slab skbuff_head_cache start c36710c0 pointer offset 0 size 184
Jun 03 20:29:31 jayofelony kernel: Register r5 information: NULL pointer
Jun 03 20:29:31 jayofelony kernel: Register r6 information: NULL pointer
Jun 03 20:29:31 jayofelony kernel: Register r7 information: NULL pointer
Jun 03 20:29:31 jayofelony kernel: Register r8 information: non-paged memory
Jun 03 20:29:31 jayofelony kernel: Register r9 information: non-paged memory
Jun 03 20:29:32 jayofelony kernel: Register r10 information: non-paged memory
Jun 03 20:29:32 jayofelony kernel: Register r11 information: slab kmalloc-1k start c3ad3400 pointer offset 528 size 1024
Jun 03 20:29:32 jayofelony kernel: Register r12 information: 15-page vmalloc region starting at 0xbf54c000 allocated at load_module+0x6c4/0x1bb4
Jun 03 20:29:32 jayofelony kernel: Process kworker/u3:2 (pid: 562, stack limit = 0x6ab42f3f)
Jun 03 20:29:32 jayofelony kernel: Stack: (0xdf9c9e18 to 0xdf9ca000)
Jun 03 20:29:32 jayofelony kernel: 9e00:                                                       c36710c0 00000000
Jun 03 20:29:32 jayofelony kernel: 9e20: 00000000 bf52586c c36710c0 00000600 00000000 4d716be9 0000000d bf53d268
Jun 03 20:29:32 jayofelony kernel: 9e40: c36710c0 c3be97c0 00000002 bf5335ec c3093198 c0056874 000162b9 df9c9e78
Jun 03 20:29:32 jayofelony kernel: 9e60: 0000007d c053b9ec c0d9b5f8 c3ad3400 0000007d 00000030 c3ad3410 4d716be9
Jun 03 20:29:32 jayofelony kernel: 9e80: c3093198 00000040 00000032 00000014 00000000 c3093000 c3ad3536 df9c9eb4
Jun 03 20:29:32 jayofelony kernel: 9ea0: 00000000 c0d9b5f8 c0d9ba80 c0ebf054 c0776cd8 055c0221 00000000 0000000c
Jun 03 20:29:32 jayofelony kernel: 9ec0: 00000000 c0045f18 ffffffff 00000401 0006fdd0 00000000 00000000 00000000
Jun 03 20:29:32 jayofelony kernel: 9ee0: 0006debc 00000000 00000000 00000000 00000000 00000000 00000000 00000000
Jun 03 20:29:32 jayofelony kernel: 9f00: 00000000 00000000 00000000 4d716be9 c09c80f0 c24f8280 c3ad3610 00000060
Jun 03 20:29:32 jayofelony kernel: 9f20: c1111b00 c1111b05 c3ad3614 c3093000 00000000 c003dc2c c1bfea1c 61c88647
Jun 03 20:29:32 jayofelony kernel: 9f40: c24f82ac c24f8280 c1bfea00 c0e01a40 c1bfea1c 61c88647 c24f82ac c003e0cc
Jun 03 20:29:32 jayofelony kernel: 9f60: c24f8280 c300fc00 df9bded8 c3a26cc0 c3093000 c003de20 c24f8280 c300fc00
Jun 03 20:29:32 jayofelony kernel: 9f80: df9bded8 00000000 00000000 c0044c40 c3a26cc0 c0044b74 00000000 00000000
Jun 03 20:29:32 jayofelony kernel: 9fa0: 00000000 00000000 00000000 c000833c 00000000 00000000 00000000 00000000
Jun 03 20:29:32 jayofelony kernel: 9fc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
Jun 03 20:29:32 jayofelony kernel: 9fe0: 00000000 00000000 00000000 00000000 00000013 00000000 00000000 00000000
Jun 03 20:29:32 jayofelony kernel:  brcmf_rx_hdrpull [brcmfmac] from brcmf_rx_frame+0x68/0x168 [brcmfmac]
Jun 03 20:29:32 jayofelony kernel:  brcmf_rx_frame [brcmfmac] from brcmf_sdio_dataworker+0xf48/0x25d8 [brcmfmac]
Jun 03 20:29:32 jayofelony kernel:  brcmf_sdio_dataworker [brcmfmac] from process_one_work+0x164/0x358
Jun 03 20:29:32 jayofelony kernel:  process_one_work from worker_thread+0x2ac/0x4ec
Jun 03 20:29:32 jayofelony kernel:  worker_thread from kthread+0xcc/0xf0
Jun 03 20:29:32 jayofelony kernel:  kthread from ret_from_fork+0x14/0x38
Jun 03 20:29:32 jayofelony kernel: Exception stack(0xdf9c9fb0 to 0xdf9c9ff8)
Jun 03 20:29:32 jayofelony kernel: 9fa0:                                     00000000 00000000 00000000 00000000
Jun 03 20:29:32 jayofelony kernel: 9fc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
Jun 03 20:29:32 jayofelony kernel: 9fe0: 00000000 00000000 00000000 00000000 00000013 00000000
Jun 03 20:29:32 jayofelony kernel: Code: e52de004 e28dd004 e3a03000 e5823000 (e5903004) 
Jun 03 20:29:32 jayofelony kernel: ---[ end trace 0000000000000000 ]---
```

DrSchottky, 2024-06-03:
That's a driver bug (sdio workqueue still procession data after a detach) but it's triggered by a module crash, non included in the log.
Not very useful.

jayofelony, 2024-06-04:
Okay, thanks for the clarification so far. I will wait for Matthias to respond if he needs anything else from me.

jayofelony, 2024-06-05:
```
[ 9266.052038] ieee80211 phy0: brcmf_fw_crashed: Firmware has halted or crashed
[ 9266.101368] ieee80211 phy0: brcmf_fil_cmd_data: bus is down. we have nothing to do.
[ 9266.101410] ieee80211 phy0: brcmf_net_mon_stop: BRCMF_C_SET_MONITOR error (-5)
[ 9266.102043] brcmfmac mmc1:0001:1 wlan0mon (unregistering): left promiscuous mode
[ 9266.104435] brcmfmac: brcmf_sdio_dpc: failed backplane access over SDIO, halting operation
[ 9266.155206] mmc1: card 0001 removed
[ 9266.200011] mmc1: queuing unknown CIS tuple 0x41 [d0 02 a6 a9] (4 bytes)
[ 9266.202562] mmc1: queuing unknown CIS tuple 0x04 [] (0 bytes)
[ 9266.203349] mmc1: bad CIS tuple 0x20 (0 bytes)
[ 9266.203398] mmc1: error -22 whilst initialising SDIO card
[ 9266.259730] mmc1: queuing unknown CIS tuple 0x41 [d0 02 a6 a9] (4 bytes)
[ 9266.262920] mmc1: queuing unknown CIS tuple 0x04 [] (0 bytes)
[ 9266.263850] mmc1: bad CIS tuple 0x20 (0 bytes)
[ 9266.263897] mmc1: error -22 whilst initialising SDIO card
[ 9266.320354] mmc1: queuing unknown CIS tuple 0x41 [d0 02 a6 a9] (4 bytes)
[ 9266.324935] mmc1: queuing unknown CIS tuple 0x04 [] (0 bytes)
[ 9266.326992] mmc1: bad CIS tuple 0x20 (0 bytes)
[ 9266.327046] mmc1: error -22 whilst initialising SDIO card
[ 9266.399394] mmc1: queuing unknown CIS tuple 0x41 [d0 02 a6 a9] (4 bytes)
[ 9266.409494] mmc1: queuing unknown CIS tuple 0x04 [] (0 bytes)
[ 9266.411932] mmc1: bad CIS tuple 0x20 (0 bytes)
[ 9266.411992] mmc1: error -22 whilst initialising SDIO card
```

This is some extra logs from `dmesg -w` whilst the crash happens. If I know the debug code I can maybe get some more specific data from the driver itself?