seemoo-lab/nexmon #608: Error installing nexmon csi

ayyp1, 2024-06-11:
root@raspberrypi:/home/ayyp/nexmon/patches/bcm43455c0/7_45_189/nexmon_csi# sudo make
Makefile:3: /firmwares/bcm43455c0/7_45_189/definitions.mk: No such file or directory
make: *** No rule to make target '/firmwares/bcm43455c0/7_45_189/definitions.mk'.  Stop.

what should I do ?

jlinktu, 2024-06-11:
The nexmon framework depends on a specific environment that gets set when running `source setup_env.sh`. When you execute `make` with `sudo`, you are changing to a different environment due to `sudo`. Due to that, nexmon does no longer run as expected. Either run `make` without `sudo` or pass your current environment to it by using `sudo -E`.

ayyp1, 2024-06-13:
ayyp@raspberrypi:~/nexmon/patches/bcm43455c0/7_45_189/nexmon_csi $ make install-firmware
Makefile:3: /firmwares/bcm43455c0/7_45_189/definitions.mk: No such file or directory
make: *** No rule to make target '/firmwares/bcm43455c0/7_45_189/definitions.mk'.  Stop.
 but still this error resides

ayyp1, 2024-06-13:
when i run make install-firmware 

COLLECTING STATISTICS read /home/ayyp/nexmon/STATISTICS.md for more information
 COMPILING src/console.c => obj/console.o (details: log/compiler.log)
/bin/sh: 1: /home/ayyp/nexmon/buildtools/gcc-arm-none-eabi-5_4-2016q2-linux-armv7l/bin/arm-none-eabi-gcc: not found
make: *** [Makefile:126: obj/console.o] Error 127

jlinktu, 2024-06-13:
As you are trying to compile for the WiFi chip  that can be found on Raspberry Pis, I assume that you are trying to do it on a Raspberry Pi.
The output you posted tells that you are missing `libmpfr.so.4`.
Follow the README on how to setup nexmon on a Raspberry Pi: http://github.com/seemoo-lab/nexmon?tab=readme-ov-file#build-patches-for-bcm43430a1-on-the-rpi3zero-w-or-bcm434355c0-on-the-rpi3rpi4-or-bcm43436b0-on-the-rpi-zero-2w-using-raspb
It tells you what to do.