seemoo-lab/nexmon #609: gen/patch.elf (details: log/linker.log, log/linker.err) make: *** [Makefile:103: gen/patch.elf] Error 1

Ilfin98, 2024-06-20:
Please help me how to fix this. Below is the end lines at log/linker.err :
/nexmon/buildtools/gcc-arm-none-eabi-5_4-2016q2-linux-armv7l/bin/arm-none-eabi-ld: Removing unused section '.text.wlc_ucodex_write_compressed_args' in file 'obj/ucode_compression_code.o'
obj/ucode_compression_code.o: In function `wlc_ucode_write_compressed':
ucode_compression_code.c:(.text.wlc_ucode_write_compressed+0x4): undefined reference to `ucode_compressed_bin_len'
ucode_compression_code.c:(.text.wlc_ucode_write_compressed+0x8): undefined reference to `ucode_compressed_bin'
ucode_compression_code.c:(.text.wlc_ucode_write_compressed+0xc): undefined reference to `ucode_compressed_bin_len'
ucode_compression_code.c:(.text.wlc_ucode_write_compressed+0x12): undefined reference to `ucode_compressed_bin'

jlinktu, 2024-06-20:
For what chip and firmware version are you trying to build something?
What patch are you trying to build?

jlinktu, 2024-06-20:
Does the answer in issue #146 solve your problem?
(Did you run `make` in the nexmon repository's root directory?)

Ilfin98, 2024-06-21:
> For what chip and firmware version are you trying to build something? What patch are you trying to build?

Bcm4358 chip and 7_112_200_17_sta. I build all patches by run "make" in nexmon/patches/bcm4358/7_112_200_17_sta/nexmon. What sould i do about that linker error? Thanks very much for your answers and helps.

Ilfin98, 2024-06-21:
> Does the answer in issue #146 solve your problem? (Did you run `make` in the nexmon repository's root directory?)

No, i did run make in the root directory of nexmon.

Ilfin98, 2024-06-21:
> > Does the answer in issue #146 solve your problem? (Did you run `make` in the nexmon repository's root directory?)
> 
> No, i did run make in the root directory of nexmon.

I mean that issue #146 not solve my problem, sorry for my bad english.

jlinktu, 2024-06-21:
Just to be clear, with nexmon's root directory I don't mean "`nexmon/patches/bcm4358/7_112_200_17_sta/nexmon`".
And if you ran `make` in the nexmon's root directory, did everything execute without errors?
On what system are you trying to build the patched firmware?