seemoo-lab/nexmon #618: Immediate crashing of firmware

jayofelony, 2024-07-25:
Apparently there is some RPi 02w that uses the brcmfmac43430b0-sdio driver and crashes immediately, any recommendation for this particular case? Looks like a mix between 43430 and 43436 chip I think.

```
[   10.456107] brcmfmac: loading out-of-tree module taints kernel.
[   10.500175] brcmfmac: F1 signature read @0x18000000=0x1542a9a6
[   10.522984] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43430b0-sdio for chip BCM43430/2
[   10.523871] usbcore: registered new interface driver brcmfmac
[   10.944060] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43430b0-sdio for chip BCM43430/2
[   10.944276] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43430b0-sdio for chip BCM43430/2
[   10.944374] brcmfmac: brcmf_c_process_clm_blob: no clm_blob available (err=-2), device may have limited channels available
[   10.945244] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM43430/2 wl0: Mar 26 2024 07:37:54 version 9.88.4.65 (nexmon.org: 2.2.2-529-g5ff4-1) FWID 01-f40f3270
[   11.313772] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled
[   14.453049] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save disabled
[   15.011888] ieee80211 phy0: brcmf_c_set_cur_etheraddr: Setting cur_etheraddr failed, -52
[   15.014925] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save disabled
[   15.018425] brcmfmac: brcmf_vif_add_validate: Attempt to add a MONITOR interface...
[   15.018471] brcmfmac: brcmf_mon_add_vif: brcmf_mon_add_vif called
[   15.018477] brcmfmac: brcmf_mon_add_vif: Adding vif "wlan0mon"
[   31.971052] brcmfmac mmc1:0001:1 wlan0mon: entered promiscuous mode
[   63.290981] ieee80211 phy0: brcmf_fw_crashed: Firmware has halted or crashed
[   63.334169] ieee80211 phy0: brcmf_fil_cmd_data: bus is down. we have nothing to do.
[   63.334197] ieee80211 phy0: brcmf_net_mon_stop: BRCMF_C_SET_MONITOR error (-5)
[   63.334602] brcmfmac mmc1:0001:1 wlan0mon (unregistering): left promiscuous mode
[   70.783265] usbcore: deregistering interface driver brcmfmac
[   72.417553] usbcore: registered new interface driver brcmfmac
[   76.119687] usbcore: deregistering interface driver brcmfmac
[   77.577683] usbcore: registered new interface driver brcmfmac
[   84.660499] usbcore: deregistering interface driver brcmfmac
[   85.280758] usbcore: registered new interface driver brcmfmac
```

jlinktu, 2024-07-26:
The firmware you are using seems to be incompatible with your setup.
1. Find the firmware that Raspberry Pi OS uses by default and verify that it works fine with your setup.
1. Add monitor mode to this firmware.

jayofelony, 2024-07-26:
Well it seems to link to the `/lib/firmware/brcm/brcmfmac43436-sdio.bin` file. Which has been patched with nexmon.

[image]
[image]

So in theory it should work, as the RPi using the 43436b0 chipset work correctly using monitor mode.

jlinktu, 2024-07-26:
Matching file names don't imply matching firmware. Check the firmware version that works in your setup.

jayofelony, 2024-07-26:
```
pi@raspberrypi:~ $ sudo dmesg | grep brcmf
[   10.585321] brcmfmac: F1 signature read @0x18000000=0x1542a9a6
[   10.600550] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43430b0-sdio for chip BCM43430/2
[   10.603142] usbcore: registered new interface driver brcmfmac
[   10.830737] brcmfmac: brcmf_c_process_txcap_blob: no txcap_blob available (err=-2)
[   10.831462] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM43430/2 wl0: Mar 31 2022 17:24:51 version 9.88.4.77 (g58bc5cc) FWID 01-3b307371
[   16.177589] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled
```

I do not own this hardware myself, I get these logs through discord from one my users. This is dmesg log from current default raspios.

jlinktu, 2024-07-26:
The log tells that the loaded firmware is version `9.88.4.77 (g58bc5cc)`, whilst the nexmon version is `9.88.4.65`.

jayofelony, 2024-07-26:
Okay, so you think or are saying that this type of wifi chip will not work with the current nexmon firmware?

jlinktu, 2024-07-27:
No, not the chip. The chip and firmware are compatible, but you also have to consider the driver.

jayofelony, 2024-07-28:
How can I fix that, or can't I?