seemoo-lab/nexmon #619: RPI ZERO 2W: don't works

qrp73, 2024-07-25:
New RPI ZERO 2W uses chip BCM43430/1, it using brcm/brcmfmac43436s-sdio firmware. 
It appears that there is no patch for this firmware...

I tried to rename patched brcmfmac43436-sdio to brcmfmac43436s-sdio, but it failed to load with error:
```
brcmfmac mmc1:0001:1: Direct firmware load for brcm/brcmfmac43436s-sdio.raspberrypi,model-zero-2-w.bin failed with error -2
```

jlinktu, 2024-07-26:
We are not related to Raspberry Pi nor Kali Linux - but if you want to get this to work I suggest you to first get Kali Linux to correctly load the original firmware. Once this works, you can think about adding monitor mode.

qrp73, 2024-07-26:
I just want to get working monitor mode, since it don't works on raspi-os, this is why I installed Kali Linux.

Original raspi-os-bookworm 6.6.31+rpt-rpi-v8  running on rpi2w uses this symlink ```/lib/firmware/brcm/brcmfmac43430-sdio.bin -> ../cypress/cyfmac43430-sdio.bin```

Here is log file from raspi-os-bookworm:
```
[   12.919275] brcmfmac: F1 signature read @0x18000000=0x1541a9a6
[   12.936804] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43430-sdio for chip BCM43430/1
[   12.937728] usbcore: registered new interface driver brcmfmac
[   13.192057] brcmfmac: brcmf_c_process_txcap_blob: no txcap_blob available (err=-2)
[   13.192698] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM43430/1 wl0: Jun 14 2023 07:27:45 version 7.45.96.s1 (gf031a129) FWID 01-70bd2af7 es7
```

jlinktu, 2024-07-26:
Well, if it doesn't matter what OS you are using, I suggest to go for Raspberry Pi OS then.

From the log output I can see that on Raspberry Pi OS, other than you suggested, [`brcmfmac43436s.bin`](https://github.com/RPi-Distro/firmware-nonfree/blob/bookworm/debian/config/brcm80211/brcm/brcmfmac43436s-sdio.bin) is loaded.

Regarding the confusion about what file is to be loaded, there are a couple of discussions there: https://github.com/RPi-Distro/firmware-nonfree/issues
TLDR: There are different versions of the Raspberry Pi Zero 2 W, with different Wi-Fi chips, which require different firmwares.
Apparently you have the version that requires [`brcmfmac43436s.bin`](https://github.com/RPi-Distro/firmware-nonfree/blob/bookworm/debian/config/brcm80211/brcm/brcmfmac43436s-sdio.bin). Kali simply seems to load the wrong firmware, thus, again suggesting to go with Raspberry Pi OS or port the related stuff to Kali.

However, all of the above has nothing to do with `nexmon` itself.
It is possible to patch [`brcmfmac43436s.bin`](https://github.com/RPi-Distro/firmware-nonfree/blob/bookworm/debian/config/brcm80211/brcm/brcmfmac43436s-sdio.bin) to support monitor mode, but I currently can't do this in my free-time. If you really need this, you can contact me by mail and we can try to find a solution.

qrp73, 2024-07-26:
> Apparently you have the version that requires [`brcmfmac43436s.bin`](https://github.com/RPi-Distro/firmware-nonfree/blob/bookworm/debian/config/brcm80211/brcm/brcmfmac43436s-sdio.bin). Kali simply [...]

Why brcmfmac43436s-sdio??? 

When I boot into original Raspi OS with working wifi firmware it shows in the log that the chip is ```BCM43430/1``` and load firmware from brcmfmac43430-sdio which is symlink to ```/lib/firmware/cypress/cyfmac43430-sdio.bin``` and this firmware version is 7.45.96.s1 (gf031a129) FWID 01-70bd2af7 es7
 

> However, all of the above has nothing to do with `nexmon` itself. It is possible to patch [...]

Currently I'm trying to setup clean Raspi-OS Lite and make the patch. If I understand correctly there is needs to build ```patches/bcm43430a1/7_45_41_46/nexmon/``` and then try to replace original firmware with version 7.45.96.s1 at ```/lib/firmware/cypress/cyfmac43430-sdio.bin```. Is it correct?

jlinktu, 2024-07-26:
The log says:
`brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM43430/1 wl0: Jun 14 2023 07:27:45 version 7.45.96.s1 (gf031a129) FWID 01-70bd2af7 es7`, which is the versioning string of [brcmfmac43436s-sdio.bin](https://github.com/RPi-Distro/firmware-nonfree/blob/bookworm/debian%2Fconfig%2Fbrcm80211%2Fbrcm%2Fbrcmfmac43436s-sdio.bin).

qrp73, 2024-07-26:
just checked, yes - ```brcmfmac43436s-sdio.bin``` contains version string ```7.45.96.s1 (gf031a129)``` 

> It is possible to patch [`brcmfmac43436s.bin`](https://github.com/RPi-Distro/firmware-nonfree/blob/bookworm/debian/config/brcm80211/brcm/brcmfmac43436s-sdio.bin) to support monitor mode, but I [...]

Thanks , I wrote you mail about it.

alexzaporozhets, 2024-08-25:
@qrp73, @jlinktu any updates on this issue?

qrp73, 2024-08-30:
@alexzaporozhets no, adding patch for 7.45.96.s1 is a paid service. 

But I found that old firmware BCM43430/1 version 7.45.41.46 (r666254 CY) which was supplied for RPI3 also works for RPI02W. And there is a patch in nexmon for this firmware version. You can found original this firmware version in this package: http://archive.raspberrypi.org/debian/pool/main/f/firmware-nonfree/firmware-brcm80211_0.43+rpi6_all.deb

rudyrdx, 2024-09-15:
> @alexzaporozhets no, adding patch for 7.45.96.s1 is a paid service.
> 
> But I found old firmware BCM43430/1 version 7.45.41.46 (r666254 CY) which was supplied for RPI3 also works for RPI02W. And there is a patch in nexmon for this firmware version.
> 
> You can found original 7.45.41.46 firmware version in this package: http://archive.raspberrypi.org/debian/pool/main/f/firmware-nonfree/firmware-brcm80211_0.43+rpi6_all.deb
> 
> This firmware has some bug which sometimes may lead to not responding state. But this is better than nothing.

Any steps or guide on how to execute it?

Xendr1k, 2024-10-12:
Any updates? I have the same problem with the same version of firmware.
dmesg | grep "Firmware: BCM43430"
[   10.316434] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM43430/1 wl0: Feb  5 2021 12:49:07 version 7.45.96 (r745790) FWI     D 01-71817851 es7.
I would like to be able to make it work in monitor mode

rudyrdx, 2024-10-22:
> Any updates? I have the same problem with the same version of firmware. dmesg | grep "Firmware: BCM43430" [ 10.316434] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM43430/1 wl0: Feb 5 2021 12:49:07 [...]

like @qrp73 suggested, first download rpi os lite, downgrade the firmware, then setup nexmon.
I have created a list of commands to execute to achieve this in one of my Rpi02w repo you can look it up.

SmillerMP, 2025-02-05:
Hi, I think I found a solution for Kali Pi Zero 2W

as @qrp73 mentioned, is necesary change the firmware, to do that run:

` wget http://archive.raspberrypi.org/debian/pool/main/f/firmware-nonfree/firmware-brcm80211_0.43+rpi6_all.deb`
`sudo apt install ./firmware-brcm80211_0.43+rpi6_all.deb -y --allow-downgrades`

This will give some warning but it is not an error

To make kali pi run this firmware instead of the original one run this:

Backup the orginal firmware:
`sudo cp /lib/firmware/brcm/brcmfmac43436s-sdio.bin /lib/firmware/brcm/brcmfmac43436s-sdio.bin.bak`
`sudo cp /lib/firmware/brcm/brcmfmac43436s-sdio.bin /lib/firmware/brcm/brcmfmac43436s-sdio.bin.bak`

Make a symlink to the new firmware
`sudo ln -sf /lib/firmware/brcm/brcmfmac43430-sdio.bin /lib/firmware/brcm/brcmfmac43436s-sdio.bin`
`sudo ln -sf /lib/firmware/brcm/brcmfmac43430-sdio.txt /lib/firmware/brcm/brcmfmac43436s-sdio.txt`
`reboot`

and now it should allow using monitor mode, from what I see the bcm43430a1 driver is well configured so it will not be necessary to recompile it

SmillerMP, 2025-02-24:
do what I mentioned above, install the firmware and compile the driver for bcm43430a1/7_45_41_46, with `iw dev wlan0 info | gawk '/wiphy/ {printf "phy" $2}'` you activate a new interface in monitor mode

utkarsharora100, 2025-02-25:
[image]
@SmillerMP Thanks for the reply!
Is this some sort of bug?
The clm_blob file is only missing for brcmfmac43436s.

utkarsharora100, 2025-02-25:
[image]
@SmillerMP Thanks for the reply!
I followed the steps given above and still I am not able to use monitor mode. It says no clm_blob available even though I have set the country code as IN.

SmillerMP, 2025-02-25:
No problem men
as far i can see, you have another version of firmware, you have the 7.45.96 version

you must have the 7.45.41.46 version, this is very important 

[image]

@rudyrdx  have a worksteps to put the interface on monitor mode, check this [](https://github.com/rudyrdx/Raspberry-Pi-Zero-2w/blob/main/Monitor%20Mode.md)

if you have the correct version driver for BCM43430, you only need to install the 7.45.41.46 firmware, the steps can be a little different on kali but you can tell me if you have any problems or if you need help with something.

utkarsharora100, 2025-02-25:
It worked finally. Thanks a lot 🫡
Earlier I had ignored this error.

Notice: Download is performed unsandboxed as root as file '/home/utkarsh/firmwar e-brcm80211_0.43+rpi6_all.deb' couldn't be accessed by user '_apt'. - pkgAcquire ::Run (13: Permission denied)

To fix this I followed these 2 steps on the terminal -->
1) Write `sudo vi /etc/apt/apt.conf.d/10sandbox`.
2)  Add the following line `APT::Sandbox::User "root";` then write and quit.

[image]

utkarsharora100, 2025-02-25:
It worked finally. Thanks a lot 🫡
Earlier I had ignored this error.

Notice: Download is performed unsandboxed as root as file '/home/utkarsh/firmwar e-brcm80211_0.43+rpi6_all.deb' couldn't be accessed by user '_apt'. - pkgAcquire ::Run (13: Permission denied)

To fix this I followed these 2 steps on the terminal -->

1) Write sudo vi /etc/apt/apt.conf.d/10sandbox.
2) Add the following line APT::Sandbox::User "root"; then write and quit.

[image]

SmillerMP, 2025-02-25:
Perfect, you're welcome. Have a happy hacking

teo2peer, 2026-11-16:
Got monitor mode working on the Zero 2 W's internal BCM43430/1 under a **current Kali kernel (6.12.34+rpt-rpi-v7, armv7l)** — no kernel downgrade needed. Sharing the driver-side port in case it helps anyone on newer kernels: **nexmon-brcmfmac-6.12** (port of `patches/driver/brcmfmac_6.6.y-nexmon` to 6.12, out-of-tree build against the stock `linux-headers-$(uname -r)`).

What it took:

1. **The firmware pair must be coherent** (thanks to this thread for the downgrade deb): nexmon 7.45.41.46 with the `clm_blob` symlink **moved aside** — the stock 2021 firmware cannot scan without its CLM (`Firmware rejected country setting` → `SCAN-FAILED ret=-22` forever), and the 2017 firmware traps if fed the 2021 CLM. Swap firmware by files + reboot, never hot-`rmmod`.
2. **8 mechanical API fixes** to compile the 6.6 tree against 6.12 headers (unaligned include, scan_width, change_beacon → `cfg80211_ap_update`, strlcpy→strscpy, void `.remove`, no_llseek, drvwrap, Makefile `$(src)`).
3. **One real bug fix**: nexmon only sets `ARPHRD_IEEE80211_RADIOTAP` in the add-vif path (which modern kernels refuse with `-95`). The path that works — `iw dev wlan0 set type monitor` (`change_iface`) — never marks the netdev type, so the radiotap-wrapped frames are delivered as Ethernet and tcpdump shows garbage. Adding the ARPHRD marking to `change_iface` fixes capture completely.

Result on 6.12:

```
listening on wlan0, link-type IEEE802_11_RADIO (802.11 plus radiotap header)
tsft 2412 MHz -72dBm signal BSSID:… Beacon (…) ESS CH: 1, PRIVACY
```

Known limitation: monitor by changing wlan0's own type; creating an extra monitor vif (airmon-ng style) is still refused with `-95`. Module is RAM-only tested (insmod; reboot restores stock driver).

Patch + build/test scripts + full README:
https://github.com/teo2peer/nexmon-brcmfmac-6.12

Also opened as a PR adding `patches/driver/brcmfmac_6.12.y-nexmon` upstream:
https://github.com/seemoo-lab/nexmon/pull/696

Thanks for nexmon and for this thread — the firmware-downgrade pointer was the missing half.