seemoo-lab/nexmon #621: "make "

Urb1101, 2024-08-07:


jlinktu, 2024-08-07:
#441 
#615

closed 2024-08-07 by jlinktu

Urb1101, 2024-08-07:
I am going to install nexmon with raspberry pi  4B, but I encountered error like the issue title when executing this point↓
https://github.com/seemoo-lab/nexmon#:~:text=bcm43436b0/9_88_4_65/nexmon/-,Compile%20a%20patched%20firmware%3A%20make,-Generate%20a%20backup
and checked "/lib/modules/<kernel version>" but couldnt find /build.
 I used "rpi-update" command to downgrade kernel so which maybe cause missing /build. 
my kernel viersion is 5.10.73-v8+ but i couldnt find linux-headers-5.10.73-v8+.(I used "sudo apt-get install ～)
there is way to install kernel header for this version or other way to solve this error?
sorry for my bad english.

Urb1101, 2024-08-07:
> #441 #615
thank you for fast reply.
#441 didnt work for me, i dont know #615 is good for me. at least I couldnt got something to solve this issue.

jlinktu, 2024-08-07:
There are plenty of resources online that describe how to install kernel headers on Raspberry Pi OS, please consult the search engine of your choice to solve that.