seemoo-lab/nexmon #623: brcmfmac.ko is undefined when executing make

Urb1101, 2024-08-13:
in Raspberrypi 4B, I executed make in patches/bcm43455c0/7_45_234_4ca95bb_CY/nexmon.
but got error:
`BUILDING DRIVER for kernel 5.10 brcmfmac_5.10.y-nexmon/brcmfmac.ko (details: log/driver.log)`
`ERROR: modpost: "brcmf_dmi_probe" [～/nexmon/patches/driver/brcmfmac_5.10.y-nexmon/brcmfmac.ko] undefined!`

kernel version:5.10.71-v8+
is there a way to solve this?

jlinktu, 2024-08-13:
Probably similar to #600 ; try the solution proposed there.

Urb1101, 2024-08-15:
I used rpi-source to download raspberrypi-kernel-headers but it didnt work, I need manual download?

Urb1101, 2024-08-15:
rpi-source dosent work? in #600 Its recommanded to use manual download but I used rpi-source to download kernel header.

Urb1101, 2024-08-15:
I tried #600 solution but it also didnt worked, is there other way to solve?

pH-Valiu, 2025-04-09:
I have the same error, only for kernel version `5.10.92-v8+` and `bcm43455c0/7_45_189`
I got rpi-source to work, still `brcmf_dmi_probe` is undefined or not available.
Has anyone found a solution for it?

pH-Valiu, 2025-04-09:
I might have found the error 🐡

pH-Valiu, 2025-04-09:
But don't they only work for 32 bit?

jlinktu, 2025-04-09:
We do not provide any precompiled / prepatched firmware and also advice against using those. You should always generate your own patched firmware.

pH-Valiu, 2025-04-09:
> I might have found the error 🐡

I managed to get nexmon_csi (and nexmon) compiled on ARM64 (5.10.92-v8+) by following the following steps:
Mostly follow the compilation guide for 64bit in the nexmon README. Additionally, if you want nexmon_csi, take this issues` comments as help: https://github.com/seemoo-lab/nexmon_csi/issues/364
Then:
- Including `dmi.o` into the target files of the `Makefile` of the driver package (`brcmfmac_5.10.y-nexmon/Makefile`) (in _nexmon_csi/_ and/or _nexmon/patches/driver_ respectively). 
- Removing the checks for architecture in the `nexmon_csi/Makefile` at lines 116 and lines 340
- The fix with rpi-source at issue #600 proved to be correct, though manually downloading the files did not do it, rather invoking `rpi-source`, waiting for it to fail, making sure to have installed `libssl-dev` and `bc` (and `xxd`, will be needed further down the road). Then call: 
```sh
make ARCH=arm64 CROSS_COMPILE=aarch64-linux-gnu- bcm2711_defconfig
make ARCH=arm64 CROSS_COMPILE=aarch64-linux-gnu- modules_prepare
```
(I defaulted all configs of the kernel headers)<br>
and make sure that `/lib/modules/5.10.92-v8+/build` links to the liunx headers directory (e.g. `/home/pi/linux-headers` or `/root/liunx-headers` depending on where you installed them)
- And then in the end, compiling nexmon (or nexmon_csi) by calling first `make ARCH=arm64 CROSS_COMPILE=aarch64-linux-gnu-` (if possible) and then `make ARCH=arm64 CROSS_COMPILE=aarch64-linux-gnu- install-firmware`

closed 2025-04-09 by jlinktu

Urb1101, 2025-04-09:
sorry I didnt know precompiled version isnt recommended. and thank you for presenting way to solve. I will try to compile again.