seemoo-lab/nexmon #624: install question

liuchengzhi314159, 2024-08-13:
I hope to use PI4 to study some thing。
first i used 2022-01-28-raspios-bullseye-arm64 with linux-kernel 5.10.xx-V8+.but when i cd patches/bcm43455c0/7 45 189/nexmon and  
make it ,it say no have /lib/module/5.10.xx-V8+/bulid,after i apt upgrade.my kernel name is 6.1.xx-V8+.
I have exchanges some image. Now ,only 2024-07-04-raspios-bullseye-arm64.img has /lib/module/6.1.xx-v8+/build.but its kernel name is linux-kernel 6.1, I think it must make somethings wrong.however,I can make brcmfmac43455-sdio.bin.but when i 'sudo make install-firmware ',it say :
Makefile:4 /firmwares/bcm43455c0/7 45 189/definitions.mk :No such file or diretory
make: *** No rule to make target '/firmwares/bcm43455c0/7 45 189/definitions.mk‘ .stop.
i don't know what happened.

jlinktu, 2024-08-13:
You forgot to `source setup_env.sh`  in the repository's root directory.

liuchengzhi314159, 2024-08-13:
？？I m sorry。i did it!

i downloaded nexmon on windows.and copy to my PI4 and  unzip nexmon-master.zip.then i did it step by step by the guide. i"m sure i 

source setup_env.sh

jlinktu, 2024-08-13:
You have to do this every time you generate a new shell environment, e.g. when opening a new emulated terminal, switching to a fresh terminal emulator tab, ...
Alternatively, you can put the source command in whatever file gets sourced by your shell when created, e.g.` ~/.bashrc` (when putting it there, remember to source that file in open shells or simply restart them).

liuchengzhi314159, 2024-08-14:
im sorry.. how can i put the source command into /.bashrc ?

jlinktu, 2024-08-14:
Please use the online search service of your choice to figure that out.

liuchengzhi314159, 2024-08-18:
Sorry, the problem is still not solved. I followed the instructions step by step in the same terminal and still have the problem in issue, known to be unsolvable