seemoo-lab/nexmon #631: Galaxy S10 bcm4375b1 patch update

Yev-henii, 2024-10-04:
Do you have any plans to update the patch for the galaxy s10 on the bcm4375b1 chip? I tried to install nexmon on my device with android 12 on one UI 4.1 but nothing worked. After reading a little on the Internet, I found reviews from people who write that due to the driver update in the kernel from version bcmdhd_101_12 to bcmdhd_101_16 along with the kernel update, nexmon is no longer suitable for these devices and there is no way to roll back the version due to Samsung protection. Perhaps you can suggest some solution?

Yev-henii, 2024-10-05:
bcmdhd_sta.bin_b1	Version=18_41_117

Yev-henii, 2024-10-05:
[bcmdhd_sta_bin_b1.zip](https://github.com/user-attachments/files/17265303/bcmdhd_sta_bin_b1.zip)

Yev-henii, 2024-10-05:
This version is the latest update that arrived over the air for the s10, I think it would be useful for many people since it is difficult to specifically saw outdated versions

savox-326, 2024-10-12:
i said that. Not sure about kernel driver version, when i last checked the new bcm binary worked fine with old kernel wifi driver, only thing which noticed was that wifi strength wasn't work. I guess this patch didn't work initially

Yev-henii, 2024-10-12:
Could you please tell me if you are talking about the new firmware version 18_41_113_sta? It is in the firmwares folder along with the definitions.mk file, but there is no patch for it in the patches folder. I want to patch it with the patch from 18_41_8_9_sta by finding and modifying functions in the local_wrapper.c file. I have found almost all the new function names using bindiff in Ida Pro, but now I am facing the issue of not being able to find this function in the 18_41_8_9 firmware to match them:

AT(CHIP_VER_BCM4375b1, FW_VER_18_41_8_9_sta, 0x1FD9BC) int called_by_wlc_ioctl(struct wlc_info *wlc, int cmd, char *arg, int len) RETURN_DUMMY

Have you tried doing this?

Also, I want to eventually try to patch my firmware version 117, but I don’t yet know how to change the values in definitions.mk for it.

I would appreciate any help because I am absolutely far from programming.

MarkusTieger, 2024-10-25:
@jlinktu you did commit the firmware for bcm4375b1 version 18.41.113.
I am assuming you have the needed knowledge to help us do the same for the 18.41.117 version?

davidrozen76, 2024-12-09:
On Samsung Galaxy S10 SM-G973F, you should be using bcm4735b1 driver version 18.38.18 in order for the nexmon patches to work.

Yev-henii, 2024-12-09:
Patch 18.38.18 works only on the initial firmware versions. The latest firmware versions for the S10 that I downloaded and unpacked have driver versions 18.41.117 or 18.41.113. Version 18.38.18 does NOT work on the latest firmware, even if you rebuild the kernel with the old driver version bcmdhd_100_10. Rolling back to a very old firmware is not possible due to Samsung's policy, so the new firmware file needs to be patched. Its functions differ significantly from version 18.38.18 and are more similar to the firmware of the S20 than the S10, but it still doesn't work.

davidrozen76, 2024-12-10:
> Patch 18.38.18 works only on the initial firmware versions. The latest firmware versions for the S10 that I downloaded and unpacked have driver versions 18.41.117 or 18.41.113. Version 18.38.18 does [...]

I can confirm that compiling the kernel with driver version 18.38.18 works.

Yev-henii, 2024-12-10:
Version 18.38.18 does not refer to the kernel driver but rather to the firmware file bcmdhd_sta.bin_b1, which you can replace by simply copying it to the vendor/firmware path in the device's memory. The chip driver, on the other hand, is located in the kernel source at drivers/net/wireless/broadcom and has version names like bcmdhd_100_10, bcmdhd_100_15, and bcmdhd_100_16.

I compiled the kernel with the bcmdhd_100_10 driver version instead of bcmdhd_100_16 and also patched the bcmdhd_sta.bin_b1 firmware file (version 18.38.18) using the Nexmon patch. I can assure you that this does not work on the latest firmware with Android 12. You can verify this by downloading the kernel built with the older driver from this link: 

https://github.com/Yev-henii/android_kernel_beyond1_lpos/releases/download/Bcmdhd_100_10/LPoS-S10-Bcmdhd_100_10-twrp-permissive.zip 

and replacing the bcmdhd_sta.bin_b1 v.18.38.18 
file patched with Nexmon, which I have attached to this message:

[bcmdhd_sta.zip](https://github.com/user-attachments/files/18078735/bcmdhd_sta.zip)

davidrozen76, 2024-12-10:
Have you tried this on Android 10?

davidrozen76, 2024-12-10:
Please try using -
nexutil -k<i>
nexutil -g0x613 -i -v2

Where <i> is a valid channel.

Yev-henii, 2024-12-10:
No, I haven't tried Android 10, and I'm not sure if such a rollback is possible, as Samsung doesn't always allow downgrading the One UI version due to bootloader updates. In any case, I would prefer to stay on Android 12. I will try to do what you wrote later.

Yev-henii, 2024-12-10:
No, I haven't tried Android 10, and I'm not sure if such a rollback is possible, as Samsung doesn't always allow downgrading the One UI version due to bootloader updates. In any case, I would prefer to stay on Android 12. I will try to do what you wrote later

Yev-henii, 2024-12-11:
So, I tried your command:

nexutil -k1  
nexutil -g0x613 -i -v2  

and it partially worked. Now, when I run the command:

sudo LD_PRELOAD=./libfakeioctl.so airodump-ng wlan0  

instead of a blank screen, I can see the surrounding Wi-Fi networks, and it looks like monitor mode. However, this only works in mode nexutil -m0 and not m2.

I didn’t even have to flash the kernel with an older driver version since it worked on the latest version, although I’m not yet fully sure if it functions as intended. Still, there’s at least some progress.

Could you explain what happens when using the command nexutil -g0x613 -i -v2?

[image]

davidrozen76, 2024-12-11:
Don't use nexutil -m0 or -m2 and you're done.

Yev-henii, 2024-12-12:
By the way, using your method, everything worked perfectly even on firmware 18.41.113

Yev-henii, 2024-12-12:
Now I have an issue with setting the channel. When I select any channel, for example, nexutils -k1, the channel doesn't lock; instead, it constantly cycles through all channels, searching for all networks on other channels.

davidrozen76, 2024-12-12:
> Now I have an issue with setting the channel. When I select any channel, for example, nexutils -k1, the channel doesn't lock; instead, it constantly cycles through all channels, searching for all networks on other channels.

Kill wpa_supplicant or make sure your wifi service is disabled.

Yev-henii, 2024-12-12:
You are correct; I had to adapt the addresses of some functions using Bindiff in IDA Pro, using patch 18.41.8.9 as a base. In fact, most of the patch for version 113 is already present in the author's repository, and it’s evident that someone has already patched the firmware for version 113, as they spent significant time adding some lines of code. It’s strange, however, that a portion necessary for completion is missing, giving the impression that the author deliberately left it out for some reason. You can find my adaptation for version 113 here:

https://github.com/Yev-henii/nexmon/tree/18_41_8_113_sta

I believe I failed to map only one address between versions 8.9 and 113, though I don’t remember exactly which one. I’ll write about it later. However, I’m 99% sure that this patch is adapted for version 113 and even seems to work to some extent.

Additionally, I came across some addresses for version 117, but since other information about addresses for this version and the definitions file is missing, I doubt I have enough expertise to adapt the patch for version 117.

In any case, I suggest continuing to develop this topic for those interested and sharing proposals for improvements.

MarkusTieger, 2024-12-12:
> You are correct; I had to adapt the addresses of some functions using Bindiff in IDA Pro, using patch 18.41.8.9 as a base. In fact, most of the patch for version 113 is already present in the [...]

I tried the 18_41_8_113_sta of your fork. Well, it doesn't work. "ifconfig wlan0 up" fails with "SIOCSIFFLAGS: Operation not permitted", and enabling wifi the normal android way doesn't work either.

This is weird. I am assuming because of your message "By the way, using your method, everything worked perfectly even on firmware 18.41.113", you got it working?

MarkusTieger, 2024-12-12:
> When I get home, I’ll send you the already patched firmware file that works for me. I didn’t use ifconfig wlan0 up but instead disabled and re-enabled Wi-Fi in the Android shell without being connected to any network (Wi-Fi itself works perfectly). Then I used the following commands:
> 
> sudo nexutil -k1 sudo nexutil -g0x613 -i -v2 sudo LD_PRELOAD=./libfakeioctl.so airodump-ng wlan0
> 
> This gave me what seems to be a working monitor mode with continuous channel hopping on the 2.4 GHz frequency. The next thing I plan to do when I have some free time is to try fixing it to a single channel. I also don’t know yet whether 5 GHz channels work or whether frame injection is functional.
> 
> Does Wi-Fi work correctly for you before patching with my version?

The unpatched firmware 18.41.8.113 (in your repository) does work just fine (normal usage). "ifconfig wlan0 up" and monitor mode does work with the 18.41.8.9 patched firmware.

I was able to see 5 Ghz networks with airodump-ng. In my case, my wifi network was using channel 52 (on 5 ghz).

```
sudo nexutil -k52
sudo nexutil -g0x613 -i -v2
sudo LD_PRELOAD=./libfakeioctl.so airodump-ng wlan0 -c 52
```

and it only showed me my wifi network on this channel. (and another one on channel 48, I guess that channel overlaps with 52).

thx in advance.

Yev-henii, 2024-12-12:
Try replacing this file in vendor/firmware. This is a pre-patched version 113 firmware file that works on my device.
[18_41_8_113_patched_nexmon.zip](https://github.com/user-attachments/files/18118351/18_41_8_113_patched_nexmon.zip)

MarkusTieger, 2024-12-13:
> Try replacing this file in vendor/firmware. This is a pre-patched version 113 firmware file that works on my device. [18_41_8_113_patched_nexmon.zip](https://github.com/user-attachments/files/18118351/18_41_8_113_patched_nexmon.zip)

This one actually works. thx

MarkusTieger, 2024-12-13:
Is there a way to reset the channel lock and monitor mode (for normal wifi usage) without rebooting my phone?
It is not possible to compile the bcmd as a module (because some exynos drivers in the kernel require it).

MarkusTieger, 2024-12-13:
> Have you figured out how to lock the channel and tried disabling wpa_supplicant? Does frame injection work? I don’t have time to try it right now. Let me know if you manage to make it work properly and what you did to achieve it.

For me at least, "nexutil -k<channel>" seems to be working (it does only take effect if "nexutil -g0x613 -i -v2" is executed after that, even if you already did it once)

frame injection didn't give an error with aireplay-ng, but I do not know yet if it works. (I have to disable protected management frames and see if devices disconnect)

I don't know what you mean by "disabling" wpa_supplicant. I just disabled wifi which stops the wpa supplicant.
Then I reenable the wifi interface with "ifconfig wlan0 up".

Yev-henii, 2024-12-13:
In my case, after executing the command sudo LD_PRELOAD=./libfakeioctl.so airodump-ng wlan0, the program scans for networks across all channels. I tried locking the channel using nexutil -k1 and adding the --channel 1 option to the airodump-ng command, but I still received information about various Wi-Fi networks that were shown as being on channel 1, even though they were actually on other channels.

MarkusTieger, 2024-12-13:
> In my case, after executing the command sudo LD_PRELOAD=./libfakeioctl.so airodump-ng wlan0, the program scans for networks across all channels. I tried locking the channel using nexutil -k1 and [...]

The other networks that are shown, is the channel number close to your selected one?

Yev-henii, 2024-12-13:
When I ran airodump-ng without the --channel 1 option, it hopped between channels and displayed networks on different channels, with the channel values matching the actual channels. However, as soon as I tried to forcibly lock the channel with the --channel 1 option, those same networks were shown as if they were all on channel 1. In other words, nexutil -k1 didn’t lock the scanning to just channel 1, and it seemed like something external was still switching the channels. Perhaps it’s necessary to run kill wpa_supplicant, as suggested in a previous comment. I’m not sure yet, and I can’t test it at the moment.

Yev-henii, 2024-12-13:
Ok, I think I'll figure out the channels later. The next topic is how to work with utilities like Wifite or Airgeddon using this. Let me know if you try to do it.

MarkusTieger, 2024-12-13:
I am unsure if management frame injection is working.
The devices do not disconnect ( wifi 4, no protected management frames and 2,4 ghz).

But in aireplay-ng I see that if the target device is connected that there are acknowledgements (indicating that some device answered to the deauthentication message). They are missing if the device is not connected.

savox-326, 2024-12-14:
Still have Note 20 Ultra (Snapdragon) with bcm4375 and also have S23 (Snapdragon) with qca6490 (qcacld-3.0). I retested, in specific cases with your method bcm's monitor really starts work, but again sticks on -1 channel, but station's channels correct. Maybe it is Snapdragon's kernel driver issue (channel -1), used my  own 18.41.8.113 patched firmware. I did it long time ago, but remember that all was good. So, checked injections... and it's not work. Enabled hotspot on my S23, connected other phone and tried `aireplay-ng -9 wlan0` it found my AP but received that `0/30 0%`. Phone also was still connected, without any wi-fi deaths. That's sad, now, I just will use S23's monitor mode (on qcacld-3.0 injections still not work also) as before, cause uts monitor easy to run and more reliable...

Yev-henii, 2024-12-14:
It’s a pity, but I have no ideas on how to fix this, as I’m not an expert in this area. It would be great if someone who has thoughts on this could help us figure out how to resolve the situation

savox-326, 2024-12-14:
I retested but in Kali Nethunter environment, used this: [https://github.com/RoninNada/libnexmonkali/tree/libnl-support](url) as ld_preload, and... channel not sticking on -1, injections also works fine. Maybe nexmon's native tools (aircrack-ng) not good for work in android environment

savox-326, 2024-12-14:
But sometimes monitor don't show anything, maybe we (or me, who on Snapdragon) need to figure out correct algorithm for monitor start

Yev-henii, 2024-12-14:
This is great news! I’ll try it later and let you know the results. My device has an Exynos processor.

savox-326, 2024-12-14:
I forgot to say `aireplay-ng -9 wlan0` also don't work on NetHunter. For the first time monitor don't receive any stations, but it enough for aireplay-ng to exit with `Found 0 APs`, same as on android, but rarely it finally catches AP. But still get that aireplay-ng on Nethunter works but not works with Nexmon's tools. aireplay with targeting certain AP works good

MarkusTieger, 2024-12-16:
> I retested but in Kali Nethunter environment, used this: [https://github.com/RoninNada/libnexmonkali/tree/libnl-support](url) as LD_PRELOAD, and... channel not sticking on -1, injections also works fine. Maybe nexmon's native tools (aircrack-ng) not good for work in android environment

I have managed to successfully deauthenticate a device using this library and nethunter.

arti2525, 2024-12-17:
> > I retested but in Kali Nethunter environment, used this: [https://github.com/RoninNada/libnexmonkali/tree/libnl-support](url) as LD_PRELOAD, and... channel not sticking on -1, injections also works fine. Maybe nexmon's native tools (aircrack-ng) not good for work in android environment
> 
> I have managed to successfully deauthenticate a device using this library and nethunter. Management frame injection seems to work
Great, could you share compiling the kernel?

Yev-henii, 2024-12-17:
Which specific kernel compilation are you interested in? I am using LPoS Kernel for S10 (Exynos).

MarkusTieger, 2024-12-17:
> I retested but in Kali Nethunter environment, used this: [https://github.com/RoninNada/libnexmonkali/tree/libnl-support](url) as LD_PRELOAD, and... channel not sticking on -1, injections also works [...]

I am using https://github.com/linux4-bringup-priv/android_kernel_samsung_exynos9820/ with crDroid 11 (Android 15).

arti2525, 2024-12-17:
> I am using https://github.com/linux4-bringup-priv/android_kernel_samsung_exynos9820/ with crDroid 11 (Android 15).

Thank you very much, could you tell me the firmware version 18.41.8.113 or 18.41.8.9 and if you have already patched it, share it?, so we can try to include it in nethunter in a new build

MarkusTieger, 2024-12-17:
> Try replacing this file in vendor/firmware. This is a pre-patched version 113 firmware file that works on my device. [18_41_8_113_patched_nexmon.zip](https://github.com/user-attachments/files/18118351/18_41_8_113_patched_nexmon.zip)

I used this one. The source code should be available here https://github.com/Yev-henii/nexmon/tree/18_41_8_113_sta, but compiling that resulted in a broken firmware (for me). The prepatched version from Yev-henii did work.

MarkusTieger, 2024-12-17:
> Try replacing this file in vendor/firmware. This is a pre-patched version 113 firmware file that works on my device. [18_41_8_113_patched_nexmon.zip](https://github.com/user-attachments/files/18118351/18_41_8_113_patched_nexmon.zip)

Can you verify there aren't any uncommited changes in your repository? I tried patching it myself again, but broken firmware again. (The patch process itself was successful though)

The other firmware versions in the official nexmon repo are working after I patch them myself, which lets me believe I didn't do a mistake here.

MarkusTieger, 2024-12-17:
> > Can you verify there aren't any uncommited changes in your repository? I tried patching it myself again, but got broken firmware again. (The patch process itself was successful though)
> > The other firmware versions in the official nexmon repo are working after I patch them myself, which lets me believe I didn't do a mistake here.
> 
> [bcmdhd_sta.zip](https://github.com/user-attachments/files/18170681/bcmdhd_sta.zip)
> 
> Here, take my one

I already have a working binary, but I want to reproduce it using the source code.

Yev-henii, 2024-12-18:
Ok, packet injection works perfectly directly from NetHunter using aireplay-ng, but when I run wifite using LD_PRELOAD=./libnexmon.so wifite -i wlan0, the program starts and finds access points, but deauthentication still doesn't work. How can I solve this?

MarkusTieger, 2024-12-18:
I haven't tried wifite at all. I only succeeded by deauthenticating using aireplay-ng, (I disabled Protected Management Frames, 2,4 Ghz only, WPA2).

savox-326, 2024-12-19:
Frame injection works only if nethunter chroot is started/mounted through a centralized "Nethunter app" (not nethunter terminal), otherwise each injection attempt will fail and perhaps will disable wlan0. Nethunter's chroot need access to kernel properties which be provided through mounting.

For me injections works good in every freq (2.4Ghz and 5Ghz), but sometimes don't find certian freq stations (2.4Ghz or 5Ghz). I have to change channel through `nexutil -k35` for example for 5Ghz or `nexutil -k1` for 2.4Ghz

Yev-henii, 2024-12-20:
I noticed that I can use Wifite normally with the version from here: 

https://github.com/awesome-pentest-gadgets/WiFite2-RPi3-nexmon

However, it has an outdated version 2.0 and doesn't support the PMKID attack. When I try to use the regular Wifite version 2.7 from the repository, it launches, scans networks fine, but then freezes on any attack. Did you modify the Wifite code in any way, or does everything work out of the box without modifications? Airgeddon doesn't work either.

arti2525, 2025-03-25:
Hello, I'm asking how you managed to write the libraries in Android 15 (LineageOS 22.1 or CDroid 11) if they are only readable in system/bin (root of course), or if you could help me with what ROM and kernel you managed to get Nexmon working in S10.

arti2525, 2025-03-25:
> I noticed that I can use Wifite normally with the version from here:
>
> https://github.com/awesome-pentest-gadgets/WiFite2-RPi3-nexmon
>
> However, it has an outdated version 2.0 and doesn't support the [...]
Hello, I'm asking how you managed to write the libraries in Android 15 (LineageOS 22.1 or CDroid 11) if they are only readable in system/bin (root of course), or if you could help me with what ROM and kernel you managed to get Nexmon working in S10.

MarkusTieger, 2025-03-25:
> Hello, I'm asking how you managed to write the libraries in Android 15 (LineageOS 22.1 or CDroid 11) if they are only readable in system/bin (root of course), or if you could help me with what ROM and kernel you managed to get Nexmon working in S10.

I got it working with both LineageOS and crDroid. I recommend LineageOS. CrDroid doesn't feel as smooth on this device there. I compile my lineageos builds myself (I have some patches), so I just replaced the firmware blob before the build, but you can probably replace it with Magisk too?.

arti2525, 2025-03-25:
Great thanks for your time, I'm trying to help v0lk3n (https://github.com/V0lk3n/nethunter_kernel_samsung_exynos9820), he made the port of the kernel for nethunter with lineageos 22.1 but he couldn't get nexmon to work. If you have any image of the rom or something that could help us include nexmon directly in nethunter it would help us. thx

MarkusTieger, 2025-03-25:
> Great thanks for your time, I'm trying to help v0lk3n (https://github.com/V0lk3n/nethunter_kernel_samsung_exynos9820), he made the port of the kernel for nethunter with lineageos 22.1 but he [...]

Do you mean the patched nexmon firmware? Yev-henii did share his patched firmware here: https://github.com/seemoo-lab/nexmon/issues/631#issuecomment-2540174388

Download: https://github.com/user-attachments/files/18118351/18_41_8_113_patched_nexmon.zip

arti2525, 2025-03-25:
> > Great thanks for your time, I'm trying to help v0lk3n (https://github.com/V0lk3n/nethunter_kernel_samsung_exynos9820), he made the port of the kernel for nethunter with lineageos 22.1 but he couldn't get nexmon to work. If you have any image of the rom or something that could help us include nexmon directly in nethunter it would help us. thx
> 
> Do you mean the patched nexmon firmware? Yev-henii did share his patched firmware here: [#631 (comment)](https://github.com/seemoo-lab/nexmon/issues/631#issuecomment-2540174388)
> 
> Download: https://github.com/user-attachments/files/18118351/18_41_8_113_patched_nexmon.zip

Hi, the big problem we have is that (Lineage 22.1 or CDroid 11) we can't use the libraries because we can't access system/bin, etc. It's in RW mode. We put the patched firmware in vendor/firmware, but the libraries are rejected because they can't mount the partition.

V0lk3n: Did you get TWRP to work? Because I can't access the system or data with it, and the Lineage installation doesn't work properly (so I'll have problems transferring the firmware to the vendor).

jlinktu, 2025-03-26:
We advise against using pre-built patched firmware blobs, you should always inspect first and built patched firmware yourself.

MarkusTieger, 2025-06-04:
Maybe interesting for you: https://github.com/seemoo-lab/nexmon/pull/662

I will also port the firmware patches after this.

MarkusTieger, 2025-06-06:
> What a great collab! How did you manage to mount /vendor rw? Or how did you place fw in /vendor/firmware?

I compiled LineageOS myself and replaced it in the vendor/samsung/exynos9820-common part. If you are using LineageOS, you can't mount it rw, because it uses erofs instead of ext4 iirc.

Yev-henii, 2025-06-06:
> What a great collab! How did you manage to mount /vendor rw? Or how did you place fw in /vendor/firmware?

If you are using TWRP recovery, then once you enter it, you can move the patched firmware file using the TWRP file manager by simply copying it—if, of course, I understood your question correctly

yesimxev, 2025-06-07:
> If you are using TWRP recovery, then once you enter it, you can move the patched firmware file using the TWRP file manager by simply copying it—if, of course, I understood your question correctly

TWRP can't mount vendor since it's encrypted. I tried to wipe system, vendor, flash LOS in LOS recovery since it fails on twrp (dynamic partition issue). i can try flasing back twrp before first boot, it may be good. Btw combining with mutlidisabler doesn't help either

[13 more replies]