seemoo-lab/nexmon #633: Can't setup nexmon for Nexus 6P

amenekowo, 2024-10-15:
# Summary
Hello there!
I installed nexmon firmware to my Nexus 6P. Because I use magisk to root and **installed Kali Nethunter by Magisk**, the `/system` is read only and I changed the utils installation path to `/vendor`  and the firmware and tools installed successfully. But when I run any monitor steps in README, it doesn't work.

<details>
<summary>Use `nexutil -m2`</summary>

```
angler:/vendor/bin # ./nexutil -m2
angler:/vendor/bin # iw wlan0 info
Interface wlan0
        ifindex 5
        wdev 0x1
        addr 98:e7:f5:xx:xx:xx
        type managed
        wiphy 0
angler:/vendor/bin # LD_PRELOAD=/vendor/lib/libnexmon.so ./airodump-ng wlan0
CANNOT LINK EXECUTABLE "sh": "/vendor/lib/libnexmon.so" is 32-bit instead of 64-bit
```
and it freezed. (I found that only `sh` binary is aarch64, both libnexmon.so and airodump-ng is arm.)
```
angler:/vendor/bin # file ./airodump-ng
./airodump-ng: ELF shared object, 32-bit LSB arm, dynamic (/system/bin/linker), stripped
angler:/vendor/bin # file /vendor/lib/libnexmon.so
/vendor/lib/libnexmon.so: ELF shared object, 32-bit LSB arm, dynamic (/system/bin/linker), stripped
```
</details>

<details>
<summary>Use iw phy `iw dev wlan0 info | gawk '/wiphy/ {printf "phy" $2}'` interface add mon0 type monitor</summary>

```
angler:/vendor/bin # iw phy phy0 interface add mon0 type monitor
command failed: Operation not supported on transport endpoint (-95)
angler:/vendor/bin # ifconfig
rmnet_ipa0 Link encap:UNSPEC
          UP RUNNING  MTU:2000  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:0 TX bytes:0

wlan0     Link encap:Ethernet  HWaddr 98:e7:f5:xx:xx:xx
          UP BROADCAST RUNNING  MTU:1500  Metric:1
          RX packets:114153 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:11006437 TX bytes:0

dummy0    Link encap:Ethernet  HWaddr 7e:c5:29:xx:xx:xx
          inet6 addr: fe80::7cc5:29ff:fexx:xxxx/64 Scope: Link
          UP BROADCAST RUNNING NOARP  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:16 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0
          RX bytes:0 TX bytes:2834

lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope: Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:204 errors:0 dropped:0 overruns:0 frame:0
          TX packets:204 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0
          RX bytes:32722 TX bytes:32722
angler:/vendor/bin # iw wlan0 info
Interface wlan0
        ifindex 5
        wdev 0x1
        addr 98:e7:f5:xx:xx:xx
        type managed
        wiphy 0
```
</details>

---

And another bug(Idk if it is called a bug) is, if I don't run `ifconfig wlan0 up`, my WLAN chip is not useable. This is mentioned [there](https://github.com/seemoo-lab/nexmon/issues/195#issuecomment-570985026).
<details>
<summary>dmesg output</summary>

```
[ 1605.705493] \x0aDongle Host Driver, version 1.201.31 (r)\x0aCompiled in drivers/net/wireless/bcmdhd on Oct 11 2018 at 19:39:21
[ 1605.705524] dhd_wlan_power Enter: power on
[ 1605.906382] dhd_bus_devreset: == Power ON ==
[ 1605.951092] dhd_bus_devreset: dhdpcie_bus_clock_start OK
[ 1605.951832] dhdpcie_dongle_attach: PCI_BAR1_WIN = 0
[ 1605.952487] dhdpcie_dongle_attach: BAR1 window val=23 mask=0
[ 1605.953211] dhdpcie_download_code_file: download firmware /vendor/firmware/fw_bcmdhd.bin
[ 1605.953253] _dhdpcie_download_firmware: dongle image file download failed
[ 1605.953263] dhd_bus_start: failed to download firmware /vendor/firmware/fw_bcmdhd.bin
[ 1605.953270] dhd_bus_devreset: dhd_bus_start: -1
[ 1605.953277] dhd_net_bus_devreset: dhd_bus_devreset: -1
[ 1605.953284] dhd_open : wl_android_wifi_on failed (-1)
[ 1605.953300] dhd_prot_ioctl : bus is down. we have nothing to do
[ 1605.953309] dhd_bus_devreset: == Power OFF ==
[ 1605.960782] dhd_bus_devreset:  WLAN OFF Done
[ 1605.960804] dhd_wlan_power Enter: power off
```

</details>

If i bring up wlan0 by `ifconfig wlan0 up`, it is working.
<details>
<summary>dmesg output</summary>

```
[ 2178.973355] \x0aDongle Host Driver, version 1.201.31 (r)\x0aCompiled in drivers/net/wireless/bcmdhd on Oct 11 2018 at 19:39:21
[ 2178.973427] dhd_wlan_power Enter: power on
[ 2179.176411] dhd_bus_devreset: == Power ON ==
[ 2179.221097] dhd_bus_devreset: dhdpcie_bus_clock_start OK
[ 2179.225276] dhdpcie_dongle_attach: PCI_BAR1_WIN = 0
[ 2179.228714] dhdpcie_dongle_attach: BAR1 window val=23 mask=0
[ 2179.230718] dhdpcie_download_code_file: download firmware /vendor/firmware/fw_bcmdhd.bin
[ 2179.313765] dhdpcie_bus_write_vars: Download, Upload and compare of NVRAM succeeded.
[ 2179.315465] Failed to open the file logstrs.bin in dhd_init_logstrs_array, /vendor/firmware/logstrs.bin
[ 2179.530879] dhd_bus_start: Initializing 42 flowrings
[ 2179.531391] dhd_bus_cmn_writeshared:
[ 2179.531424] dhd_bus_cmn_writeshared:
[ 2179.531456] dhd_bus_cmn_writeshared:
[ 2179.531486] dhd_bus_cmn_writeshared:
[ 2179.531516] dhd_bus_cmn_writeshared:
[ 2179.531543] dhd_bus_cmn_writeshared:
[ 2179.531616] dhd_bus_cmn_writeshared:
[ 2179.585120] dhd_prot_ioctl: status ret value is -5
[ 2179.587384] dhd_preinit_ioctls lpc fail WL_DOWN : 0, lpc = 1
[ 2179.590423] dhd_prot_ioctl: status ret value is -23
[ 2179.618453] dhd_prot_ioctl: status ret value is -26
[ 2179.654451] dhd_rtt_init : FTM is supported
[ 2179.658422] dhd_bus_devreset: WLAN Power On Done
```
</details>

Any ideas is welcome. Thanks!

# Environment
Nexus 6P running Android Oreo (8.1) and Kali Nethunter
<details>
<summary>`nexutil -v` output</summary>

```
angler:/vendor/bin # nexutil -V
platform Nexus 6P
firmware 7.112.300.14 (r707445) FWID 01-3242a45b
vendorid 0x14e4
deviceid 0x43e9
radiorev 0x2e2069
chipnum 0x4358
chiprev 0x3
chippackage 0x2
corerev 0x30
boardid 0x7a1
boardvendor 0x14e4
boardrev P100
driverrev 0x77012c0
ucoderev 0x3c3013d
bus 0x0
phytype 0xb
phyrev 0x11
anarev 0x0
nvramrev 0x7a1f2
```
</details>

Kernel version
`Linux kali 3.10.73-g309d642 #1 SMP PREEMPT Thu Oct 11 19:39:39 UTC 2018 aarch64`

jlinktu, 2024-10-17:
> # Summary
> Hello there! I installed nexmon firmware to my Nexus 6P. Because I use magisk to root and **installed Kali Nethunter by Magisk**, the `/system` is read only and I changed the utils installation path to `/vendor` and the firmware and tools installed successfully. But when I run any monitor steps in README, it doesn't work.
> 
> Use `nexutil -m2`
> ```
> angler:/vendor/bin # ./nexutil -m2
> angler:/vendor/bin # iw wlan0 info
> Interface wlan0
>         ifindex 5
>         wdev 0x1
>         addr 98:e7:f5:xx:xx:xx
>         type managed
>         wiphy 0
> angler:/vendor/bin # LD_PRELOAD=/vendor/lib/libnexmon.so ./airodump-ng wlan0
> CANNOT LINK EXECUTABLE "sh": "/vendor/lib/libnexmon.so" is 32-bit instead of 64-bit
> ```
> 
> and it freezed. (I found that only `sh` binary is aarch64, both libnexmon.so and airodump-ng is arm.)
> 
> ```
> angler:/vendor/bin # file ./airodump-ng
> ./airodump-ng: ELF shared object, 32-bit LSB arm, dynamic (/system/bin/linker), stripped
> angler:/vendor/bin # file /vendor/lib/libnexmon.so
> /vendor/lib/libnexmon.so: ELF shared object, 32-bit LSB arm, dynamic (/system/bin/linker), stripped
> ```

You answered it yourself already. Compile `libnexmon.so` and `airodump-ng` for the correct achitecture.
`iw` will not show the device as being of type `monitor`. That's what you need `libnexmon.so` for. It let's programs see the interface as `monitor` interface even though the driver is not aware of it.

> 
> Use iw phy `iw dev wlan0 info | gawk '/wiphy/ {printf "phy" $2}'` interface add mon0 type monitor
> ```
> angler:/vendor/bin # iw phy phy0 interface add mon0 type monitor
> command failed: Operation not supported on transport endpoint (-95)
> angler:/vendor/bin # ifconfig
> rmnet_ipa0 Link encap:UNSPEC
>           UP RUNNING  MTU:2000  Metric:1
>           RX packets:0 errors:0 dropped:0 overruns:0 frame:0
>           TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
>           collisions:0 txqueuelen:1000
>           RX bytes:0 TX bytes:0
> 
> wlan0     Link encap:Ethernet  HWaddr 98:e7:f5:xx:xx:xx
>           UP BROADCAST RUNNING  MTU:1500  Metric:1
>           RX packets:114153 errors:0 dropped:0 overruns:0 frame:0
>           TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
>           collisions:0 txqueuelen:1000
>           RX bytes:11006437 TX bytes:0
> 
> dummy0    Link encap:Ethernet  HWaddr 7e:c5:29:xx:xx:xx
>           inet6 addr: fe80::7cc5:29ff:fexx:xxxx/64 Scope: Link
>           UP BROADCAST RUNNING NOARP  MTU:1500  Metric:1
>           RX packets:0 errors:0 dropped:0 overruns:0 frame:0
>           TX packets:16 errors:0 dropped:0 overruns:0 carrier:0
>           collisions:0 txqueuelen:0
>           RX bytes:0 TX bytes:2834
> 
> lo        Link encap:Local Loopback
>           inet addr:127.0.0.1  Mask:255.0.0.0
>           inet6 addr: ::1/128 Scope: Host
>           UP LOOPBACK RUNNING  MTU:65536  Metric:1
>           RX packets:204 errors:0 dropped:0 overruns:0 frame:0
>           TX packets:204 errors:0 dropped:0 overruns:0 carrier:0
>           collisions:0 txqueuelen:0
>           RX bytes:32722 TX bytes:32722
> angler:/vendor/bin # iw wlan0 info
> Interface wlan0
>         ifindex 5
>         wdev 0x1
>         addr 98:e7:f5:xx:xx:xx
[... 68 lines cut ...]
> angler:/vendor/bin # nexutil -V
> platform Nexus 6P
> firmware 7.112.300.14 (r707445) FWID 01-3242a45b
> vendorid 0x14e4
> deviceid 0x43e9
> radiorev 0x2e2069
> chipnum 0x4358
> chiprev 0x3
> chippackage 0x2
> corerev 0x30
> boardid 0x7a1
> boardvendor 0x14e4
> boardrev P100
> driverrev 0x77012c0
> ucoderev 0x3c3013d
> bus 0x0
> phytype 0xb
> phyrev 0x11
> anarev 0x0
> nvramrev 0x7a1f2
> ```
> 
> Kernel version `Linux kali 3.10.73-g309d642 #1 SMP PREEMPT Thu Oct 11 19:39:39 UTC 2018 aarch64`

For an example on how to use Magisk to install patched firmware, have a look at https://github.com/seemoo-lab/nexmon/tree/master/patches/bcm4389c1/20_101_57_r1035009/nexmon .

amenekowo, 2024-12-09:
Thank you for your reply.

Now I have time to work with it, and I references to [this](https://github.com/seemoo-lab/nexmon/tree/master/patches/bcm4389c1/20_101_57_r1035009/nexmon),and I have a working Nexmon patch on Magisk module.

For the module, if you don't want to build it yourself, you could use mine provided here.

[nexmon-magisk-nexus6p.zip](https://github.com/user-attachments/files/18062231/nexmon-magisk-nexus6p.zip)

If anyone is interested in how I do this, I made a memo for this. You can find it [here](https://github.com/amenekowo/zheteng/blob/main/241209-nexmon-on-nexus6p.md).

Thank for your support!

closed 2024-12-09 by amenekowo

jlinktu, 2024-12-10:
We advise against using pre-built patched firmware blobs, you should always inspect first and built patched firmware yourself. By following the memo @amenekowo linked, you should be able to do this.