seemoo-lab/nexmon #637: Debugger - Base address

sigmoid-bar, 2025-01-10:
Hello,
I was attempting to understand the debugger project (https://github.com/seemoo-lab/nexmon_debugger). 
From that I understand that the memory mapped interface is being accessed via the base address defined here  https://github.com/seemoo-lab/nexmon/blob/ef25ce36700acf60f91ebc58c43f0b2d8dc372cd/patches/include/debug.h#L38

Could you help me understand how that address is found?
In some chips there is a cortex A7 instead of the R4, so this does not work as well as requiring updates to the offsets of the breakpoint and watchpoint register locations (https://github.com/seemoo-lab/nexmon/blob/ef25ce36700acf60f91ebc58c43f0b2d8dc372cd/patches/include/debug.h#L166)

Any ideas or advice would be appreciated, and thank you for this great work.

sigmoid-bar, 2025-01-10:
Well I should have read a bit further. 
From the paper The Nexmon firmware analysis: ... it says you read the Debug ROM address from the DBGDRAR register. 
You executed an MRC instruction that recovered this and dumped it to the chips console.
I will try the same.

jlinktu, 2025-01-10:
You can find more details about this by searching ARM's official Technical Reference Manual (TRM) for the corresponding core types.

sigmoid-bar, 2025-01-10:
Thanks - I had a look at the TRM for cr4 and ca7, nothing is explicitly mentioned in terms of memory mapped base addresses but will read it further.

I tried reading the `DBGDRAR` register using the coprocessor interface, however it reports 0x0. `asm volatile ("mrc p14, 0, %0, c1, c0, 0\n" : "=r"(dbgdrar));`

Reading the DBGDSAR reports 0x10000. Which looks valid for the CPU.
`asm volatile ("mrc p14, 0, %0, c2, c0, 0\n" : "=r"(dbgdsar));`

I may have to resort to accessing the co-processor interface to use the breakpoint and watchpoint features as an alternative to the memory mapped interface.

Do you have any suggestions on another approach to find the Debug ROM address?

jlinktu, 2025-01-10:
According to the Cortex-A7 TRM the validity of the value read from `DBGDRAR` and `DBGDSAR` is indicated in its lowest bit, which seems to be `0` in your case and thus invalid. I would therefore assume that "[...] no memory-mapped debug components are implemented [...]" - https://developer.arm.com/documentation/ddi0406/cb/Debug-Architecture/The-Debug-Registers/Register-descriptions--in-register-order/DBGDRAR--Debug-ROM-Address-Register?lang=en

sigmoid-bar, 2025-01-10:
Ah okay - thank you. I will try implementing the rest by accessing the co-processor interface instead.
this was for the Pixel 6, running BCM4389c1. 

Thanks for a quick response.