seemoo-lab/nexmon #655: Maybe, there is no Ucode.bin :( I cant "make" nexmon firmware on RPi02W

Sonys9, 2025-04-08:
I trying to make a patched firmware for Raspberry Pi Zero 2W (based on Raspberry OS legacy 64BIT lite because this system is lightweigth and cool)

Error:
```
root@raspberrypi:/home/admin/nexmon/patches/bcm43430a1/7_45_41_46# cd nexmon
root@raspberrypi:/home/admin/nexmon/patches/bcm43430a1/7_45_41_46/nexmon# make

                                           <logo>

                The C-based Firmware Patching Framework

                           !!! WARNING !!!
    Our software may damage your hardware and may void your hardware’s
     warranty! You use our tools at your own risk and responsibility

  COLLECTING STATISTICS read /home/admin/nexmon/STATISTICS.md for more information
  COMPILING src/version.c => obj/version.o (details: log/compiler.log)
make: *** No rule to make target '/home/admin/nexmon/firmwares/bcm43430a1/7_45_41_46/ucode.bin', needed by 'gen/ucode.bin'.  Stop.
root@raspberrypi:/home/admin/nexmon/patches/bcm43430a1/7_45_41_46/nexmon#

root@raspberrypi:/home/admin/nexmon/patches/bcm43430a1/7_45_41_46/nexmon/log# cat error.log
cat: gen/nexmon.pre: No such file or directory
cat: gen/nexmon.pre: No such file or directory
root@raspberrypi:/home/admin/nexmon/patches/bcm43430a1/7_45_41_46/nexmon/log#

root@raspberrypi:/home/admin/nexmon/patches/bcm43430a1/7_45_41_46/nexmon/log# cat compiler.log
decl_name: b_hndrte_idle
decl_name: gp4_wl_send_hook
decl_name: gp4_wlc_ioctl_hook
decl_name: wlc_hwtimer_alloc_timeout
decl_name: wlc_hwtimer_add_timeout
decl_name: wlc_hwtimer_del_timeout
decl_name: bl_flash_patch_179
decl_name: bl_wlc_ucode_write_compressed
decl_name: gp4_hndrte_reclaim_0_end
decl_name: gp4_templateram_bin
decl_name: gp4_version_patch
decl_name: gp4_version_patch
decl_name: gp4_version_patch
decl_name: gp4_version_patch
root@raspberrypi:/home/admin/nexmon/patches/bcm43430a1/7_45_41_46/nexmon/log#
```

What can i do?

jlinktu, 2025-04-08:
You probably did not run `make` in the root directory (or at least it did not run without errors), which would otherwise had created the missing ucode file.

Sonys9, 2025-04-09:
```
nexmon
root@raspberrypi:/home/admin# cd nexmon
root@raspberrypi:/home/admin/nexmon# ls
app         CITATION.cff  firmwares  ios_utilities  Makefile    patches    REVINFO.md   setup_env.sh   utilities
buildtools  DEVICES.md    gfx        LICENSE.txt    nexmon.bib  README.md  SECURITY.md  STATISTICS.md
root@raspberrypi:/home/admin/nexmon# source setup_env.sh
root@raspberrypi:/home/admin/nexmon# make
  BUILDING BUILDTOOLS
make[1]: Entering directory '/home/admin/nexmon/buildtools'
  BUILDING flash_patch_extractor
make[2]: Entering directory '/home/admin/nexmon/buildtools/flash_patch_extractor'
make[2]: 'fpext' is up to date.
make[2]: Leaving directory '/home/admin/nexmon/buildtools/flash_patch_extractor'
  BUILDING ucode_extractor
make[2]: Entering directory '/home/admin/nexmon/buildtools/ucode_extractor'
make[2]: 'ucodeext' is up to date.
make[2]: Leaving directory '/home/admin/nexmon/buildtools/ucode_extractor'
  BUILDING b43/assembler/b43-asm.bin
make[2]: Entering directory '/home/admin/nexmon/buildtools/b43/assembler'
     YACC     parser.c
/bin/sh: 1: bison: not found
make[2]: *** No rule to make target 'parser.c', needed by 'dep/parser.d'.  Stop.
make[2]: Leaving directory '/home/admin/nexmon/buildtools/b43/assembler'
make[1]: *** [Makefile:21: b43/assembler/b43-asm.bin] Error 2
make[1]: Leaving directory '/home/admin/nexmon/buildtools'
make: *** [Makefile:9: buildtools] Error 2
root@raspberrypi:/home/admin/nexmon#
```

> You probably did not run `make` in the root directory (or at least it did not run without errors), which would otherwise had created the missing ucode file.

Sonys9, 2025-04-09:
No no non no fixed
apt install bison

Sonys9, 2025-04-09:
new error
```
e[1]: Entering directory '/home/admin/nexmon/firmwares'
  EXECUTING MAKE FOR CHIP VERSION bcm4330/
make -C bcm4330/
make[2]: Entering directory '/home/admin/nexmon/firmwares/bcm4330'
  EXECUTING MAKE FOR FIRMWARE VERSION 5_90_100_41_sta/
make -C 5_90_100_41_sta/
make[3]: Entering directory '/home/admin/nexmon/firmwares/bcm4330/5_90_100_41_sta'
  EXTRACTING UCODE
/buildtools/ucode_extractor/ucodeext -r bcmdhd_sta.bin -b 0x2D830 -l 0x8B82 -o ucode.bin
make[3]: /buildtools/ucode_extractor/ucodeext: No such file or directory
make[3]: *** [Makefile:7: ucode.bin] Error 127
make[3]: Leaving directory '/home/admin/nexmon/firmwares/bcm4330/5_90_100_41_sta'
make[2]: *** [Makefile:8: 5_90_100_41_sta/] Error 2
make[2]: Leaving directory '/home/admin/nexmon/firmwares/bcm4330'
make[1]: *** [Makefile:7: bcm4330/] Error 2
make[1]: Leaving directory '/home/admin/nexmon/firmwares'
make: *** [Makefile:5: firmwares] Error 2
root@raspberrypi:/home/admin/nexmon#
```

jlinktu, 2025-04-09:
Bison is part of the dependencies listed in the [README](https://github.com/seemoo-lab/nexmon?tab=readme-ov-file#build-patches-for-bcm43430a1-on-the-rpi3zero-w-or-bcm434355c0-on-the-rpi3rpi4-or-bcm43436b0-on-the-rpi-zero-2w-using-raspbianraspberry-pi-os-recommended). Make sure you follow it.

Sonys9, 2025-04-09:
```

e[1]: Entering directory '/home/admin/nexmon/firmwares'
  EXECUTING MAKE FOR CHIP VERSION bcm4330/
make -C bcm4330/
make[2]: Entering directory '/home/admin/nexmon/firmwares/bcm4330'
  EXECUTING MAKE FOR FIRMWARE VERSION 5_90_100_41_sta/
make -C 5_90_100_41_sta/
make[3]: Entering directory '/home/admin/nexmon/firmwares/bcm4330/5_90_100_41_sta'
  EXTRACTING UCODE
/buildtools/ucode_extractor/ucodeext -r bcmdhd_sta.bin -b 0x2D830 -l 0x8B82 -o ucode.bin
make[3]: /buildtools/ucode_extractor/ucodeext: No such file or directory
make[3]: *** [Makefile:7: ucode.bin] Error 127
make[3]: Leaving directory '/home/admin/nexmon/firmwares/bcm4330/5_90_100_41_sta'
make[2]: *** [Makefile:8: 5_90_100_41_sta/] Error 2
make[2]: Leaving directory '/home/admin/nexmon/firmwares/bcm4330'
make[1]: *** [Makefile:7: bcm4330/] Error 2
make[1]: Leaving directory '/home/admin/nexmon/firmwares'
make: *** [Makefile:5: firmwares] Error 2
root@raspberrypi:/home/admin/nexmon#

```

jlinktu, 2025-04-09:
Follow the README from the beginning to end.

Sonys9, 2025-04-09:
```

  COLLECTING STATISTICS read /home/admin/nexmon/STATISTICS.md for more information
  COMPILING src/version.c => obj/version.o (details: log/compiler.log)
  GENERATING LINKER FILE gen/nexmon.pre => gen/nexmon.ld
  GENERATING LINKER FILE gen/nexmon.pre => gen/flashpatches.ld
  LINKING OBJECTS => gen/patch.elf (details: log/linker.log, log/linker.err)
  GENERATING MAKE FILE gen/nexmon.pre => gen/nexmon.mk
  GENERATING MAKE FILE gen/nexmon.pre => gen/flashpatches.mk
  APPLYING FLASHPATCHES gen/flashpatches.mk => brcmfmac43430-sdio.bin (details: log/flashpatches.log)
  APPLYING PATCHES gen/nexmon.mk => brcmfmac43430-sdio.bin (details: log/patches.log)
  BUILDING DRIVER for kernel 6.1 brcmfmac_6.1.y-nexmon/brcmfmac.ko (details: log/driver.log)
root@raspberrypi:/home/admin/nexmon/patches/bcm43430a1/7_45_41_46/nexmon# ls
brcmfmac43430-sdio.bin       brcmfmac43430-sdio.raspberrypi,3-model-b.clm_blob  gen      Makefile  src
brcmfmac43430-sdio.bin.orig  brcmfmac.patch                                     include  obj
brcmfmac43430-sdio.clm_blob  BUILD_NUMBER                                       log      patch.ld
root@raspberrypi:/home/admin/nexmon/patches/bcm43430a1/7_45_41_46/nexmon# cat log/
compiler.log      error.log         linker.err        patches.log
driver.log        flashpatches.log  linker.log
root@raspberrypi:/home/admin/nexmon/patches/bcm43430a1/7_45_41_46/nexmon# cat log/
compiler.log      error.log         linker.err        patches.log
driver.log        flashpatches.log  linker.log
root@raspberrypi:/home/admin/nexmon/patches/bcm43430a1/7_45_41_46/nexmon# cat log/driver.log
make[1]: Entering directory '/usr/src/linux-headers-6.1.21-v8+'
make[1]: Leaving directory '/usr/src/linux-headers-6.1.21-v8+'
root@raspberrypi:/home/admin/nexmon/patches/bcm43430a1/7_45_41_46/nexmon#
```

where is this driver?

Sonys9, 2025-04-09:
root@raspberrypi:/home/admin# sudo find /home/admin -name "brcmfmac.ko" 2>/dev/null
/home/admin/nexmon/patches/driver/brcmfmac_6.1.y-nexmon/brcmfmac.ko
root@raspberrypi:/home/admin#
found!

closed 2025-04-09 by jlinktu