seemoo-lab/nexmon #656: Aircrack-ng and Iw dev error -95

Sonys9, 2025-04-09:
i see there is a "* monitor" in iw list, i made all what written in Readme, but i got error -95. My device is RPI02W

jlinktu, 2025-04-09:
see #500

closed 2025-04-09 by jlinktu

Sonys9, 2025-04-09:
is nexmon supports kernel 6.x? 
i have got 524 error and Airdump-ng is empty:(

Sonys9, 2025-04-09:
[image]

Sonys9, 2025-04-09:
[   20.652170] Bluetooth: MGMT ver 1.22
[   20.672614] NET: Registered PF_ALG protocol family
[  589.579490] brcmfmac: brcmf_vif_add_validate: Attempt to add a MONITOR interface...
[  589.579545] brcmfmac: brcmf_mon_add_vif: brcmf_mon_add_vif called
[  589.579566] brcmfmac: brcmf_mon_add_vif: Adding vif "mon0"
[  592.981409] device mon0 entered promiscuous mode
root@raspberrypi:/home/admin#