seemoo-lab/nexmon #660: Support for Pixel 9's bcm4390

Ma5hmo, 2025-04-28:
Hey,

I've tried adding support for the bcm4390 (fw_bcmdhd.bin_4390_b1) chip by modifying bcm4398d0's definitions.mk and reverse engineering the firmware. However the problem is when running the firmware on the device, the chip crashes. I can understand from the kernel logs that the chip trapped on the _hnd_die function, however I don't know how to dump the call stack of the chip.

Moreover, I tried dumping the ROM using dhdutil, but got this error:
tokay:/# dhdutil -i wlan0 membytes -r 0x2a0000 3
dhdutil: __dhd_driver_io: Operation not supported on transport endpoint

More information:
<FW-TAG>4390b1-roml/config_pcie_release2 Version=27.843.27.1 (wlan=r1073982 b1 release2) Date=2024-07-12T07:05:28Z FWID=01-5eef7ad0
<UCODE-TAG>path=.../dot11_firmware/branches/UR2710Ucode_twig_46614_100@46819 major=46614 minor=100 diff_ct=0 prebuilt=0 build=/projects/wcc_swbuild_ext20_release/build_linux_ext20_release/URIAL2710T843R27_REL_27_843_27_1/hndrte-dongle-wl/2024.7.11.0/build/dongle/4390b1-roml/config_pcie_release2@gg/ucode_build

If more information is necessary please update me.
Thanks for any support!

Kavanyy, 2025-05-06:
Hey!
I'd like to know how are you dumping the ROM using dhdutil.
Thanks in advance!

Kavanyy, 2025-05-06:
Could you please explain how you determined the offset 0x2a0000 for the dhdutil membytes command? In other words, where did you get that exact address from documentation, a prior dump, or a memory‐map?

Ma5hmo, 2025-05-06:
0x2a0000 is the RAM start, 0x10000 is the ROM start. I've figured these numbers from the bcm4398d0 definitions.mk file in the firmware folder, which aligns with the `fw_bcmdhd.map_4390_b1` file found in the /vendor/firmware folder in the Pixel 9 device.

`dhdutil membytes` doesn't work in the ROM field either, I've just used these numbers as an example for showing it can't even dump memory from the RAM.

> Could you please explain how you determined the offset 0x2a0000 for the dhdutil membytes command? In other words, where did you get that exact address from documentation, a prior dump, or a memory‐map?

Kavanyy, 2025-05-06:
Yes, since you compiled dhdutil from Nexmon, the tool includes support for commands like membytes. However, this will only work if the kernel's dhd.ko module actually implements the necessary ioctls.

The error
__dhd_driver_io: Operation not supported on transport endpoint
usually means that the driver doesn't support that command—either because the feature is not compiled in, or because it's intentionally blocked.

To make membytes work, the dhd driver must be rebuilt with certain flags enabled. These include:

CONFIG_BCMDHD_FULL_SEP_BIN=y
DHD_EXPORT_CNTL_FILE=y
DHD_DEBUG=y
DHD_DUMP_MEM_READ=y

If you're working with a Pixel 9, Google provides the driver source here:
https://android.googlesource.com/kernel/google-modules/wlan/bcmdhd

You can clone it with:
git clone https://android.googlesource.com/kernel/google-modules/wlan/bcmdhd

After modifying the config and rebuilding dhd.ko, you can push the new module to your device and test dhdutil membytes again.

I don't own a Pixel 9 myself, so I can't test this directly so if you're able to rebuild the module and try it out, let me know how it goes.

Ma5hmo, 2025-05-07:
Thanks a lot! I will try it out and update you

Kavanyy, 2025-05-12:
Hello again!

I tried following my own instructions, but I still can’t use dhdutil. I’ve looked into the issue and believe they’re now using a custom ioctl.

There are several Nexmon projects that include a ROM-extractor tool, which lets you obtain the clean ROM before the flash patches are applied, as it’s loaded into RAM during the phone’s boot process.

I’m not entirely sure how to use the ROM-extraction tool referenced for other chips, so I’d appreciate some guidance on how it works and on how to obtain the u-code offsets. Is there another ioctl to obtain a live image of the ram?

Please keep me updated if you find anything useful!

Ma5hmo, 2025-05-12:
Hey!

I don't know much about other rom extraction tools, and I don't think there's another ioctl to obtain a live image of the ram. I'll dig deeper into that after building the driver. Can you actually give me some guidance on how you have done that? That will help me a bunch!

Thanks!

Kavanyy, 2025-05-12:
I really think you should check this issue, where it is explained everything related to the dhdutil not working on modern SoC.
https://github.com/seemoo-lab/nexmon/issues/247

I also encourage you to check this valuable project from the Pixel8 (thank you Jakob for the effort): 
https://github.com/seemoo-lab/nexmon/tree/master/patches/bcm4398d0/24_671_6_9_r1031525/rom_extraction/src

I still dont know how that project works as its not documented yet, i'll keep an eye on it as for it know im not absolutely certain how to use it tho. I have so much questions yet, but im still trying to do an effort here..

Anyways im still concern and dont really understand how he obtains the ucode offsets to compress them with the ucode compression tool. 

Keep me posted if you find anything interesting!

Ma5hmo, 2025-05-26:
> Keep me posted if you find anything interesting!

Hey!

I've actually been trying to patch the firmware manually, but it keeps crashing on init. I've tried adding a custom ioctl hook by adding it right before the __pathdr entries and replacing the original address of the wlc_doioctl_patch with the address of the new patched function.

Do you have some ideas to what could be the crash reason?

Also, if you have an idea to how to retrieve a coredump from the chip it would be relevant aswell.

Thanks in advance.

jlinktu, 2025-06-04:
Added.

closed 2025-06-04 by jlinktu